merbanan/rtl_433 #75: Adding WS2310 weather station support

ygator, 2015-01-17:
Can someone help me.  I'd like to implement the handling from a WS2310 weather station.
In analyse mode it is picking it up fine (except for the bits are flipped).  How do a go about adding this decoding based on the following output (two transmissions) from analyse mode?  If needed I can post more.  Thanks

*** signal_start = 110666412, signal_end = 111177557
signal_len = 511145,  pulses = 416
Iteration 1. t: 207    min: 68 (223)    max: 346 (193)    delta 8
Iteration 2. t: 207    min: 68 (223)    max: 346 (193)    delta 0
Pulse coding: Short pulse length 68 - Long pulse length 346

Short distance: 325, long distance: 357, packet distance: 37641

p_limit: 207

[00] {52} f6 f1 28 1d 69 29 20 : 11110110 11110001 00101000 00011101 01101001 00101001 00100000
[01] {52} f6 a1 28 19 dd 62 10 : 11110110 10100001 00101000 00011001 11011101 01100010 00010000
[02] {52} f6 91 28 1f ff 00 20 : 11110110 10010001 00101000 00011111 11111111 00000000 00100000
[03] {52} f6 81 28 00 1f fe 00 : 11110110 10000001 00101000 00000000 00011111 11111110 00000000
[04] {52} f6 b1 28 0d 69 29 d0 : 11110110 10110001 00101000 00001101 01101001 00101001 11010000
[05] {52} f6 e1 28 09 dd 62 40 : 11110110 11100001 00101000 00001001 11011101 01100010 01000000
[06] {52} f6 d1 28 0f ff 00 50 : 11110110 11010001 00101000 00001111 11111111 00000000 01010000
[07] {52} f6 81 28 00 1f fe 00 : 11110110 10000001 00101000 00000000 00011111 11111110 00000000

*** signal_start = 142666549, signal_end = 143048815
signal_len = 382266,  pulses = 312
Iteration 1. t: 208    min: 70 (159)    max: 347 (153)    delta 13
Iteration 2. t: 208    min: 70 (159)    max: 347 (153)    delta 0
Pulse coding: Short pulse length 70 - Long pulse length 347

Short distance: 323, long distance: 355, packet distance: 37850

p_limit: 208

[00] {52} f6 f1 2a 1d 69 29 40 : 11110110 11110001 00101010 00011101 01101001 00101001 01000000
[01] {52} f6 a1 2a 19 dd 62 30 : 11110110 10100001 00101010 00011001 11011101 01100010 00110000
[02] {52} f6 81 2a 00 1f fe 20 : 11110110 10000001 00101010 00000000 00011111 11111110 00100000
[03] {52} f6 b1 2a 0d 69 29 f0 : 11110110 10110001 00101010 00001101 01101001 00101001 11110000
[04] {52} f6 e1 2a 09 dd 62 60 : 11110110 11100001 00101010 00001001 11011101 01100010 01100000
[05] {52} f6 81 2a 00 1f fe 20 : 11110110 10000001 00101010 00000000 00011111 11111110 00100000

rct, 2015-01-17:
It looks like your station is a LaCrosse, is that correct?  

I might be able to modify the LaCrosse TX temperature and humidity code I added to handle this.

Can you run rtl_433 analyze module (-a) again with -t to capture the signals into files?   Then post the files with what the readings should be when parsed?

ygator, 2015-01-18:
I appreciate that rct.  Yes it is WS-2310TWC from La Crosse Technology.  More information can be found at http://www.lacrossetechnology.com/2310twc/
I'll post a link to the information tomorrow.

Thanks again

ygator, 2015-01-18:
Here is the information you requested.  https://drive.google.com/file/d/0B5D4Lm5bUBiQb0xVYVZiZnBxNjg/view
Let me know if you need anything else.

Thanks

rct, 2015-01-18:
@ygator - Ok, you got pretty far along before I got back to you.  Here are a few notes

See http://www.f6fbb.org/domo/sensors/tx3_th.php 

1. As you realized, the bits are inverted - LaCrosse uses a longer pulse for zero, the opposite of some other signals.
2. That extra trailing 0xF is from rtl_433, which is padding the odd number of nybbles into a full byte.
3. The temperature for the LaCrosse TX sensors is sent with +50, which makes the minimum possible reading be -50.0 C (there is no negative coding in the transmission, hence the added +50.   It looks like you've noticed for your sensor that it is actually +30, instead of +50.   This makes your minimum possible value -30 C (-22 F).  That seems reasonable, but just to check are you sure about the value?  
4. For the TX sensors, it was guessed the second nybble which is always 0xA was the number of nybbles in the transmission (not including the start == 0).  However your's is 12 nybbles and a value of 9, so the interpretation was wrong / different.
5. In the TX sensors, there is an address which is set randomly when the device powers up.  You need to have the reset both the receiver and the transmitter when changing batteries so they sync up on the address byte.   It might be the same for your sensor.

ygator, 2015-01-18:
Yes, it is definitely 30 which also matches the base station memory map for the device found at http://www.lavrsen.dk/foswiki/bin/view/Open2300/OpenWSMemoryMap.

Another difference in regards to the linked decoding you included is that the repeated first and second nibble values are inverted and not just duplicated.

The sensor/transmitter is not easily gotten to so I'll have to wait for some better weather before I'll be able to test reset and hooking up the rain and wind sensors.

ygator, 2015-02-02:
I have it working.  Need to test other sensors when I get a chance to get at the sensor/transmitter when the weather gets better.

closed 2015-02-02 by ygator