merbanan/rtl_433 #78: Add plugin support for device types

Batilan, 2015-01-17:
@merbanan: good to see you back and active again. Thanks for this great piece of software!

Lately I have been working on a plugin infrastructure for rtl_433 so that device types can be added at runtime by simply adding or removing a library file per device type. This has been implemented and seems to be working  OK. See my branch plugins at https://github.com/Batilan/rtl_433/tree/plugins.

Besides the device callbacks I'm also thinking about implementing the PWM decoders as plugins. This would allow for  easier development of alternative (PWM) decoders. Currently the PWM decoders are called by all devices with different time settings for each received message. I think this can probably be made more efficient. 

I think for low end devices (like Raspberry) this construct together with calling all all availabale device callbacks as is done now stresses some of these devices quite heavily already and is not really scalable if we want to add more and more devices. The factoring out of the device callbacks by @helgew is quite an improvement and makes adding devices and avoiding conflicts easier (I needed to to the same for the plugin module).

The reason I have not yet submitted a pull request is that I would like to improve the plugin interface so that it is usable in a more generic way. Also the plugin module needs some adaptions to make it work for Windows environments (is this software actually compiled and used by Windows users?). I have tested it only for different Linux environments like Raspberry and x86 distributions.

I would like to exchange some idea's and prevent duplicating of work, however a Github  issue is probably not the best communication medium for this?. Lately I have also been exchanging ideas with @rct who has also been doing some nice improvements on rtl_433. I'm motivated  to share idea's with any other developer that has idea's about improving the great rtl_433 software. I understand @merbanan  also has idea's about improving the software but has limited time, I would like to help with structural improvements making the rtl_433 more scalable so we can remove the line "This software is mostly useable for developers right now."

Looking forward for any responses.

merbanan, 2015-01-17:
How about starting a mailing list ? I agree on what you write. There needs to be a better "packet" format we can send to decoders, right now we are missing the length of the bits in ms, a lot of formats can be separated by that. And we need to add test recordings of the supported formats so we know when we break things or not.

helgew, 2015-01-17:
This might work as a good place for discussions: http://www.discourse.org/ . I have not tried it myself though.

merbanan, 2015-01-17:
I have created the google group rtl_433. Can you try it ?

merbanan, 2015-01-17:
https://groups.google.com/d/forum/rtl_433

merbanan, 2015-01-17:
If this works fine I'll add it on the main page.

helgew, 2015-01-17:
Apparently, this is a closed group... not sure that's what you wanted :)

Batilan, 2015-01-17:
I just signed up

0xFACE, 2015-01-17:
Works for me ;-)

merbanan, 2015-01-17:
Soon emails may even work :)

deennoo, 2015-01-24:
I just signes

merbanan, 2015-01-26:
I will close this issue. Reopen if needed.

closed 2015-01-26 by merbanan