merbanan/rtl_433 #79: Some? callbacks not occuring with the new merged code base

rct, 2015-01-18:
something unexpected has changed with the new merged code base from pull request #74  I'm not getting any LaCrosse output.  I'm also getting a very low number of call backs.

rct, 2015-01-18:
Looks like @helgew merges lost some fixes that were in other pull requests that merbanan had merged.

Probably @helgew changes should have been rebased first??

So far missing are:
* #67 
* #41 
* #48 / #62

so far the three fixes I checked got overwritten.

helgew, 2015-01-18:
I'll have a look at this. I made the mistake to factor out the device code first and then merged in all those pull requests. That involved multiple rebasings and conflict resolutions... :/

helgew, 2015-01-18:
Here is what was missing:
* #69's fix (fixed #67)  to signal handling, which was also in #63 (fixed issue #41)
* PR #62's change on line 952 (fixed #48)

These are fixed in PR #82.

rct, 2015-01-22:
Just pulled down the latest.   I doesn't compile because of a missing closing brace in oregon_scientific.  After fixing that. See Issue #89 

 The good news is the LaCrosse code works.   However, the SIGPIPE fix is STILL MISSING.   It got applied to the WINDOWS ifdef, where I'm not sure if it is even needed or will compile. 

The SIGPIPE FIX needs to be in the non windows signal handler.  IF NOT, and you pipe stderr into another program to do something with the output, when you CTRL-C, rtl_433 goes into an infinite loop, using %100 cpu, because the signal handler writes to stderr, which it shouldn't do, but that causes, another SIGPIPE because stderr is closed, so the signal handler gets invoked again, ....

It should look like this:  

https://github.com/rct/rtl_433/commit/ab932e04f20f7abc2b750ea42985f41faaf45b63

```
 #else
static void sighandler(int signum)
{
- fprintf(stderr, "Signal caught, exiting!\n");
+ if (signum == SIGPIPE) {
+ signal(SIGPIPE,SIG_IGN);
+ } else {
+ fprintf(stderr, "Signal caught, exiting!\n");
+ }
do_exit = 1;
rtlsdr_cancel_async(dev);
}
```
NOTE: it is after the #else for the Windows #ifdef.

helgew, 2015-01-23:
eventually, I will get it right again :/ (see PR #90)

rct, 2015-01-23:
Diff looks good -- Thanks!

merbanan, 2015-01-26:
I'll close this now, reopen if needed.

closed 2015-01-26 by merbanan