merbanan/rtl_433 #81: error making last realase

deennoo, 2015-01-18:
on make action i'm facing some error :+1: 
/home/pi/rtl_433plug/include/rtl_433.h:48:30: error: unknown type name ‘uint8_t’
/home/pi/rtl_433plug/include/rtl_433.h:49:1: warning: no semicolon at end of struct or union [enabled by default]
/home/pi/rtl_433plug/include/rtl_433.h:52:20: error: unknown type name ‘uint8_t’

and 

/home/pi/rtl_433plug/src/devices/silvercrest.c:3:33: error: unknown type name ‘uint8_t’
/home/pi/rtl_433plug/src/devices/silvercrest.c:33:30: error: ‘silvercrest_callback’ undeclared here (not in a function)
/home/pi/rtl_433plug/src/devices/silvercrest.c:33:29: warning: excess elements in struct initializer [enabled by default]
/home/pi/rtl_433plug/src/devices/silvercrest.c:33:29: warning: (near initialization for ‘silvercrest’) [enabled by default]

helgew, 2015-01-18:
There is an include missing, which is addressed in PR #77.

deennoo, 2015-01-18:
OK donne sorry

closed 2015-01-18 by deennoo