merbanan/rtl_433 #85: Code fails to compile (unknown type name 'uint8_t') ('silvercrest_callback' undeclared here)

shodanx2, 2015-01-20:
Hi,

I'm trying to control etekcity remote controlled outlets (http://goo.gl/gB4jFB), I'm going to use them to control various stuff in my house, lights, blinds, fans, water pump etc.. And also to get input from various security system sensors (beam breakers, magnetic switches, IR sensors, glass break detector etc..)

I tried compiling the software but it fails, here is the console output.

----------------------------------------------------------

-----------------------------------------------------------

motoko build # cmake ../
-- The C compiler identification is GNU 4.8.3
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.19
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib64/libusb-1.0.so
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Udev rules not being installed, install them with -DINSTALL_UDEV_RULES=ON
-- Configuring done
-- Generating done
-- Build files have been written to: /root/rtl_433-master/build

motoko build # make

Scanning dependencies of target rtlsdr_shared
[  3%] Building C object src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.o
/root/rtl_433-master/src/librtlsdr.c: In function 'e4000_exit':
/root/rtl_433-master/src/librtlsdr.c:104:22: warning: unused parameter 'dev' [-Wunused-parameter]
 int e4000_exit(void *dev) { return 0; }
                      ^
/root/rtl_433-master/src/librtlsdr.c: In function 'fc0012_exit':
/root/rtl_433-master/src/librtlsdr.c:148:23: warning: unused parameter 'dev' [-Wunused-parameter]
 int fc0012_exit(void *dev) { return 0; }
                       ^
/root/rtl_433-master/src/librtlsdr.c: In function 'fc0012_set_bw':
/root/rtl_433-master/src/librtlsdr.c:154:25: warning: unused parameter 'dev' [-Wunused-parameter]
 int fc0012_set_bw(void *dev, int bw) { return 0; }
                         ^
/root/rtl_433-master/src/librtlsdr.c:154:34: warning: unused parameter 'bw' [-Wunused-parameter]
 int fc0012_set_bw(void *dev, int bw) { return 0; }
                                  ^
/root/rtl_433-master/src/librtlsdr.c: In function 'fc0012_set_gain_mode':
/root/rtl_433-master/src/librtlsdr.c:156:32: warning: unused parameter 'dev' [-Wunused-parameter]
 int fc0012_set_gain_mode(void *dev, int manual) { return 0; }
                                ^
/root/rtl_433-master/src/librtlsdr.c:156:41: warning: unused parameter 'manual' [-Wunused-parameter]
 int fc0012_set_gain_mode(void *dev, int manual) { return 0; }
                                         ^
/root/rtl_433-master/src/librtlsdr.c: In function 'fc0013_exit':
/root/rtl_433-master/src/librtlsdr.c:159:23: warning: unused parameter 'dev' [-Wunused-parameter]
 int fc0013_exit(void *dev) { return 0; }
                       ^
/root/rtl_433-master/src/librtlsdr.c: In function 'fc0013_set_bw':
/root/rtl_433-master/src/librtlsdr.c:163:25: warning: unused parameter 'dev' [-Wunused-parameter]
 int fc0013_set_bw(void *dev, int bw) { return 0; }
                         ^
/root/rtl_433-master/src/librtlsdr.c:163:34: warning: unused parameter 'bw' [-Wunused-parameter]
 int fc0013_set_bw(void *dev, int bw) { return 0; }
                                  ^
[... 70 lines cut ...]
/root/rtl_433-master/include/rtl_433.h:48:30: error: unknown type name 'uint8_t'
     int     (*json_callback)(uint8_t bits_buffer[BITBUF_ROWS][BITBUF_COLS],int16_t bits_per_row[BITBUF_ROWS]) ;
                              ^
/root/rtl_433-master/include/rtl_433.h:49:1: warning: no semicolon at end of struct or union [enabled by default]
 } r_device;
 ^
/root/rtl_433-master/include/rtl_433.h:52:20: error: unknown type name 'uint8_t'
 int debug_callback(uint8_t buffer[BITBUF_ROWS][BITBUF_COLS], int16_t bits_per_row[BITBUF_ROWS]);
                    ^
/root/rtl_433-master/src/devices/silvercrest.c:3:33: error: unknown type name 'uint8_t'
 static int silvercrest_callback(uint8_t bb[BITBUF_ROWS][BITBUF_COLS],int16_t bits_per_row[BITBUF_ROWS]) {
                                 ^
/root/rtl_433-master/src/devices/silvercrest.c:33:30: error: 'silvercrest_callback' undeclared here (not in a function)
     /* .json_callback  = */ &silvercrest_callback,
                              ^
/root/rtl_433-master/src/devices/silvercrest.c:33:29: warning: excess elements in struct initializer [enabled by default]
     /* .json_callback  = */ &silvercrest_callback,
                             ^
/root/rtl_433-master/src/devices/silvercrest.c:33:29: warning: (near initialization for 'silvercrest') [enabled by default]
src/CMakeFiles/rtl_433.dir/build.make:80: recipe for target 'src/CMakeFiles/rtl_433.dir/devices/silvercrest.c.o' failed
make[2]: *** [src/CMakeFiles/rtl_433.dir/devices/silvercrest.c.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'src/CMakeFiles/rtl_433.dir/all' failed
make[1]: *** [src/CMakeFiles/rtl_433.dir/all] Error 2
Makefile:116: recipe for target 'all' failed
make: *** [all] Error 2

helgew, 2015-01-20:
This is a duplicate of #76 and a pull request with the fix has been submitted as part of pull request #82. For now, you can add '#include <stdint.h>' in include/rtl_433.h to fix the problem.

shodanx2, 2015-01-20:
> '#include ' in include/rtl_433.h to fix the problem.

Added

#include <stdint.h>

to rtl_433-master/include/rtl_433.h and now it compiled successfully !

[100%] Built target rtlsdr_static

thanks !

-- 
Jean-Francois Perreault
Professional geek

Inanimate objects can be classified scientifically into three major
categories;
those that don't work, those that break down and those that get lost.
-*Russell Baker *

"It ain't what you don't know that gets you into trouble. It's what you
know for sure that just ain't so."
-* Mark **Twain*

Computers are like Old Testament gods; lots of rules and no mercy.

- *Joseph Campbell*

shodanx2, 2015-01-20:
And it totally works 

*** signal_start = 56791174, signal_end = 56876103
signal_len = 84929,  pulses = 275
Iteration 1. t: 96    min: 44 (190)    max: 149 (85)    delta 61
Iteration 2. t: 96    min: 44 (190)    max: 149 (85)    delta 0
Pulse coding: Short pulse length 44 - Long pulse length 149

Short distance: 35, long distance: 145, packet distance: 1463

p_limit: 96

[00] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[01] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[02] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[03] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[04] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[05] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[06] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[07] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[08] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[09] {25} 05 17 03 00 : 00000101 00010111 00000011 00000000
[10] {25} 05 15 00 00 : 00000101 00010101 00000000 00000000

*** signal_start = 58143679, signal_end = 58228599
signal_len = 84920,  pulses = 275
Iteration 1. t: 93    min: 41 (190)    max: 146 (85)    delta 61
Iteration 2. t: 93    min: 41 (190)    max: 146 (85)    delta 0
Pulse coding: Short pulse length 41 - Long pulse length 146

Short distance: 38, long distance: 148, packet distance: 1468

p_limit: 93

[00] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[01] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[02] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[03] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[04] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[05] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[06] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[07] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[08] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[09] {25} 05 17 0c 00 : 00000101 00010111 00001100 00000000
[10] {25} 05 15 00 00 : 00000101 00010101 00000000 00000000

helgew, 2015-01-20:
Awesome!

Would you mind closing the issue?

shodanx2, 2015-01-20:
Is there a place to post the results I have with various 433 mhz devices and rtl_433 ?
I could also make scope traces of the transmitters with circuit pictures etc..

closed 2015-01-20 by shodanx2