merbanan/rtl_433 #86: Device dongle detected incorrectly sometimes

ygator, 2015-01-21:
Every once in a while my device dongle will be detected as an E4000 when it is a R820T.
Does anyone else have this issue?

# ./rtl_433
Registering protocol[01] Rubicson Temperature Sensor
Registering protocol[02] Silvercrest Remote Control
Registering protocol[03] ELV EM 1000
Registering protocol[04] ELV WS 2000
Registering protocol[05] Waveman Switch Transmitter
Registering protocol[06] Steffen Switch Transmitter
Registering protocol[07] Acurite 5n1 Weather Station
Registering protocol[08] Acurite Temperature and Humidity Sensor
Registering protocol[09] Acurite 896 Rain Gauge
Registering protocol[10] LaCrosse TX Temperature / Humidity Sensor
Registering protocol[11] Oregon Scientific Weather Sensor
Registering protocol[12] KlikAanKlikUit Wireless Switch
Registering protocol[13] AlectoV1 Weather Sensor
Registering protocol[14] Intertechno 433
Registering protocol[15] Mebus 433
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: ezcap USB 2.0 DVB-T/DAB/FM dongle
Found Elonics E4000 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Sample rate decimation set to 0. 250000->250000
Bit detection level set to 10000.
Tuner gain set to Auto.
Reading samples in async mode...
[E4K] PLL not locked for 433920000 Hz!
WARNING: Failed to set center freq.
^CSignal caught, exiting!

User cancel, exiting...

# ./rtl_433
Registering protocol[01] Rubicson Temperature Sensor
Registering protocol[02] Silvercrest Remote Control
Registering protocol[03] ELV EM 1000
Registering protocol[04] ELV WS 2000
Registering protocol[05] Waveman Switch Transmitter
Registering protocol[06] Steffen Switch Transmitter
Registering protocol[07] Acurite 5n1 Weather Station
Registering protocol[08] Acurite Temperature and Humidity Sensor
Registering protocol[09] Acurite 896 Rain Gauge
Registering protocol[10] LaCrosse TX Temperature / Humidity Sensor
Registering protocol[11] Oregon Scientific Weather Sensor
Registering protocol[12] KlikAanKlikUit Wireless Switch
Registering protocol[13] AlectoV1 Weather Sensor
Registering protocol[14] Intertechno 433
Registering protocol[15] Mebus 433
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: ezcap USB 2.0 DVB-T/DAB/FM dongle
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Sample rate decimation set to 0. 250000->250000
Bit detection level set to 10000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.

merbanan, 2015-01-23:
That is an error from the rtl_sdr libraries. rtl_433 is not responsible for that and can not affect that.

ygator, 2015-01-24:
It is happening a lot more now that I moved to the latest code.  I also see that it is no longer compiled in and is using the library.  I am on archlinuxarm and the rtl-sdr is version rtl-sdr-20140210-1.  I'll have to check if this is really the latest or not.  Since it is no longer compiled in I will close this issue.
Thanks.

closed 2015-01-24 by ygator