merbanan/rtl_433 #87: make: *** [all] Error 2

pclov3r, 2015-01-21:
I'm unable to compile this on CentOS 7

The <stdint.h> patch doesn't fix the problem as it did for others it appears. 

[root@VMware-CentOS build]# cmake ../
-- The C compiler identification is GNU 4.8.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.27.1")
-- checking for module 'libusb-1.0'
--   found libusb-1.0, version 1.0.15
-- Found libusb-1.0: /usr/include/libusb-1.0, /usr/lib64/libusb-1.0.so
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Udev rules not being installed, install them with -DINSTALL_UDEV_RULES=ON
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/rtl_433/build
[root@VMware-CentOS build]# make
Scanning dependencies of target rtlsdr_shared
[  3%] Building C object src/CMakeFiles/rtlsdr_shared.dir/librtlsdr.c.o
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘e4000_exit’:
/usr/local/src/rtl_433/src/librtlsdr.c:104:22: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int e4000_exit(void *dev) { return 0; }
                      ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc0012_exit’:
/usr/local/src/rtl_433/src/librtlsdr.c:148:23: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int fc0012_exit(void *dev) { return 0; }
                       ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc0012_set_bw’:
/usr/local/src/rtl_433/src/librtlsdr.c:154:25: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int fc0012_set_bw(void *dev, int bw) { return 0; }
                         ^
/usr/local/src/rtl_433/src/librtlsdr.c:154:34: warning: unused parameter ‘bw’ [-Wunused-parameter]
 int fc0012_set_bw(void *dev, int bw) { return 0; }
                                  ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc0012_set_gain_mode’:
/usr/local/src/rtl_433/src/librtlsdr.c:156:32: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int fc0012_set_gain_mode(void *dev, int manual) { return 0; }
                                ^
/usr/local/src/rtl_433/src/librtlsdr.c:156:41: warning: unused parameter ‘manual’ [-Wunused-parameter]
 int fc0012_set_gain_mode(void *dev, int manual) { return 0; }
                                         ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc0013_exit’:
/usr/local/src/rtl_433/src/librtlsdr.c:159:23: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int fc0013_exit(void *dev) { return 0; }
                       ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc0013_set_bw’:
/usr/local/src/rtl_433/src/librtlsdr.c:163:25: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int fc0013_set_bw(void *dev, int bw) { return 0; }
                         ^
/usr/local/src/rtl_433/src/librtlsdr.c:163:34: warning: unused parameter ‘bw’ [-Wunused-parameter]
 int fc0013_set_bw(void *dev, int bw) { return 0; }
                                  ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc2580_exit’:
/usr/local/src/rtl_433/src/librtlsdr.c:167:23: warning: unused parameter ‘dev’ [-Wunused-parameter]
 int fc2580_exit(void *dev) { return 0; }
                       ^
/usr/local/src/rtl_433/src/librtlsdr.c: In function ‘fc2580_set_bw’:
/usr/local/src/rtl_433/src/librtlsdr.c:171:34: warning: unused parameter ‘bw’ [-Wunused-parameter]
 int fc2580_set_bw(void *dev, int bw) { return fc2580_SetBandwidthMode(dev, 1); }
                                  ^
[... 81 lines cut ...]
/usr/local/src/rtl_433/src/devices/oregon_scientific.c: In function ‘get_os_temperature’:
/usr/local/src/rtl_433/src/devices/oregon_scientific.c:3:63: warning: unused parameter ‘sensor_id’ [-Wunused-parameter]
 float get_os_temperature(unsigned char *message, unsigned int sensor_id) {
                                                               ^
/usr/local/src/rtl_433/src/devices/oregon_scientific.c: In function ‘get_os_humidity’:
/usr/local/src/rtl_433/src/devices/oregon_scientific.c:11:67: warning: unused parameter ‘sensor_id’ [-Wunused-parameter]
 unsigned int get_os_humidity(unsigned char *message, unsigned int sensor_id) {
                                                                   ^
/usr/local/src/rtl_433/src/devices/oregon_scientific.c: In function ‘oregon_scientific_v2_1_parser’:
/usr/local/src/rtl_433/src/devices/oregon_scientific.c:55:88: warning: unused parameter ‘bits_per_row’ [-Wunused-parameter]
 static int oregon_scientific_v2_1_parser(uint8_t bb[BITBUF_ROWS][BITBUF_COLS], int16_t bits_per_row[BITBUF_ROWS]) {
                                                                                        ^
/usr/local/src/rtl_433/src/devices/oregon_scientific.c: In function ‘oregon_scientific_v3_parser’:
/usr/local/src/rtl_433/src/devices/oregon_scientific.c:251:2: error: expected ‘}’ before ‘else’
  else if ((msg[0] == 0x19) && (msg[1] == 0x84)) {
  ^
/usr/local/src/rtl_433/src/devices/oregon_scientific.c:192:86: warning: unused parameter ‘bits_per_row’ [-Wunused-parameter]
 static int oregon_scientific_v3_parser(uint8_t bb[BITBUF_ROWS][BITBUF_COLS], int16_t bits_per_row[BITBUF_ROWS]) {
                                                                                      ^
make[2]: *** [src/CMakeFiles/rtl_433.dir/devices/oregon_scientific.c.o] Error 1
make[1]: *** [src/CMakeFiles/rtl_433.dir/all] Error 2
make: *** [all] Error 2
[root@VMware-CentOS build]# grep stdint.h ../include/rtl_433.h
#include <stdint.h>
[root@VMware-CentOS build]#

pclov3r, 2015-01-21:
Fails to build on Fedora 21 as well with the same error output.

pclov3r, 2015-01-21:
changed line 251 of /src/devices/oregon_scientific.c to  
} else if ((msg[0] == 0x19) && (msg[1] == 0x84)) { 

And it compiled without a problem.

rct, 2015-01-22:
Same here, I don't think the OS matters, no compiler should accept that.

Are people submitting pull requests without making sure they even compile?

pclov3r, 2015-01-22:
I would hope not. Guess it happens tho or was overlooked.

closed 2015-01-23 by pclov3r