merbanan/rtl_433 #92: Running section of README.md is out of date

ygator, 2015-01-26:
Current version help shows
rtl_433, an ISM band generic data receiver for RTL2832 based DVB-T receivers

Usage:  [-d device_index (default: 0)]
        [-g gain (default: 0 for auto)]
        [-a analyze mode, print a textual description of the signal]
        [-t signal auto save, use it together with analyze mode (-a -t)
        [-l change the detection level used to determine pulses (0-3200) default: 10000]
        [-f [-f...] receive frequency[s], default: 433920000 Hz]
        [-s samplerate (default: 250000 Hz)]
        [-S force sync output (default: async)]
        [-r read data from file instead of from a receiver]
        [-p ppm_error (default: 0)]
        [-r test file name (indata)]
        [-m test file mode (0 rtl_sdr data, 1 rtl_433 data)]
        [-D print debug info on event
        [-z override short value
        [-x override long value
        filename (a '-' dumps samples to stdout)

merbanan, 2015-01-27:
Fixed. Next time please send a patch.

closed 2015-01-27 by merbanan

ygator, 2015-01-27:
I tried fixing it myself, but it would not let me update your git.  If you tell me the git commands I need to edit this work and submit you a pull request (if that is how it is suppose to work) then I can make the changes myself.  I've tried to figure out what I needed to do, but everything I have tried so far did not work.

merbanan, 2015-01-27:
https://help.github.com/articles/using-pull-requests/