merbanan/rtl_433 #96: rtl_433.c:(.text+0x33d9): undefined reference to `fineoffset_WH2' - Fails to compile on windows

pclov3r, 2015-01-28:
It appears that merge https://github.com/merbanan/rtl_433/pull/94 broke the ability to compile it under windows. 

There is also this odd output on Linux 

Registering protocol[01] Rubicson Temperature Sensor
Registering protocol[02] Prologue Temperature Sensor
Registering protocol[03] Silvercrest Remote Control
Registering protocol[04] ELV EM 1000
Registering protocol[05] ELV WS 2000
Registering protocol[06] Waveman Switch Transmitter
Registering protocol[07] Steffen Switch Transmitter
Registering protocol[08] Acurite 5n1 Weather Station
Registering protocol[09] Acurite Temperature and Humidity Sensor
Registering protocol[10] Acurite 896 Rain Gauge
Registering protocol[11] LaCrosse TX Temperature / Humidity Sensor
Registering protocol[12] Oregon Scientific Weather Sensor
Registering protocol[13] KlikAanKlikUit Wireless Switch
Registering protocol[14] AlectoV1 Weather Sensor
Registering protocol[15] Intertechno 433
Registering protocol[16] Mebus 433
Max number of protocols reached 15
Registering protocol[17] Fine Offset Electronics, WH-2 Sensor
Max number of protocols reached 15

"Max number of protocols reached 15" is shows up twice?

0xFACE, 2015-01-28:
In include/rtl_433.h file 
#define MAX_PROTOCOLS           15
should be increased.

pclov3r, 2015-01-28:
Solved the compile issue. My compiler was doing something weird I have no idea why.

Still the one weird Max number of protocols output issue which 0xFACE suggestion fixed.

merbanan, 2015-02-03:
Fixed, changed to 20.

closed 2015-02-03 by merbanan