merbanan/rtl_433 #97: Feature Request: Disable and enable specific devices.

pclov3r, 2015-02-12:
Not sure how this would be to implement but it would be nice if you could disable or enable specific devices. 

Basically for example if i only wanted to see LaCrosse sensors i could just enable that one device.

helgew, 2015-02-12:
Addressed in #98 :smile:

merbanan, 2015-02-16:
Merged.

closed 2015-02-16 by merbanan

pclov3r, 2015-02-16:
How do I use this new feature with the -R? What do input for the device name?

helgew, 2015-02-16:
When you run rtl_433 with the -h option, it will tell you what devices are supported and each will have a number associated with it. Use -R and one of those numbers to enable just that device. You can use the -R option multiple times to enable additional devices.

pclov3r, 2015-02-19:
Attempting to use this feature right now results in a segfault for me on
linux and doesn't work on Windows

[root@localhost ~]# rtl_433 -R [08] LaCrosse TX Temperature / Humidity
Sensor
Segmentation fault (core dumped)
[root@localhost ~]# rtl_433 -R [08]
Segmentation fault (core dumped)
[root@localhost ~]# rtl_433 -R 08
Segmentation fault (core dumped)
[root@localhost ~]# rtl_433 -R 8
Segmentation fault (core dumped)
[root@localhost ~]#

Am i doing something wrong?