merbanan/rtl_433 #100: Feature Request :: RTL_TCP support

Tehrasha, 2015-02-13:
I leave my dongle on 24/7 in an rlt_tcp session so I can access it from any of my computers.
Currently I can only run rtl_433 from the same machine the dongle is on, and I have to stop the tcp session, and then restart it after each use.

merbanan, 2015-02-16:
Well the idea is that you should decode the signals on the system with the dongle and send the message from there. But it might be possible to push the stream to a FIFO and open it from there. I myself will not develop or investigate this kind of feature. But if someone else sends patches I will accept them.

helgew, 2015-02-17:
There is a [TCP enabled version of librtlsdr](http://www.rtl-sdr.com/tcp-enabled-version-librtlsdr/) being developed, that may fit your needs. I could not get it to work for me, but you may have more luck.

jmccorm, 2017-09-28:
+1 Similar problem to OP. My SDR dongle is dedicated to a Raspberry Pi 3 in the attic, and connected to a fixed antenna (for optimum receptivity). In my household, all SDR software (running on diverse platforms such as Windows PCs, Android and iOS tablets, and Linux on the Raspberry Pi 3 itself) connect to the SDR antenna over the network via rtl_tcp. They share connectivity to the dongle dynamically over the network. ("SDR services on demand" if you want to call it something fancy.)

Today, so many rtl_sdr projects support TCP/IP connectivity these days that it is almost a standard feature.  Please consider prioritizing this feature request?

roger-, 2018-07-04:
Workaround: http://flux242.blogspot.com/2017/04/how-to-connect-rtl433-to-remote-rtltcp.html

merbanan, 2018-11-30:
@zuckschwerdt is this now supported through SoapySDR?

zuckschwerdt, 2018-12-04:
As a first test this should work:
```
rtl_tcp -f 433920000 -s 250000 &
socat tcp:127.0.0.1:1234 - | rtl_433 -q -c 0 -C si -F json -r -
```

zuckschwerdt, 2018-12-04:
There is some basic support in #894.
Closing this, further discussion and tracking in PR.

closed 2018-12-04 by zuckschwerdt