merbanan/rtl_433 #102: TFA 30.3125 compatible?

lanwin, 2015-02-18:
Ive looking for a way to get temperature and humidity for all of our rooms within for a reasonable price in Germany. 

I already have a simple TFA weather station with a simple temperature sensor at 433mhz and for testing also bought a TFA 30.3180.IT at 868mhz, but I never manged to get any useful package from booth. I saw a reaction when it sends, but the data is to short. I played with various parameters, but never got more or better data. Maybe this could also cause this http://www.ebay.de/itm/251673693904 device I bought is broken. Its detected and works in Linux and Windows but I never found any channel with the Windows software it provides.

Got someone TFA 303125 working with rtl_433 or can someone recommand an nice looking alternatives (the most sensor are looking like crap)?

merbanan, 2015-02-21:
Right now most people are running on 434MHz. I had equipment that operated on 868MHz but I never got reliable readings from it. I suspect that better demodulation code and a higher sampling rate is needed to get it working properly.

ygator, 2015-02-26:
This looks interesting, but I have not looked into it much.
http://wirelesstag.net/

lanwin, 2015-04-15:
I can confirm that 303125 is compatible and works out of the box as lacrosse sensor. Its exactly the same sensor as 303180 but at 433mhz.

closed 2015-04-15 by lanwin

devZer0, 2019-01-27:
but 30.3125 is more expensive then 30.3180