merbanan/rtl_433 #103: Assistance decoding data for 433mhz transmitter?

mpmc, 2015-02-26:
Hey folks,

I have a [cheap Maplin weather station](http://www.maplin.co.uk/p/maplin-wireless-desktop-weather-station-with-barometer-a01qx) I've set it up and I'm able to receive data from it using rtl_433.

The problem is actually decoding this data and I can't for the life of me figure it out! I've looked online and nothing seems to match the output from the one I have, I've had a look inside the sensor and can't even see a model number!

I've uploaded a large sample [here](https://gist.githubusercontent.com/mpmc/4454ea982ecacd8e36c0/raw/gistfile1.txt)

Any help would be appreciated :)

mpmc, 2015-03-01:
I've been running rtl_433 for a good few hours (trying to work out the data). as well as recording the temps (in c) and humidity percentages.

```
5.6c 77%    =  000001011000001001001001011101010000001101000000
5.9c 77%    =  000001011000001111001001011101010000001010000000
6.1c 77%    =  000001011000000001001101011101010000001010000000
9.5c 81%    =  000001011000001111011001010001010100001000000000
9.6c 81%    =  000001011000000000011101010001010100001110000000
10.1c 81%   =  000001011000001001011101010001010100001101000000
9.7c 81%    =  000001011000000011011101010001010100000100000000
9.8c 81%    =  000001011000000101011101010001010100000011000000
9.9c 81%    =  000001011000000111011101010001010100001111000000
10.1c 81%   =  000001011000001001011101010001010100001101000000
10.2c 82%   =  000001011000001100011101010010010100001111000000
10.3c 82%   =  000001011000001101011101010010010100001001000000
```
I suspect I'm either receiving two sensors (unlikely) or it's sending other data (Air pressure?) along with temp and humidity.

mpmc, 2015-03-02:
I've opened it up again and looked a bit harder (Read: Got someone else with better eyesight) and managed to find the teeny tiny model number this time. __REV01 ST608B__ the board also has __S014HO__ on it.

This is the result when switching channels:-

```
3

00000010 01001000 00011101 10011000 10000010 01000000 

2

00000010 01000100 00011101 10011000 10000001 00000000

1

00000010 01000000 10011101 10100000 10000000 01000000 

```

merbanan, 2015-04-01:
Can you post the hex data from the weather station ?

mpmc, 2015-04-01:
I'm guessing you mean the hex output from rtl_433? (I'm a bit thick & should have included it anyway, sorry!)
```
[00] {0} 00 : 00000000
[01] {0} 00 : 00000000
[02] {42} 02 42 5d 4d 01 c0 : 00000010 01000010 01011101 01001101 00000001 11000000
[03] {0} 00 : 00000000
[04] {42} 02 42 5d 4d 01 c0 : 00000010 01000010 01011101 01001101 00000001 11000000
[05] {0} 00 : 00000000
[06] {42} 02 42 5d 4d 01 c0 : 00000010 01000010 01011101 01001101 00000001 11000000
[07] {0} 00 : 00000000
[08] {42} 02 42 5d 4d 01 c0 : 00000010 01000010 01011101 01001101 00000001 11000000
[09] {0} 00 : 00000000
[10] {42} 02 42 5d 4d 01 c0 : 00000010 01000010 01011101 01001101 00000001 11000000
[11] {0} 00 : 00000000
[12] {42} 02 42 5d 4d 01 c0 : 00000010 01000010 01011101 01001101 00000001 11000000
[13] {0} 00 : 00000000

= 10.1 67%
```
```
[00] {0} 00 : 00000000
[01] {0} 00 : 00000000
[02] {42} 02 42 5d 51 03 c0 : 00000010 01000010 01011101 01010001 00000011 11000000
[03] {0} 00 : 00000000
[04] {42} 02 42 5d 51 03 c0 : 00000010 01000010 01011101 01010001 00000011 11000000
[05] {0} 00 : 00000000
[06] {42} 02 42 5d 51 03 c0 : 00000010 01000010 01011101 01010001 00000011 11000000
[07] {0} 00 : 00000000
[08] {42} 02 42 5d 51 03 c0 : 00000010 01000010 01011101 01010001 00000011 11000000
[09] {0} 00 : 00000000
[10] {42} 02 42 5d 51 03 c0 : 00000010 01000010 01011101 01010001 00000011 11000000
[11] {0} 00 : 00000000
[12] {42} 02 42 5d 51 03 c0 : 00000010 01000010 01011101 01010001 00000011 11000000
[13] {0} 00 : 00000000

= 10.1 68%
```

Apologies for the limited number of readings I'll try to get more as the temp goes down! Unfortunately I'm unable to go out & get the sensor myself (so I can put it in the fridge or something) due to limited mobility.

Any help understanding this thing is much appreciated!

enavarro222, 2015-12-10:
I test some sample from here http://1drv.ms/1JQqLjG

    $ rtl_433 -r tmp/gfiles/gfile004.data  2>/dev/null
    Esperanza EWS: TemperatureF=43.6 TemperatureC=6.4 Humidity=67 Device_id=5 Channel=1 

This issue might be closed :)

merbanan, 2015-12-11:
Ok, closing. Thanks for verifying.

closed 2015-12-11 by merbanan

mpmc, 2015-12-11:
@enavarro222 Awesome! I totally forgot about this, thanks for the reminder!

@merbanan Haha it totally slipped my mind that as well as working on this you also write many of the other software that I use, you're everywhere! :+1:

merbanan, 2015-12-11:
@mpmc Thanks. I do my best.

mpmc, 2015-12-11:
@enavarro222 @merbanan 

Looks like this doesn't work correctly for mine still :(
```
Esperanza EWS: TemperatureF=40.4 TemperatureC=4.7 Humidity=71 Device_id=2 Channel=1
Fine Offset Electronics, WH2:
ID          = 0x1D
temperature = 6.8 C
humidity    = 255 %
```
As you can see it's nowhere near whats being reported by the screen..

[image]

It picks up my other sensor fine (that's in the shed).