merbanan/rtl_433 #105: support for WT450H

fojtp, 2015-03-06:
please, can you add support for WT450H?
more info:
http://ala-paavola.fi/jaakko/doku.php?id=wt450h
https://netcompsys.wordpress.com/2011/11/20/

thank you

helgew, 2015-03-06:
The implementation looks fairly straightforward but one would need some samples for testing. Can you please submit a couple to https://github.com/merbanan/rtl_433_tests <https://github.com/merbanan/rtl_433_tests> with a description and a listing of what the readout should be from the sensor?

fojtp, 2015-03-11:
submited

foldynl, 2015-09-27:
Hello,

I would like to ask you if there is any progress to support this type of device in rtl_433. I have a similar device (WT 260H) with same RF protocol.

foldynl, 2015-09-30:
The initial implementation of RF protocol for device like WT450 is available.
https://github.com/foldynl/rtl_433/tree/WT450
Test signals from testing signal repository look good.
The implementation was tested with WT260H sensor in real environment.

Could anybody with a real WT450 sensor test it?

milahti, 2015-10-01:
Excellent! 
I recently purchased a DVB-T for this purpose and tested with some RTL_433 version from which the binary output did not make any sense. I thought there would be either something wrong on my DVB-T dongle or the lack of support is causing the strange values. Luckily I found this thread and started to follow. Now I downloaded the code and out of the box RTL_433 started to output really good looking results:
2015-10-01 21:35:13 WT450 sensor: House Code 1, Channel 1, Temperature 10.13 C, Humidity 0 %
2015-10-01 21:35:13 WT450 sensor: House Code 1, Channel 1, Temperature 10.15 C, Humidity 0 %
2015-10-01 21:35:14 WT450 sensor: House Code 1, Channel 1, Temperature 10.16 C, Humidity 0 %
2015-10-01 21:35:47 WT450 sensor: House Code 1, Channel 2, Temperature 23.04 C, Humidity 58 %
2015-10-01 21:35:47 WT450 sensor: House Code 1, Channel 2, Temperature 23.06 C, Humidity 58 %
2015-10-01 21:35:47 WT450 sensor: House Code 1, Channel 2, Temperature 23.09 C, Humidity 58 %

I have two WT450H sensors (with humidity) and one WT450. One WT450H is actually outside and seems to have signal range issues. After opening a door I was able to get results from that as well:
2015-10-01 21:39:10 WT450 sensor: House Code 1, Channel 0, Temperature 11.18 C, Humidity 52 %
2015-10-01 21:39:10 WT450 sensor: House Code 1, Channel 0, Temperature 11.21 C, Humidity 52 %

Temperature and humidity values seem realistic. I have to later compare them in more detail. Channel numbering of the WT450(H) screen goes from 1 to 3, here from 0 to 2.

merbanan, 2015-10-01:
Try add -l 2000 to the command line.

foldynl, 2015-10-01:
Thanks...It sounds good. Is the sensor able to send any information about the state of battery (low battery) ? Please, could you try to catch the message with low battery information (rtl_433 -D or better -t) and send me the output?

milahti, 2015-10-02:
I had to find some old batteries to test with... now I'm able to see low battery indicator on my Channel 2 device but at least on the human readable results there were no indications about low battery. Sample with rtl_433 -D:

Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
AcuRite Rain Gauge Total Rain is 1409.5mm
Raw Message: 34 7b 03 00 00
pwm_d_decode(): Acurite 896 Rain Gauge 
bitbuffer:: Number of rows: 1 
[00] {25} 34 7b 03 00 : 00110100 01111011 00000011 0
2015-10-02 23:45:18 WT450 sensor: House Code 1, Channel 1, Temperature 21.56 C, Humidity 0 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 70 04 77 10 : 11000001 01110000 00000100 01110111 0001
2015-10-02 23:45:18 WT450 sensor: House Code 1, Channel 1, Temperature 21.58 C, Humidity 0 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 70 04 77 40 : 11000001 01110000 00000100 01110111 0100
AcuRite Rain Gauge Total Rain is 1408.5mm
Raw Message: 34 7b 01 00 00
pwm_d_decode(): Acurite 896 Rain Gauge 
bitbuffer:: Number of rows: 1 
[00] {26} 34 7b 01 00 : 00110100 01111011 00000001 00
2015-10-02 23:45:19 WT450 sensor: House Code 1, Channel 1, Temperature 21.61 C, Humidity 0 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 70 04 77 b0 : 11000001 01110000 00000100 01110111 1011
AcuRite Rain Gauge Total Rain is 325.5mm
Raw Message: 34 c2 8b 00 00
pwm_d_decode(): Acurite 896 Rain Gauge 
bitbuffer:: Number of rows: 1 
[00] {26} 34 c2 8b 00 : 00110100 11000010 10001011 00
2015-10-02 23:45:46 WT450 sensor: House Code 1, Channel 0, Temperature 21.33 C, Humidity 30 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 31 e4 74 30 : 11000001 00110001 11100100 01110100 0011
2015-10-02 23:45:46 WT450 sensor: House Code 1, Channel 0, Temperature 21.35 C, Humidity 30 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 31 e4 74 60 : 11000001 00110001 11100100 01110100 0110
2015-10-02 23:45:46 WT450 sensor: House Code 1, Channel 0, Temperature 21.36 C, Humidity 30 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 31 e4 74 90 : 11000001 00110001 11100100 01110100 1001
2015-10-02 23:45:57 WT450 sensor: House Code 1, Channel 2, Temperature 21.41 C, Humidity 57 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 bb 94 75 20 : 11000001 10111011 10010100 01110101 0010
2015-10-02 23:45:57 WT450 sensor: House Code 1, Channel 2, Temperature 21.43 C, Humidity 57 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 bb 94 75 70 : 11000001 10111011 10010100 01110101 0111
2015-10-02 23:45:57 WT450 sensor: House Code 1, Channel 2, Temperature 21.44 C, Humidity 57 %
pulse_demod_clock_bits(): WT450 
bitbuffer:: Number of rows: 1 
[00] {36} c1 bb 94 75 80 : 11000001 10111011 10010100 01110101 1000
^CSignal caught, exiting!

I did recordings with rtl_433 -a -t as well, where should I send those?

BTW it seems that the rain gauge information comes also from my WT450 sensors. I tested by removing batteries from all my devices, but it seems that some part of WT450 sample matches with that device's protocol as well:
AcuRite Rain Gauge Total Rain is 1408.5mm

foldynl, 2015-10-04:
Great work. It seems that Bit 13 is battery indicator.
Please,  could you clone https://github.com/merbanan/rtl_433_tests and submit your samples to the cloned git repository?

I am not a RTL_433 expert therefore I am not able to say why WT450 shows also AcuRite Rain information. Sometimes I see a similar situation with my WT260H. I have to investigate the issue.
 
I have to fix a  channel numbering too.

Thanks a million for data.

foldynl, 2015-10-05:
Thanks for helping. The patch was accepted and WT450 support was merged into the master branch.

milahti, 2015-10-05:
Really big thanks for you for making this work!
If it's just for file sharing purpose, I uploaded my sample files to Google Docs, you can download them using this link:
https://drive.google.com/file/d/0B1Pib2KJb5s5QlhQcTVGaGJLZ1E/view?usp=sharing

merbanan, 2015-11-24:
Thanks for the samples, next time try to send a git pull request.

closed 2015-11-24 by merbanan