merbanan/rtl_433 #106: Humidity and temperature sensor Ea2 BL999

khseal, 2015-03-08:
Received data from the sensor. 
15.7 94
34 4b 90 5f f0
8.8 96
34 41 a0 3f 10
2.8 97
34 43 80 bf 90	
-2.6 97
34 4b 7f bf 60
-3.3 97
34 46 7f bf f0
-7.4 96
34 46 df 3f d0
-8 95	
34 40 df df 20 
-2.3 93
34 29 7f 9f 00
2.9 92
34 2b 80 1f e0
23.2 35 
34 01 70 fd 70
Positive temperature 4 and 5 tetrad only written backwards.
15.7 94 1001 1101 0101 1110
34 4b 90 5f f0
0010 0010 0100 1011 1001 0000 0101 1111 1111 0000 
Humidity 7 and 8 tetrads, but the dependence not understandable. Negative temperature in question.

khseal, 2015-03-09:
Negative temperatures are additional binary code. Temperature 4-6 tetrad only written backwards.
-0.7 1111 1001 20 0001 0100
34 49 ff 0d 40 : 0011 0100 0100 1001 1111 1111 0000 1101 0100 0000

-2.5 20 1110 0111 0001 0100
34 4e 7f 0d f0 : 0011 0100 0100 1110 0111 1111 0000 1101 1111 0000

-11.5 20 1000 1101 0001 0100
34 4b 1f 0d f0 : 0011 0100 0100 1011 0001 1111 0000 1101 1111 0000

khseal, 2015-03-09:
[image]

merbanan, 2015-04-01:
Can you elaborate what this issue is about?

khseal, 2015-04-01:
And what exactly clarify?
What kind of sensors?
This weather station sensors http://www.ea2world.com/ea2p.html
Something not understand on the algorithm?

merbanan, 2015-04-01:
Do you want someone to add support for this sensor to rtl_433 ?

khseal, 2015-04-01:
Perhaps this information will be useful. I could not find information on the Internet on the sensor.
I can not add support in rtl_433.

merbanan, 2015-04-01:
You can submit a signal sample and text describing the format to the rtl_433_tests repository. Then someone might pick that up and add a decoder for this sensor.

khseal, 2015-04-01:
Ok, I'll try to do it today.

khseal, 2015-12-12:
TFA-Twin-Plus = Ea2 BL999 sensor
add to support device list

sprilukin, 2015-12-13:
Sorry to post here, but i see that this issue is about using BL999 temperature humidity sensor.
I did not found any arduino library over the internet for this sensor so i wrote one by myself.
You could find it here: https://github.com/sprilukin/lib_BL999.
Feel free to rise issue or pull request 'coz i'm just a beginner in arduino programming.

khseal, 2015-12-14:
Thanks for the link. Your job is useful for me.