merbanan/rtl_433 #113: Flushing output

bonnefoi, 2015-04-12:
Could you flush the stdout output at the end of the information returned by a device ?
Thus, it could be used in an interactive shell command through pipe as in the following command : 

`sudo rtl_433 -R14 2> /dev/null | awk -W interactive '/ADDR|command/ { slave=$4; getline; master=$4; getline; print slave ":"  master "," $3; }'`

Serveurperso, 2015-05-01:
Hi, same need here (wanna make Cacti graphs of my weather sensors), this is very annoying but I go check the code.

Serveurperso, 2015-05-01:
I solved this issue by disabling the buffering for stdin and stderr :

(root|~/rtl_433/src) git diff
diff --git a/src/rtl_433.c b/src/rtl_433.c
index fc85f0d..48406ef 100644
--- a/src/rtl_433.c
+++ b/src/rtl_433.c
@@ -987,6 +987,9 @@ int main(int argc, char **argv) {
     char vendor[256], product[256], serial[256];
     int have_opt_R = 0;

+    setbuf(stdout, NULL);
+    setbuf(stderr, NULL);
+
     demod = malloc(sizeof (struct dm_state));
     memset(demod, 0, sizeof (struct dm_state));

merbanan, 2015-05-10:
Word around committed.

closed 2015-05-10 by merbanan