merbanan/rtl_433 #115: stdout to file does not work

andreaaizza, 2015-04-28:
Hi, 

I am running changest 1def756a32913da5fd419cf5c038434fe53703c0. 

When issuing "#rtl_433 2>&1 > output_file.dat &" output_file.dat is filled only with initial text (see below), whilst no decoded sensor data is written to file (e.g. Thermo Sensor THN132N, Channel 1, Battery: Ok, Rolling-code 0xDA, Temp: 14.4C  57.9F). When "#killall -TERM rtl_433" is issued and rtl_433 exits, then file is output_file.dat is filled in. How can I get data written to file immediately as it is decoded? 

Cheers,
Andrea

Serveurperso, 2015-05-01:
Hi, same need here (wanna make Cacti graphs of my weather sensors), this is very annoying but I go check the code.

Serveurperso, 2015-05-01:
I solved this issue by disabling the buffering for stdin and stderr :

(root|~/rtl_433/src) git diff
diff --git a/src/rtl_433.c b/src/rtl_433.c
index fc85f0d..48406ef 100644
--- a/src/rtl_433.c
+++ b/src/rtl_433.c
@@ -987,6 +987,9 @@ int main(int argc, char **argv) {
     char vendor[256], product[256], serial[256];
     int have_opt_R = 0;

+    setbuf(stdout, NULL);
+    setbuf(stderr, NULL);
+
     demod = malloc(sizeof (struct dm_state));
     memset(demod, 0, sizeof (struct dm_state));

merbanan, 2015-05-10:
Workaround committed.

closed 2015-05-10 by merbanan