merbanan/rtl_433 #117: Help to implement Sensor GT-WT-02 (Aldi, Globaltronics..)

ludwich66, 2015-05-01:
Hi, 
i need help to implement a "new" Sensor.
I am not a hacker, but i had tested it with an excel code and it calculate the data of the Sensor correct.
See my example.
Who can help me here?

Sorry for my bad english!

ludwich

outputfie
*** signal_start = 113970967, signal_end = 114212857
signal_len = 241890,  pulses = 240
Iteration 1. t: 135    min: 129 (239)    max: 141 (1)    delta 9
Iteration 2. t: 135    min: 129 (239)    max: 141 (1)    delta 0
Distance coding: Pulse length 135
Short distance: 519, long distance: 1028, packet distance: 2257
p_limit: 135
[00] {0} 00 : 00000000 
[01] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
[02] {1} 00 : 00000000 
[03] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
[04] {1} 00 : 00000000 
[05] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
[06] {1} 00 : 00000000 
[07] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
[08] {1} 00 : 00000000 
[09] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
[10] {1} 00 : 00000000 
[11] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
[12] {0} 00 : 00000000

/* 2 examp les
/* [01] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000
/* code, BatOK,not-man-send, Channel1, +23,7°C, 35%
/*
/* [01] {37} 34 8f 87 15 90 : 00110100 10001111 10000111 00010101 10010000
/* code, BatOK,not-man-send, Channel1,-12,1°C, 10%

/* SENSOR: GT-WT-02 (ALDI Globaltronics..)
/* TYP AAAAAAAABCDDEFFFFFFFFFFFFGGGGGGGxxxxx
/* BIT 0000000001111111111222222222233333333
/* BIT 1234567890123456789012345678901234567
/* 
/* TYPDescriptian
/* A = Rolling Device Code, Change after battarie
/* B = Batt 0=OK 1=LOW
/* C = Manual Send Button Pressed 0=not pressed 1=pressed
/* D = Chanal 00=CH1, 01=CH2, 11=CH3
/* E = Temp 0=positiv 1=negativ
/* F =  PositivTemp =12 Bit bin2dez Temp, 
/* F = negativ Temp = 4095+1- F (12Bit bin2dez) , Factor  Divid  F / 10 (1Dezimal)
/* G =  Humidity = 7 Bit bin2dez 00-99
/* x = unknown

ortylp, 2015-08-09:
Can you create a longer recording using
rtl_sdr -g 20 -f 433920000 -s 250000 filname.raw

and make it available somehow to me?

ludwich66, 2015-08-10:
There is a raw file with 22Mb
http://koreander.h062lg51ufymmrfv.myfritz.net/GT-WT-02.raw
Is it ok?
ludwich

ortylp, 2015-08-10:
The recording is too weak. Can you record with the antenna about one meter from the transmitter?

I do not promise anything. At the moment I am still analysing the code of rtl_433 to find the cause of unstable demodulation of my other sensors...

ludwich66, 2015-08-10:
i take a new sample tomorrow
ludwich

ortylp, 2015-08-10:
no need to hurry.
you can convert the file to a wav file and check, if the power level is strong enough.
sox -c 2 -L -r 250000 -b 8 -e unsigned-integer -t raw GT-WT-02.raw GT-WT-02.wav
audacity GT-WT-02.wav

ludwich66, 2015-08-16:
Hi ORTYLP,

i dont use sox and audacity on my mac.
it is possible to maka  a new file and place it on my hompage.
i have problems with github :-) i dont check the way to add pictures - its a mysterium :-)

What data in Moment needed to add my sensor to rtl_433

i have added at Moment:

- Picture Sensor Front
- Picture Sensor Back
- A detailed Protocoldecription (Testet with output in excel - works well)
- a gfile003.data <https://github.com/merbanan/rtl_433_tests/blob/master/tests/GT-WT-02/gfile003.data> (i dont know, is it OK?)

you analyze only the receiving quality at moment?

ludwich

PS: Languages -> German, broken-english, Bavarian

ortylp, 2015-08-16:
We need more test data and ideas to reverse engineer the unknown bits.

ludwich66, 2015-08-16:
There are more encryptet data from  GT-WT-02, but the Unknown bits are not relevant ( CRC??) IMHO.
Do you need more samples??

ludwich

ortylp, 2015-08-16:
I need as much data as you can provide.  I mean different valid bit sequences.  The easiest way to do this is to put the sensor into a fridge (let it go to -18°C) get it out and record data as it comes back to the room temperature. 

I do not need raw data.  Sequences like
[01] {37} 34 00 ed 47 60 : 00110100 00000000 11101101 01000111 01100000 
is the information we need.  As many different bit sequences as you can provide.

The CRC (if it is one) is the key to limit false positive rate, we need to find out what it is.
Did you analyse these bis sequences yourself, or you can provide some other sources?

ludwich66, 2015-08-16:
i made the dataset with your Fridge-Hint :-)
ok i sample more data next day. we take the last 6 bits :-)
do you have any ideas, how does other sensors do?
CRC, Parity, ??
ludwich

ortylp, 2015-08-16:
No need to hurry, I will get back to it probably next weekend, if I find time.
There are some "standard" methods for doing this, I'll try to figure out what it could be, or brute force it.

ortylp, 2015-08-16:
BTW: the data sample you provided was really strange, my algorithm for enhancing sensitivity went insane.  It looks like you have DC offset and strong IQ imbalance in the receiver... or really oversteered it.  A fine sample for making sure the algorithm does not fail under such conditions.

It seems one of my neighbours also bought this sensor, but I can receive it only when the algorithm is working well :)

magellannh, 2015-08-16:
I haven't been following this thread closely, so apologies in advance if this is off base.  

However, when I was trying to figure out the CRC algorithm used by an efergy electricity sensor, I found this online CRC calculator to be very helpful:
  http://www.lammertbies.nl/comm/info/crc-calculation.html

ortylp, 2015-08-16:
@magellannh thanks for the link.  As we are talking about cheap sensors I expect something trivial and maximum 8bit long, but these other algorithms might be worth a look.

ludwich66, 2015-08-16:
Thanks i try any combination next week 

ludwich

ortylp, 2015-08-16:
I see we need this CRC stuff ASAP.  I let it run directly from receiver and I see some reasonable reports and also rubbish :(  I see false positives once in a few minutes...

ludwich66, 2015-08-16:
now lets look:-)

rct, 2015-08-17:
There is an 8 bit crc calculator in src/util.c that @vestom committed around a month back.

It takes the polynomial representation as an 8 bit argument.   
* 0x31, is used by Dallas/Maxim 1-wire and a number of other devices including one of the fineoffset sensors.
* 0x07 is CRC-8 CCITT, which is also pretty common.

You could always try all 256 to see if it uses that CRC8 algorithm.   (I tried that for DSC, but non of them worked so it must be using a diffierent algorithm, or processing the bits differently.)

rct, 2015-08-17:
Two other on-line calculators that are a little more flexible:
* http://depa.usst.edu.cn/chenjq/www2/software/crc/CRC_Javascript/CRCcalculation.htm  
* https://ghsi.de/CRC/index.php

The last one shows the flip flops and code to implement the CRC.

ortylp, 2015-08-17:
@rct the crc8 function  was the first thing I tried, all 256 polynomials, none fitted.  Thanks for further hints.

rct, 2015-08-17:
@ortylp, ok, I'm having the same issue with the DSC sensors.  I suspect the bits may be run through the calculation differently on the actual devices.  I need to experiment with a bit shifting implementation that might more closely resemble hardware.

Also if you haven't already tried it make sure you try with / without any framing data that might be in the packet.

ludwich66, 2015-08-17:
there are more datas. i dont know at moment
- Parity e/o -> no
-CRC -> 6 Bit is very short…

look at line 87 and 88
HUM 110 = HH = out of range

ludwich

ludwich66, 2015-08-18:
Checksum is not crc :-)
Do It!!

bin2dez(Bit1;Bit2;Bit3;Bit4)+ #rolling code
bin2dez(Bit5;Bit6;Bit7;Bit8)+ #rolling code
bin2dez(Bit9;Bit10;Bit11;Bit12)+  # send, bat , ch
ignore Bit 13 (+/-)
bin2dez(Bit14;Bit15;Bit16;Bit17)+   #temp1
bin2dez(Bit18;Bit19;Bit16;Bit20)+  #temp2
bin2dez(Bit21;Bit22;Bit23;Bit24)+  #temp3
bin2dez(Bit25;Bit26;Bit27;Bit28)+ #hum1
bin2dez(Bit29;Bit30;Bit31;Bit13?) #hum2
bin2dez(Bit32;Bit33;Bit34;Bit35;Bit36;Bit37) #checksum

works with positive, negative values need a manipulation 
i try it later
ludwich

ludwich66, 2015-08-18:
is the sensor now working ?
i compiled rtl_433 and find the as device 25 - it does not work - only in rtl_433 -a Mode
ludwich

ortylp, 2015-08-18:
It should work.  I have just tested:

git clone git@github.com:merbanan/rtl_433.git
cmake .
make
./src/rtl_433 -r ~/src/rtl_433/tests/tests/GT-WT-02/gfile003.data 
Registering protocol "Silvercrest Remote Control"
Registering protocol "Rubicson Temperature Sensor"
Registering protocol "Prologue Temperature Sensor"
Registering protocol "Waveman Switch Transmitter"
Registering protocol "Steffen Switch Transmitter"
Registering protocol "ELV EM 1000"
Registering protocol "ELV WS 2000"
Registering protocol "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol "Acurite 5n1 Weather Station"
Registering protocol "Acurite 896 Rain Gauge"
Registering protocol "Acurite Temperature and Humidity Sensor"
Registering protocol "Oregon Scientific Weather Sensor"
Registering protocol "Mebus 433"
Registering protocol "Intertechno 433"
Registering protocol "KlikAanKlikUit Wireless Switch"
Registering protocol "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol "Cardin S466-TX2"
Registering protocol "Fine Offset Electronics, WH-2 Sensor"
Registering protocol "Nexus Temperature & Humidity Sensor"
Registering protocol "Ambient Weather Temperature Sensor"
Registering protocol "Calibeur RF-104 Sensor"
Registering protocol "X10 RF"
Registering protocol "DSC Security Contact"
Registering protocol "Brennstuhl RCS 2044"
Registering protocol "GT-WT-02 Sensor"
Test mode active. Reading samples from file: /home/ortyl/src/rtl_433/tests/tests/GT-WT-02/gfile003.data
2015-08-18 22:05:44 GT-WT-02 Sensor 34: battery OK, channel 0, button 0, temperature 22.2 C / 72.0 F, humidity 59%
2015-08-18 22:05:44 GT-WT-02 Sensor 34: battery OK, channel 0, button 0, temperature 22.2 C / 72.0 F, humidity 59%
2015-08-18 22:05:44 GT-WT-02 Sensor 34: battery OK, channel 0, button 0, temperature 22.2 C / 72.0 F, humidity 59%
2015-08-18 22:05:44 GT-WT-02 Sensor 34: battery OK, channel 0, button 0, temperature 22.2 C / 72.0 F, humidity 59%
2015-08-18 22:05:44 GT-WT-02 Sensor 34: battery OK, channel 0, button 0, temperature 22.2 C / 72.0 F, humidity 59%
2015-08-18 22:05:44 GT-WT-02 Sensor 34: battery OK, channel 0, button 0, temperature 22.2 C / 72.0 F, humidity 59%
Test mode file issued 6 packets

ortylp, 2015-08-18:
So with the file you provided it works...

It great, you figured out the checksum :D
I'll test the formula for checksum as soon as I get to it, at the moment I do not know, when I'll get free time slot for programming again.

ludwich66, 2015-08-18:
I have used the normal build process - i try it twice :-)

->The calculate checksum is ok

ludwich66, 2015-08-18:
Thanks you please inform me if you are fine with programming

ludwich

ludwich66, 2015-08-19:
Problems to open device 25?
Any hints?
Jurgens-MBP:build juergenfrey$ rtl_433 -d 25
Registering protocol "Silvercrest Remote Control"
Registering protocol "Rubicson Temperature Sensor"
Registering protocol "Prologue Temperature Sensor"
Registering protocol "Waveman Switch Transmitter"
Registering protocol "Steffen Switch Transmitter"
Registering protocol "ELV EM 1000"
Registering protocol "ELV WS 2000"
Registering protocol "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol "Acurite 5n1 Weather Station"
Registering protocol "Acurite 896 Rain Gauge"
Registering protocol "Acurite Temperature and Humidity Sensor"
Registering protocol "Oregon Scientific Weather Sensor"
Registering protocol "Mebus 433"
Registering protocol "Intertechno 433"
Registering protocol "KlikAanKlikUit Wireless Switch"
Registering protocol "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol "Cardin S466-TX2"
Registering protocol "Fine Offset Electronics, WH-2 Sensor"
Registering protocol "Nexus Temperature & Humidity Sensor"
Registering protocol "Ambient Weather Temperature Sensor"
Registering protocol "Calibeur RF-104 Sensor"
Registering protocol "X10 RF"
Registering protocol "DSC Security Contact"
Registering protocol "Brennstuhl RCS 2044"
Registering protocol "GT-WT-02 Sensor"
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Using device 25: 
Failed to open rtlsdr device #25.

ortylp, 2015-08-19:
Use -R not -d

ludwich66, 2015-08-19:
Grrrr. it does not work for me.
i made  a clone from git, and build it.
It work only mode rtl_433 -a ??

Response of Buildprozess
Jurgens-MBP:~ juergenfrey$ git clone git@github.com:merbanan/rtl_433.git
Cloning into 'rtl_433'...
remote: Counting objects: 2586, done.
remote: Total 2586 (delta 0), reused 0 (delta 0), pack-reused 2586
Receiving objects: 100% (2586/2586), 732.87 KiB | 816.00 KiB/s, done.
Resolving deltas: 100% (1835/1835), done.
Checking connectivity... done.
Jurgens-MBP:~ juergenfrey$ rtl_433
-bash: rtl_433: command not found
Jurgens-MBP:~ juergenfrey$ cd rtl_433
Jurgens-MBP:rtl_433 juergenfrey$ mkdir build
Jurgens-MBP:rtl_433 juergenfrey$ cd build
Jurgens-MBP:build juergenfrey$ cmake ../
-- The C compiler identification is AppleClang 6.1.0.6020053
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.28") 
-- checking for module 'librtlsdr'
--   found librtlsdr, version 
-- Found librtlsdr: /opt/local/include, /opt/local/lib/librtlsdr.dylib
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/juergenfrey/rtl_433/build
Jurgens-MBP:build juergenfrey$ make
Scanning dependencies of target rtl_433
[  3%] Building C object src/CMakeFiles/rtl_433.dir/rtl_433.c.o
[  7%] Building C object src/CMakeFiles/rtl_433.dir/baseband.c.o
[ 10%] Building C object src/CMakeFiles/rtl_433.dir/bitbuffer.c.o
[ 14%] Building C object src/CMakeFiles/rtl_433.dir/pulse_demod.c.o
/Users/juergenfrey/rtl_433/src/pulse_demod.c:32:8: warning: taking the absolute
      value of unsigned type 'unsigned int' has no effect [-Wabsolute-value]
                if ((abs(pulses->pulse[n] - device->short_limit) < TOLER...
                     ^
/Users/juergenfrey/rtl_433/src/pulse_demod.c:32:8: note: remove the call to
      'abs' since unsigned values cannot be negative
                if ((abs(pulses->pulse[n] - device->short_limit) < TOLER...
                     ^~~
/Users/juergenfrey/rtl_433/src/pulse_demod.c:33:9: warning: taking the absolute
      value of unsigned type 'unsigned int' has no effect [-Wabsolute-value]
                 && ((abs(pulses->pulse[n] + pulses->gap[n] - periods * ...
                      ^
/Users/juergenfrey/rtl_433/src/pulse_demod.c:33:9: note: remove the call to
      'abs' since unsigned values cannot be negative
                 && ((abs(pulses->pulse[n] + pulses->gap[n] - periods * ...
                      ^~~
2 warnings generated.
[ 17%] Building C object src/CMakeFiles/rtl_433.dir/pulse_detect.c.o
[ 21%] Building C object src/CMakeFiles/rtl_433.dir/util.c.o
[ 25%] Building C object src/CMakeFiles/rtl_433.dir/devices/silvercrest.c.o
[ 28%] Building C object src/CMakeFiles/rtl_433.dir/devices/rubicson.c.o
[ 32%] Building C object src/CMakeFiles/rtl_433.dir/devices/prologue.c.o
[ 35%] Building C object src/CMakeFiles/rtl_433.dir/devices/waveman.c.o
[ 39%] Building C object src/CMakeFiles/rtl_433.dir/devices/steffen.c.o
[ 42%] Building C object src/CMakeFiles/rtl_433.dir/devices/elv.c.o
[ 46%] Building C object src/CMakeFiles/rtl_433.dir/devices/lacrosse.c.o
[ 50%] Building C object src/CMakeFiles/rtl_433.dir/devices/acurite.c.o
[ 53%] Building C object src/CMakeFiles/rtl_433.dir/devices/oregon_scientific.c.o
[ 57%] Building C object src/CMakeFiles/rtl_433.dir/devices/mebus.c.o
[ 60%] Building C object src/CMakeFiles/rtl_433.dir/devices/intertechno.c.o
[ 64%] Building C object src/CMakeFiles/rtl_433.dir/devices/alecto.c.o
[ 67%] Building C object src/CMakeFiles/rtl_433.dir/devices/newkaku.c.o
[ 71%] Building C object src/CMakeFiles/rtl_433.dir/devices/cardin.c.o
[ 75%] Building C object src/CMakeFiles/rtl_433.dir/devices/fineoffset.c.o
[ 78%] Building C object src/CMakeFiles/rtl_433.dir/devices/nexus.c.o
[ 82%] Building C object src/CMakeFiles/rtl_433.dir/devices/ambient_weather.c.o
[ 85%] Building C object src/CMakeFiles/rtl_433.dir/devices/calibeur.c.o
[ 89%] Building C object src/CMakeFiles/rtl_433.dir/devices/x10_rf.c.o
[ 92%] Building C object src/CMakeFiles/rtl_433.dir/devices/dsc.c.o
[ 96%] Building C object src/CMakeFiles/rtl_433.dir/devices/brennstuhl_rcs_2044.c.o
[100%] Building C object src/CMakeFiles/rtl_433.dir/devices/gt_wt_02.c.o
Linking C executable rtl_433
[100%] Built target rtl_433
Jurgens-MBP:build juergenfrey$ sudo make install
Password:
[100%] Built target rtl_433
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/include/rtl_433.h
-- Installing: /usr/local/include/rtl_433_devices.h
-- Installing: /usr/local/bin/rtl_433
Jurgens-MBP:build juergenfrey$ rtl_433 -R 25
Registering protocol "GT-WT-02 Sensor"
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Sample rate decimation set to 0. 250000->250000
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
^CSignal caught, exiting!

ortylp, 2015-08-19:
A I have already said, the example you have sent works.  Could you please record some more examples for us to analyse?

If it works with -a it must work also with demodulator.  Can you please provide full output of a "rtl_433 -a" run?

ludwich66, 2015-08-19:
There is a full -a dump (Thanks for your help)
Jurgens-MBP:build juergenfrey$ rtl_433 -a
Registering protocol "Silvercrest Remote Control"
Registering protocol "Rubicson Temperature Sensor"
Registering protocol "Prologue Temperature Sensor"
Registering protocol "Waveman Switch Transmitter"
Registering protocol "Steffen Switch Transmitter"
Registering protocol "ELV EM 1000"
Registering protocol "ELV WS 2000"
Registering protocol "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol "Acurite 5n1 Weather Station"
Registering protocol "Acurite 896 Rain Gauge"
Registering protocol "Acurite Temperature and Humidity Sensor"
Registering protocol "Oregon Scientific Weather Sensor"
Registering protocol "Mebus 433"
Registering protocol "Intertechno 433"
Registering protocol "KlikAanKlikUit Wireless Switch"
Registering protocol "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol "Cardin S466-TX2"
Registering protocol "Fine Offset Electronics, WH-2 Sensor"
Registering protocol "Nexus Temperature & Humidity Sensor"
Registering protocol "Ambient Weather Temperature Sensor"
Registering protocol "Calibeur RF-104 Sensor"
Registering protocol "X10 RF"
Registering protocol "DSC Security Contact"
Registering protocol "Brennstuhl RCS 2044"
Registering protocol "GT-WT-02 Sensor"
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Sample rate decimation set to 0. 250000->250000
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
*** signal_start = 51590, signal_end = 292785
signal_len = 241195,  pulses = 240
Iteration 1. t: 130    min: 129 (15)    max: 131 (225)    delta 4
Iteration 2. t: 130    min: 129 (2)    max: 131 (238)    delta 0
Distance coding: Pulse length 130

Short distance: 514, long distance: 1023, packet distance: 2253

p_limit: 130
bitbuffer:: Number of rows: 13 
[00] {0} : 
[01] {37} 34 80 91 dd 90 : 00110100 10000000 10010001 11011101 10010
[02] {1} 00 : 0
[03] {37} 34 80 91 dd 90 : 00110100 10000000 10010001 11011101 10010
[04] {1} 00 : 0
[05] {37} 34 80 91 dd 90 : 00110100 10000000 10010001 11011101 10010
[06] {1} 00 : 0
[07] {37} 34 80 91 dd 90 : 00110100 10000000 10010001 11011101 10010
[08] {1} 00 : 0
[09] {37} 34 80 91 dd 90 : 00110100 10000000 10010001 11011101 10010
[10] {1} 00 : 0
[11] {37} 34 80 91 dd 90 : 00110100 10000000 10010001 11011101 10010
[12] {0} :

ludwich66, 2015-08-19:
Oh is see raw 00  is missing
Raw 01 is cuted to 37 Bits, the are 5 bytes before, the last 3 ones are 0
??

ortylp, 2015-08-19:
With the file you provided I get the following:

Test mode active. Reading samples from file: /home/ortyl/Downloads/gfile003.data
*** signal_start = 16983, signal_end = 389398
signal_len = 372415,  pulses = 348
Iteration 1. t: 100    min: 2 (108)    max: 199 (240)    delta 81
Iteration 2. t: 100    min: 2 (108)    max: 199 (240)    delta 0
Pulse coding: Short pulse length 2 - Long pulse length 199

Short distance: 531, long distance: 1540, packet distance: 3376

p_limit: 100
bitbuffer:: Number of rows: 13 
[00] {1} 80 : 1
[01] {59} 6b 7f 79 aa ad ae ba a0 
[02] {3} 60 : 011
[03] {53} ba bf 7d 75 9f 75 78 
[04] {2} c0 : 11
[05] {54} b5 b7 76 77 6f ba f4 
[06] {2} c0 : 11
[07] {55} f6 f6 dd cb 5f 56 da 
[08] {4} 50 : 0101
[09] {55} 6e b5 ee ad 6e fa de 
[10] {3} a0 : 101
[11] {56} 55 dd eb fb 6b 57 75 
[12] {1} 80 : 1
Test mode file issued 6 packets

__I need more samples from you.__

The distances:
"Short distance: 531, long distance: 1540, packet distance: 3376"
Do not fit with yours at all:
"Short distance: 514, long distance: 1023, packet distance: 2253"

ludwich66, 2015-08-19:
-> The problem with missing even rows, and the shorter data row is located first after  new compilin rt._433. i don't know ...
*** signal_start = 9520953, signal_end = 9765049
signal_len = 244096,  pulses = 240
Iteration 1. t: 136    min: 130 (239)    max: 142 (1)    delta 4
Iteration 2. t: 136    min: 130 (239)    max: 142 (1)    delta 0
Distance coding: Pulse length 136

Short distance: 515, long distance: 1023, packet distance: 2247

p_limit: 136
bitbuffer:: Number of rows: 13 
[00] {0} : 
[01] {37} a0 00 e7 77 60 : 10100000 00000000 11100111 01110111 01100
[02] {1} 00 : 0
[03] {37} a0 00 e7 77 60 : 10100000 00000000 11100111 01110111 01100
[04] {1} 00 : 0
[05] {37} a0 00 e7 77 60 : 10100000 00000000 11100111 01110111 01100
[06] {1} 00 : 0
[07] {37} a0 00 e7 77 60 : 10100000 00000000 11100111 01110111 01100
[08] {1} 00 : 0
[09] {37} a0 00 e7 77 60 : 10100000 00000000 11100111 01110111 01100
[10] {1} 00 : 0
[11] {37} a0 00 e7 77 60 : 10100000 00000000 11100111 01110111 01100
[12] {0} :

ortylp, 2015-08-19:
Please record data with rtl_433 -a -t and make the gfile???.data files available to us.

rct, 2015-08-19:
Two things:
1. Try -D (with/without -a) to see the analysis @vestom's code generates.

2. A recent commit changed the filtering and default detection level..  That might change the width of the pulses from what was seen previously.

ludwich66, 2015-08-19:
http://koreander.h062lg51ufymmrfv.myfritz.net/gfile01.data
http://koreander.h062lg51ufymmrfv.myfritz.net/gfile02.data
last activity for this day - bye 
ludwich

ortylp, 2015-08-19:
These files have different timing!
In the recent pull request I have modified gap length threshold to correspond to your files above.

ludwich66, 2015-08-20:
Hi ortylp,
it works :-) Thanks
- There is not a value "Checksum OK", what do you do if there are wrong dates? Skip this dataset without info?
- do you need more?
ludwich

Jurgens-MBP:build juergenfrey$ rtl_433
Registering protocol "Silvercrest Remote Control"
Registering protocol "Rubicson Temperature Sensor"
Registering protocol "Prologue Temperature Sensor"
Registering protocol "Waveman Switch Transmitter"
Registering protocol "Steffen Switch Transmitter"
Registering protocol "ELV EM 1000"
Registering protocol "ELV WS 2000"
Registering protocol "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol "Acurite 5n1 Weather Station"
Registering protocol "Acurite 896 Rain Gauge"
Registering protocol "Acurite Temperature and Humidity Sensor"
Registering protocol "Oregon Scientific Weather Sensor"
Registering protocol "Mebus 433"
Registering protocol "Intertechno 433"
Registering protocol "KlikAanKlikUit Wireless Switch"
Registering protocol "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol "Cardin S466-TX2"
Registering protocol "Fine Offset Electronics, WH-2 Sensor"
Registering protocol "Nexus Temperature & Humidity Sensor"
Registering protocol "Ambient Weather Temperature Sensor"
Registering protocol "Calibeur RF-104 Sensor"
Registering protocol "X10 RF"
Registering protocol "DSC Security Contact"
Registering protocol "Brennstuhl RCS 2044"
Registering protocol "GT-WT-02 Sensor"
Registering protocol "Danfoss CFR Thermostat"
Registering protocol "Energy Count 3000 (868.3 MHz)"
No supported devices found.
Jurgens-MBP:build juergenfrey$ rtl_433
Registering protocol "Silvercrest Remote Control"
Registering protocol "Rubicson Temperature Sensor"
Registering protocol "Prologue Temperature Sensor"
Registering protocol "Waveman Switch Transmitter"
Registering protocol "Steffen Switch Transmitter"
Registering protocol "ELV EM 1000"
Registering protocol "ELV WS 2000"
Registering protocol "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol "Acurite 5n1 Weather Station"
Registering protocol "Acurite 896 Rain Gauge"
Registering protocol "Acurite Temperature and Humidity Sensor"
Registering protocol "Oregon Scientific Weather Sensor"
Registering protocol "Mebus 433"
Registering protocol "Intertechno 433"
Registering protocol "KlikAanKlikUit Wireless Switch"
Registering protocol "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol "Cardin S466-TX2"
Registering protocol "Fine Offset Electronics, WH-2 Sensor"
Registering protocol "Nexus Temperature & Humidity Sensor"
Registering protocol "Ambient Weather Temperature Sensor"
Registering protocol "Calibeur RF-104 Sensor"
Registering protocol "X10 RF"
Registering protocol "DSC Security Contact"
Registering protocol "Brennstuhl RCS 2044"
Registering protocol "GT-WT-02 Sensor"
Registering protocol "Danfoss CFR Thermostat"
Registering protocol "Energy Count 3000 (868.3 MHz)"
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
2015-08-20 16:45:27 GT-WT-02 Sensor a0: battery OK, channel 0, button 0, temperature 21.9 C / 71.4 F, humidity 49%
2015-08-20 16:46:16 GT-WT-02 Sensor a0: battery OK, channel 0, button 0, temperature 21.9 C / 71.4 F, humidity 49%

ortylp, 2015-08-20:
If checksum is wrong, no output is produced.  It is the only way to avoid false positives.  I still need to add a few lines to exclude the message, where all bits are zero (which is actually a valid one).  This one comes relatively often and is very unlikely to be actual transmission from the sensor.

We need new data samples in the test repository.  Please use the trick with the fridge, record data with rtl_433 -a -t, as the temperature rises "change" battery ( == change the sensor ID), try to provoke all possible states of the sensor, put weak batteries (battery status). temperature over 51.2°C would also be interesting.

There is no such thing as too much test data.   After this recording is done please choose about 20-30 "interesting" files and push it to github repository.

ludwich66, 2015-08-20:
Crazy i do receive the sensor out of the fridge, its complete Metal only the rubber lips :-)
over 50 degrees, tomorrow i make a electronic-cake :-)
Why do you mean the critical temp is over 51.2 , my tipp 102,3 ;-)
ludwich

ortylp, 2015-08-20:
Then make sure you do not melt the device!

ludwich66, 2015-08-20:
Displayrange for humidity is LL=10%, HH=110%

ludwich66, 2015-08-20:
Channel numbering on device is 1, 2, 3 bit numbering 00, 01,10

ortylp, 2015-08-21:
Does it mean, that humidity is 10% off?

ludwich66, 2015-08-21:
RTL_433 now run on a ODROID C1 (quad core A5 1,5GHZ, 1GRAM) CPU usage are 6%

ludwich66, 2015-08-21:
theat mean
the working range is 20-90 % 
if „LL“ in display view it sends 10 %
if „HH“ in display view it sends 110%

i have no exact datarows at moment, any idea  how to simulate a 20 to 90 Percent range with a slow rate of 50 seconds?
tomorrow i take a long shower :-)

It works from 7 AM to 23 PM fine on the Odroid C1

-> do you modify the chanel assigment ?

00 = 1 (now0)
01 = 2 (now1)
10 = 3 (now2)

bye ludwich

ludwich66, 2015-08-22:
New Temperatur and Humidity Row.
Temperature up to 56,3 C (corresponding with Sensordisplay)
Humidity if Hum < 20 % display change to "LL", data change to "10%"
Humidity > 90% isn't tested at moment
2015-08-22 10:58:51 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 37.7 C / 99.9 F, humidity 10%
2015-08-22 10:59:41 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 40.4 C / 104.7 F, humidity 10%
2015-08-22 11:00:32 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 42.9 C / 109.2 F, humidity 10%
2015-08-22 11:01:21 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 45.0 C / 113.0 F, humidity 10%
2015-08-22 11:02:11 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 42.1 C / 107.8 F, humidity 10%
2015-08-22 11:03:01 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 48.7 C / 119.7 F, humidity 10%
2015-08-22 11:03:51 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 49.4 C / 120.9 F, humidity 10%
2015-08-22 11:04:41 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 55.3 C / 131.5 F, humidity 10%
2015-08-22 11:05:31 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 56.3 C / 133.3 F, humidity 10%
2015-08-22 11:06:21 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 49.5 C / 121.1 F, humidity 10%
2015-08-22 11:07:11 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 47.7 C / 117.9 F, humidity 10%
2015-08-22 11:08:01 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 46.1 C / 115.0 F, humidity 10%
2015-08-22 11:08:51 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 44.6 C / 112.3 F, humidity 10%
2015-08-22 11:09:41 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 43.2 C / 109.8 F, humidity 10%
2015-08-22 11:10:31 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 41.9 C / 107.4 F, humidity 10%
2015-08-22 11:11:21 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 40.6 C / 105.1 F, humidity 10%
2015-08-22 11:12:11 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 39.3 C / 102.7 F, humidity 10%
2015-08-22 11:13:01 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 38.2 C / 100.8 F, humidity 10%
2015-08-22 11:13:51 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 37.1 C / 98.8 F, humidity 10%
2015-08-22 11:14:41 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 35.9 C / 96.6 F, humidity 10%
2015-08-22 11:15:31 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 34.8 C / 94.6 F, humidity 10%
-->End of "LL"
2015-08-22 11:16:21 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 33.9 C / 93.0 F, humidity 21%
2015-08-22 11:17:11 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 33.1 C / 91.6 F, humidity 22%

ludwich66, 2015-08-22:
Humidity Row
There are a interesting couple of values on the upper End
36% on Display to, rising
110% on Display "HH"
36% on Display to, falling
i don't see  this on the lower range End, ?but this  was a faster test.
-->  now i clean the wet bathroom first :-)

2015-08-22 11:29:41 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 26.3 C / 79.3 F, humidity 63%
2015-08-22 11:31:21 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 25.9 C / 78.6 F, humidity 71%
2015-08-22 11:32:11 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 25.7 C / 78.3 F, humidity 79%
2015-08-22 11:33:01 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 25.6 C / 78.1 F, humidity 83%
2015-08-22 11:34:41 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 25.5 C / 77.9 F, humidity 36%
2015-08-22 11:35:31 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 25.6 C / 78.1 F, humidity 110%
2015-08-22 11:37:11 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 25.6 C / 78.1 F, humidity 36%
2015-08-22 11:41:21 GT-WT-02 Sensor a9: battery OK, channel 0, button 0, temperature 24.9 C / 76.8 F, humidity 75%

ortylp, 2015-08-25:
@ludwich66 I'll update the decoder as soon as I get back to coding, but I cannot promise when...

ludwich66, 2015-08-26:
Its OK :-), i do more tests (falling Humidiy) this week.
It works fine for me now ;-)
ludwich

ortylp, 2015-08-31:
@ludwich66 if it works now as expected, then please close this issue.

ludwich66, 2015-09-01:
Hi ortylp,
cannel assigment works
the humidity begin end level (LL/HH) works
Thank you :-)

closed 2015-09-01 by ludwich66

ludwich66, 2015-09-05:
Hi Ortylp,

is it possible to add a space between the humidity value and % sign.
with the LL and HH logic it a string, thats now a problem with charts.
Any idea??

ludwich