merbanan/rtl_433 #119: rtl_433 -S gives me segmentation fault

jcarduino, 2015-06-24:
Playing with the parameters when starting rtl_433 -S causes a segmentation error right at the startup.

Any thoughts?

jcarduino, 2015-09-17:
still active and current with build on today...

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000.
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in sync mode...
Segmentation fault (core dumped)

LuccoJ, 2015-10-01:
Yep, this happens to me as well.

robho, 2015-10-26:
sync mode writes samples to an output file. If no output file is specified rtl_433 will crash.

jcarduino, 2015-10-26:
Clear. Can this be updated in documentation? This can be closed afaik. 

Mvg Johan Corstjens

Verstuurd via mobile

merbanan, 2018-12-03:
Option -S is no longer a valid option and thus no longer crash. Closing issue.

closed 2018-12-03 by merbanan