merbanan/rtl_433 #130: building with visual studio

vsboost, 2015-07-12:
Hi there,
Any instructions for building in windows using visual studio ?

pclov3r, 2015-08-03:
I've built it just fine using dev-c++ 

http://sourceforge.net/projects/orwelldevcpp/

vsboost, 2015-08-03:
Thanks, got heaps of errors, even tried copying the rtl-sdr.h file that it was asking for and still got errors.  Are you able to give some specific instructions to a compiling newb ??

pclov3r, 2015-08-03:
You'll need to setup the Linker so it can link the librtlsdr dll 

Can you post your build log?

vsboost, 2015-08-04:
Had to cut it down, was too big.

Compiling single file...
--------
- Filename: C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl_433.c
- Compiler Name: TDM-GCC 4.9.2 64-bit Release

Processing C source file...
--------
- C Compiler: %BinDir0%\gcc.exe
- Command: gcc.exe "C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl_433.c" -o "C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl_433.exe"  -I"%CppIncludeDir0%" -I"%CppIncludeDir1%" -I"%CppIncludeDir2%" -L"%LibDir0%" -L"%LibDir1%" -static-libgcc
In file included from C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl_433.c:23:0:
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:1:1: error: expected identifier or '(' before '<' token
 <!DOCTYPE html PUBLIC "-\\W3C\\DTD XHTML 1.0 Transitional\\EN"
 ^

C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:3:13: warning: character constant too long for its type
 <html xmlns='http:\\www.w3.org\1999\xhtml' xml:lang='en' lang='en'>
             ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:3:53: warning: multi-character character constant [-Wmultichar]
 <html xmlns='http:\\www.w3.org\1999\xhtml' xml:lang='en' lang='en'>
                                                     ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:3:63: warning: multi-character character constant [-Wmultichar]
 <html xmlns='http:\\www.w3.org\1999\xhtml' xml:lang='en' lang='en'>
                                                               ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:6:12: warning: character constant too long for its type
 <meta name='generator' content='cgit v0.9.1'\>
            ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:6:32: warning: character constant too long for its type
 <meta name='generator' content='cgit v0.9.1'\>
                                ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:7:12: warning: character constant too long for its type
 <meta name='robots' content='index, nofollow'\>
            ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:7:29: warning: character constant too long for its type
 <meta name='robots' content='index, nofollow'\>
                             ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:8:11: warning: character constant too long for its type
 <link rel='stylesheet' type='text\css' href='\cgit.css'\>
           ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:8:29: warning: character constant too long for its type
 <link rel='stylesheet' type='text\css' href='\cgit.css'\>
                             ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:8:45: warning: character constant too long for its type
 <link rel='stylesheet' type='text\css' href='\cgit.css'\>
                                             ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:9:11: warning: character constant too long for its type
 <link rel='alternate' title='AUser feed' href='http:\\cgit.osmocom.org\rtl-sdr\aUser\include\rtl-sdr.h?h=master' type='application\aUser+xml'\>
           ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:9:29: warning: character constant too long for its type
 <link rel='alternate' title='AUser feed' href='http:\\cgit.osmocom.org\rtl-sdr\aUser\include\rtl-sdr.h?h=master' type='application\aUser+xml'\>
                             ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:9:46: warning: character constant too long for its type
 <link rel='alternate' title='AUser feed' href='http:\\cgit.osmocom.org\rtl-sdr\aUser\include\rtl-sdr.h?h=master' type='application\aUser+xml'\>
                                              ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:9:117: warning: character constant too long for its type
 <link rel='alternate' title='AUser feed' href='http:\\cgit.osmocom.org\rtl-sdr\aUser\include\rtl-sdr.h?h=master' type='application\aUser+xml'\>
                                                                                                                     ^
In file included from C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl_433.c:23:0:
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:12:9: warning: multi-character character constant [-Wmultichar]
 <div id='cgit'><table id='header'>
         ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:12:26: warning: character constant too long for its type
 <div id='cgit'><table id='header'>
                          ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:14:11: warning: multi-character character constant [-Wmultichar]
 <td class='logo' rowspan='2'>[image]<\a><\td>
           ^

C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:14:51: warning: character constant too long for its type
[... 232 lines cut ...]
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:809:1: error: expected identifier or '(' before '<' token
 <\code><\pre><\td><\tr><\table>
 ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:811:12: warning: character constant too long for its type
 <div class='footer'>generated  by cgit v0.9.1 at 2015-08-03 08:26:44 (UTC)<\div>
            ^
[... cut ...]
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h:811:61: error: invalid digit "8" in octal constant
 <div class='footer'>generated  by cgit v0.9.1 at 2015-08-03 08:26:44 (UTC)<\div>
                                                             ^
C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl_433.c:24:21: fatal error: rtl_433.h: No such file or directory
 #include "rtl_433.h"
                     ^
compilation terminated.

Compilation results...
--------
- Errors: 50
- Warnings: 1613
- Compilation Time: 2.14s

pclov3r, 2015-08-04:
Can you attach the full build log instead?

vsboost, 2015-08-04:
The only other log i can find from DevC++ is the compiler log generated after building.

3	13	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
3	53	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
3	63	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
6	12	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
6	32	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
7	12	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
7	29	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
8	11	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
8	29	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
8	45	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
9	11	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
9	29	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
9	46	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
9	117	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
12	9	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
12	26	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
14	11	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
14	51	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
14	67	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
15	11	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
15	51	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
15	66	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
15	104	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
15	124	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
15	137	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Error] empty character constant
16	27	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
17	15	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
17	33	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
18	23	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
18	37	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Error] empty character constant
18	46	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
19	15	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
19	76	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
20	14	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
21	9	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	40	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	73	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	122	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	136	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	186	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	240	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
21	290	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long for its type
[... cut ...]
392	10	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
392	18	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
392	30	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] multi-character character constant [-Wmultichar]
392	42	C:\Users\User\Downloads\rtl_433-master\rtl_433-master\src\rtl-sdr.h	[Warning] character constant too long f

vsboost, 2015-11-12:
@pclov3r could you please post a step by step?

merbanan, 2018-11-30:
There is now a Visual Studio project file.

closed 2018-11-30 by merbanan