merbanan/rtl_433 #133: Support for Fine Offset wh1081

maaw, 2015-07-19:
Hi, this is an extremely popular weather station, with a very crappy receiver. Could you please add support for this one? How can I help you?
Thank you for your work :)
Regards

rct, 2015-07-20:
First, you can probably get more help in the rtl_433 google group.

The next step would be to record some sample signals using rtl_433 -p NNN -a -t and to note what readings they correspond to (tempeature, humdity, wind speed, direction, etc.)

You might have to lower the signal detection threshold with -l 5000 or so.    By default it is 10,000 which is ok if the device is fairly close or you have a good antenna.

Also make sure you figure our your rtl_433 stick's frequency offset (PPM error) to make sure you are listening on the right frequency.

maaw, 2015-07-20:
Thank you very much for your answer @rct. I don't have access to the group. How can i

rct, 2015-07-21:
The google group rtl_433 can be found at: http://groups.google.com/group/rtl_433.

I see you closed this issue, hopefully you will still see this response.  I think you should have left this open until there was support for this weather station.

maaw, 2015-07-21:
Sorry and thank you for your answer. I signed up to the group and found nothing related, so I guess I'll have to do some research before posting.
I'm not sure how I closed the issue.
Again thank you for your attention and please excuse me for not replying before

rct, 2015-07-21:
Take a look at my comment above.

There is a little bit of a process to get things going.

First you need to make sure you are receiving a signal, which usually means getting the right frequency after correcting for the PPM error.  and a reasonable detection threshold (-l) depending on how far away your station is from your receiver.

After that -a -t, can help record signals that can be picked apart.  -D will also help analyze the signal.

Also, since the code you referenced already decodes the protocol, some protocol documentation as well as code could be pulled out of that.

maaw, 2015-07-21:
Yes, I'll try the steps you described. 
I'm pretty close to the transmitter, but all I get is a column of 15 80s and 00s every forty-something seconds when scanning with -A argument. But I'm not sure what parameter should I change to get a more detailed dump. I tried increasing the detection threshold and the message seemed to fragment, but again I'm not very sure what's happening. 
I also have an AlectoV1 Weather Sensor next to the WH1081 transmitter and that message, of course, is decoded properly, so I can cross out signal / receiver issues.

vestom, 2015-08-23:
Hi,

Did you ever succeed in getting it to work? If you add some test files to https://github.com/merbanan/rtl_433_tests, I can try to help. I did the code for the Fine Offset WH2, and they might have something in common.

We are also very interested in test files for the AlectoV1, as they are currently missing. Even though it works now, we might have regressions without test files.

I do not believe it should be necessary to mess with the PPM offset. The detection actually works best, when the signal is offset from the tuner centre frequency, as the tuner may have a so called DC filter.

Just try to record some files with -a -t and maybe try with the (brand new) pulse analyzer function by using -A (big A).

/Tommy

9m2tpt, 2015-09-10:
Anybody has any luck on this one? I've just come across rtl_433 and thinking of having my wh1080 decoded by rtl_433 rather than using the console.

marianowadel, 2015-09-10:
Hi, sorry for not replying before, I haven't had the time. I've managed to make it work with ookd's decoder (https://github.com/jimstudt/ook-decoder). You need only launch ookd, and then run wh1080. 
I'd like to get it to work with rtl_433, but I won't be able to give it a try again until october (Q3 end here at work).
Ideally, I'd like to replace fowsr's functionality (https://code.google.com/p/fowsr/), that pulls data via USB from the wh1080's console, with rtl_433 + a barometer. 
The reason is the console is very unstable, and hangs every 1 or 2 weeks. Plus the built in barometer started failing.
I promise I'll keep you guys updated.
Cheers

maaw, 2015-09-10:
Hi, sorry for not replying before, I haven't had the time. I've managed to make it work with ookd's decoder (https://github.com/jimstudt/ook-decoder). You need only launch ookd, and then run wh1080. 
I'd like to get it to work with rtl_433, but I won't be able to give it a try again until october (Q3 end here at work).
Ideally, I'd like to replace fowsr's functionality (https://code.google.com/p/fowsr/), that pulls data via USB from the wh1080's console, with rtl_433 + a barometer. 
The reason is the console is very unstable, and hangs every 1 or 2 weeks. Plus the built in barometer started failing.
I promise I'll keep you guys updated.
Cheers

merbanan, 2015-09-10:
@9m2tpt and @maaw. You can start with sending recordings of the signal to the rtl_433_tests repository. Then others can aid in development also.

9m2tpt, 2015-09-11:
I have done the recording but having hard time to find a way to upload them
to the repository. Will find a way to upload.

9m2tpt, 2015-09-21:
Hi @merbanan , since I've no idea how to upload them into the repository, the gdata files can be found in my shared google drive.

https://drive.google.com/folderview?id=0B1aqySKViHWSVENtNldTUWVmZ1E&usp=sharing

there are four files there, let me know if you need extra.

ovrheat, 2016-02-12:
Good news: WH1080 is now supported on rtl_433 (tested on its 868.3 Mhz version, it **_should_** work for other versions too: just adjust -f  (frequency) parameter, and use -l 0 as level parameter (auto)).
For my 868.3 version the working command line is 'rtl_433 -f 868250000 -l 0' .
Time signal decoding is also full working.
I need some feedback, so let me know.

maaw, 2016-02-12:
ovrheat, that's awesome. I've broken my raspbian microsd card so I'm not
able to test it right now, but i'll get it fixed asap in order to try this.
Thank you for your work, and sorry for not being so helpful.
I'll let you know when I test it.
Cheers!

9m2tpt, 2016-02-12:
Just tested with my WH1080 using the standard frequency at 433.920MHz (got mine few years ago from Maplin) and it works great. The only problem probably the frequency drifting on my RTL_SDR, I need to take a look into it. Thanks @ovrheat for the good job.

closed 2016-02-12 by merbanan