merbanan/rtl_433 #134: Add FSK (Frequency Shift Keying) demodulation

rct, 2015-07-20:
Currently, rtl_433, only demodulates OOK, (On-Off-Keying), a modulation method were the signal is turned off, (no carrier), between pulses.  

See issue #14.  

Some other devices like LaCrosse 915 Mhz devices use FSK, while the LaCrosse 433 Mhz devices use OOK.

(I'm opening this issue to collect the info and track progress in one place.) 

@magellannh has code that uses rtl_fm https://github.com/magellannh/RPI_Efergy,   rtl_wx https://github.com/magellannh/rtl-wx, decodes an Efergy device, but I'm not sure if that is the same FSK one or not. 

Note: @vestom and @merbanan recently discussed adding FSK decoding in the rtl_433 google group.

Batilan, 2015-07-21:
Yes this would indeed alow rtl_433 to decode a whole new range of devices. 

One example that might give some pointers is ec3k (https://pypi.python.org/pypi/ec3k which decodes Voltcraft Energy Count 3000 / Velleman NETBSEM4 / La Crosse Techology Remote Cost Control Monitor – RS3620. It uses gnu-radio to do that, but the way the blocks are used might give some pointers. 

Note that ec3k is quite CPU intensive (probably because cascading gnu-blocks isn't that efficient. A strange effect that can be observed here is the strange sensitivity around the center frequency. See also https://batilanblog.wordpress.com/2015/02/17/using-ec3k-with-raspberry-pi/ (around 2/3 of the page see the bargraph). 

Probably a DSP savy guy can make some sense of this.

Batilan, 2015-07-21:
Ah, just saw on the google group that a renowned FSK savy guy already gave a very good hint ;-). Probably forget about the ec3k gnu-radio blocks and go with the advice from Benjamin

merbanan, 2015-10-18:
FSK demod has been added some time ago.

closed 2015-10-18 by merbanan