merbanan/rtl_433 #136: Please implement  gracefull exit when usb stick becomes inactive/removed/whatever

jcarduino, 2015-08-01:
When the stick is removed rtl_433 does not exit.

keeps repeating:

rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
r82xx_write: i2c wr failed=-4 reg=17 len=1
r82xx_set_freq: failed=-4
rtlsdr_demod_write_reg failed with -4
rtlsdr_demod_read_reg failed with -4
WARNING: Failed to set center freq.
Failed to submit transfer 0!

in stead of gracefull exit or better try reinit/reconnect to stick

killerbees19, 2017-06-04:
+1 for this! 🤓

It's using 100% CPU…

zuckschwerdt, 2017-06-04:
What are the exact command line options you use, and what specific OS?

killerbees19, 2017-06-04:
OS: Debian Jessie
Command: `rtl_433 -R 19 -R 50 -R 55 -g 50 -l 1 -F json -T 900`

But I think I've found a simple fix: https://github.com/froonix/rtl_433/commit/8e606bde58269578fdb6e51c13cb07a2e73d7423

```
cb transfer status: 1, canceling...
WARNING: async read failed.

Library error -5, exiting...
```

killerbees19, 2017-06-05:
Thanks for merging! :-)

I think this issue could be closed now, what do you think?

closed 2017-06-06 by merbanan