merbanan/rtl_433 #141: TFA Twin Plus 30.3049

iotzo, 2015-08-16:
Hi!
I've created little bash script to support TFA Twin Plus 30.3049 thermo/hygrometer protocol:

 rtl_433 -f 433920000 -l 500 -a -g 18 2>&1 |mawk -W interactive -F ":" '/\[01\] {36}/{print $2}' | sed -u 's/ //g'| stdbuf -oL rev | mawk -W interactive '{ printf"ibase=2; %s\n",substr($0,6,7) ; printf"ibase=2; %s\n", substr($0,16,9) }' | bc | mawk -W interactive 'NR==1{ print"----------------------------\n" ; system("date") ; printf"Humidity\t: %s\n",$1-28 } NR==2{ printf"Temperature\t: %s\n",$1/10 ; NR=0 }'

Here is some sample data:
[04] {36} e4 4b 70 73 00 : 111001000100 101101110 000 0111001 10000    ---> temp/hum:23.7/50
temp num-->13-21bit(9bits) in reverse order in this case "011101101"=237
positive temps ( with 000 in bits 22-24) : temp=num/10 (in this case 23.7 C)
negative temps (with 111 in bits 22-24) :  temp=(512-num)/10
negative temps example:
[03] {36} e4 4c 1f 73 f0 : 111001000100 110000011 111 0111001 11111 temp: -12.4 

Humidity:
hum num-->25-32bit(7bits) in reverse order : in this case "1001110"=78
humidity=num-28 --> 78-28=50

It would be nice if we have official support for that model/protocol!
10x!

merbanan, 2015-08-16:
Hi, add some sample files, look at:

https://github.com/merbanan/rtl_433_tests

This will help people to write the code for your protocol. And I will not accept new protocols without test files.

iotzo, 2015-08-16:
OK!
Done!
Sending pull request for the sample files!

ortylp, 2015-08-17:
I have working implementation based on your description.  I need to clean the code a bit, add some more tests and push to github.  I hope I get to it tomorrow at the evening.

We need to decode the purpose of other bits and kind of CRC, if any.  THIS IS IMPORTANT, otherwise we get false positives and that means disaster.  Can you please test if there are things like "channel" or "battery status", do bits change after battery replacement?

iotzo, 2015-08-19:
It works now, thank You!
The last 4bits maybe is some kind of CRC. Here is some samples:
{36} e4 25 c8 87 50 : 111001000010 010111001 000 1000011 1 0101
{36} e4 2e c8 bb 40 : 111001000010 111011001 000 1011101 1 0100
{36} e4 26 c8 9b b0 : 111001000010 011011001 000 1001101 1 1011
{36} e4 2a c8 6b 90 : 111001000010 101011001 000 0110101 1 1001
{36} e4 22 c8 cb a0 : 111001000010 001011001 000 1100101 1 1010
{36} e4 2c c8 0b 80 : 111001000010 110011001 000 0000101 1 1000
{36} e4 24 c8 b3 30 : 111001000010 010011001 000 1011001 1 0011
{36} e4 28 c8 d3 90 : 111001000010 100011001 000 1101001 1 1001
{36} e4 4f 48 63 f0 : 111001000100 111101001 000 0110001 1 1111
{36} e4 47 48 a3 b0 : 111001000100 011101001 000 1010001 1 1011
{36} e4 47 48 23 30 : 111001000100 011101001 000 0010001 1 0011
{36} e4 40 50 a3 60 : 111001000100 000001010 000 1010001 1 0110
{36} e4 44 e0 43 40 : 111001000100 010011100 000 0100001 1 0100
{36} e4 4f 20 c3 b0 : 111001000100 111100100 000 1100001 1 1011
{36} e4 42 c0 43 00 : 111001000100 001011000 000 0100001 1 0000
{36} e4 4b 80 c3 10 : 111001000100 101110000 000 1100001 1 0001
{36} e4 4d 00 c3 a0 : 111001000100 110100000 000 1100001 1 1010

{36} e4 4d ff 23 20 : 111001000100 110111111 111 0010001 1 0010
{36} e4 4b 7f 23 a0 : 111001000100 101101111 111 0010001 1 1010
{36} e4 40 7f a3 90 : 111001000100 000001111 111 1010001 1 1001
{36} e4 4a bf 93 80 : 111001000100 101010111 111 1001001 1 1000
{36} e4 43 3f e3 a0 : 111001000100 001100111 111 1110001 1 1010
{36} e4 42 3f e3 b0 : 111001000100 001000111 111 1110001 1 1011
{36} e4 43 df 13 a0 : 111001000100 001111011 111 0001001 1 1010
{36} e4 4a df 93 f0 : 111001000100 101011011 111 1001001 1 1111
{36} e4 47 5f 93 e0 : 111001000100 011101011 111 1001001 1 1110
{36} e4 41 5f 53 40 : 111001000100 000101011 111 0101001 1 0100
{36} e4 49 9f 33 20 : 111001000100 100110011 111 0011001 1 0010
{36} e4 41 1f b3 c0 : 111001000100 000100011 111 1011001 1 1100
{36} e4 4c 1f 73 f0 : 111001000100 110000011 111 0111001 1 1111

ortylp, 2015-08-24:
@iotzo Until we have the CRC/checksum working, this decoder will not be merged into the upstream.

merbanan, 2015-08-24:
I am not that strict but I really prefer that the crc/checksum is working. You can validate in other ways also. But as noone has sent a pull request for any code there is nothing to review and nothing to merge.

iotzo, 2015-08-24:
I tried, but i can't identify the type of this checksum:(

ortylp, 2015-08-25:
As I am running with my "enhanced sensitivity" I am successfully receiving many more sensors from the neighbourhood.  As the amount of properly demodulated emissions raised dramatically. the number of not properly recognised bit packages went also unproportionally high.  The prototype for "TFA Twin Plus 30.3049" is also there, despite check on the number of bits in the package false positives come quite often.

merbanan, 2015-08-25:
Regarding the checksum I think this gives some vital information.

e4 43 3f e3 a : 111001000100 001100111 111 1110001 1 1010
e4 42 3f e3 b : 111001000100 001000111 111 1110001 1 1011

The difference is just one number 3 vs 2. But the checksum value is 0xa vs 0xb. So there is difference of off 1 but not one more it is one less.

merbanan, 2015-08-25:
e4 47 48 a3 b
e4 47 48 23 3

a-2 = 8
b-3 = 8

So there seems to be a linear relationship here.

iotzo, 2015-08-25:
Good observations!
I think that the check sum is the sum of the nibbles in reverse order(from the result we should take least significant bits). For example:
1) if we reverse "111001000100 001100111 111 1110001 1" , then the result is:
1 1000111 111 111001100 001000100111, so lets sum all the nibbles:
echo "ibase=2;obase=2;1100+0111+1111+1100+1100+0010+0010+0111"|bc
1000101
, and if we read the result in reverse order we get: 1010...lets take only first 4bits(from right to left)
Lets do the same with the second line:
echo "111001000100 001000111 111 1110001 1"|rev
1 1000111 111 111000100 001000100111
echo "ibase=2;obase=2;1100+0111+1111+1100+0100+0010+0010+0111"|bc
111101
Reading the result from right to left : 1011..
I've checked other lines and the results are consistent!
PS:
One more thing!
Notice, that sometimes hex sequence begins with "e44" and sometimes with "e42"

merbanan, 2015-08-25:
I think that most 4 bit checksums are all nibbles summed together without the overflow. Either in big or little endian.

ortylp, 2015-08-25:
It looks really promising :)

merbanan, 2015-08-25:
I checked the last and the sum matched, good work.

merbanan, 2015-08-25:
@ortylp will you update your code and send a pull ?

ortylp, 2015-08-25:
yep, but unfortunately not today, my target is around Sunday

iotzo, 2015-08-25:
I'll try these days! I've noticed that on the LCD screen on the internal panel there is CH1[23], but CH1 i always active and i don't know how to force channel change:(

iotzo, 2015-08-27:
I have channel number bits(5,6 in reverse order) and low battery bit(9)   :)
Actually in this model you can't change channel number(it's always CH1), because there is NO channel switch on the circuit board. I had to solder some wires and fiddle with them to change channel. Here some samples with different channels(1,2 or 3), low batteries and some consecutive resets(it seems that the  1,2,3,4,7,8 bits changes randomly on every reset/battery change):
[01] {36} 24 01 f0 13 80 : 0010 0100 0000 0001 11110000 00010011 1000   CH1
[01] {36} 84 02 08 83 a0 : 1000 0100 0000 0010 00001000 10000011 1010   CH1
[01] {36} 96 02 08 83 80 : 1001 0110 0000 0010 00001000 10000011 1000   CH1
[01] {36} c4 0c 08 83 60 : 1100 0100 0000 1100 00001000 10000011 0110   CH1
[01] {36} 46 0c 08 83 90 : 0100 0110 0000 1100 00001000 10000011 1001   CH1
[01] {36} 75 0c 08 83 90 : 0111 0101 0000 1100 00001000 10000011 1001   CH1
[01] {36} 75 04 08 03 e0 : 0111 0101 0000 0100 00001000 00000011 1110   CH1
[01] {36} 96 8e 88 c3 10 : 1001 0110 1000 1110 10001000 11000011 0001   CH1/LOW BAT
[01] {36} 96 81 88 83 e0 : 1001 0110 1000 0001 10001000 10000011 1110   CH1/LOW BAT
[01] {36} 96 86 88 83 a0 : 1001 0110 1000 0110 10001000 10000011 1010   CH1/LOW BAT

[01] {36} a8 02 48 c3 30 : 1010 1000 0000 0010 01001000 11000011 0011   CH2
[01] {36} a8 40 88 83 e0 : 1010 1000 0100 0000 10001000 10000011 1110   CH2
[01] {36} 88 05 08 83 50 : 1000 1000 0000 0101 00001000 10000011 0101   CH2
[01] {36} 0b 09 08 43 a0 : 0000 1011 0000 1001 00001000 01000011 1010   CH2
[01] {36} 5b 0a 08 83 50 : 0101 1011 0000 1010 00001000 10000011 0101   CH2
[01] {36} 7a 0c 08 83 20 : 0111 1010 0000 1100 00001000 10000011 0010   CH2
[01] {36} da 04 08 83 00 : 1101 1010 0000 0100 00001000 10000011 0000   CH2
[01] {36} b8 08 08 83 b0 : 1011 1000 0000 1000 00001000 10000011 1011   CH2
[01] {36} fb 08 08 83 d0 : 1111 1011 0000 1000 00001000 10000011 1101   CH2
[01] {36} 68 00 08 83 a0 : 0110 1000 0000 0000 00001000 10000011 1010   CH2
[01] {36} ea 00 08 83 50 : 1110 1010 0000 0000 00001000 10000011 0101   CH2
[01] {36} 69 00 08 83 b0 : 0110 1001 0000 0000 00001000 10000011 1011   CH2
[01] {36} f8 00 08 83 70 : 1111 1000 0000 0000 00001000 10000011 0111   CH2
[01] {36} e9 0f f0 83 d0 : 1110 1001 0000 1111 11110000 10000011 1101   CH2
[01] {36} 1b 0f f0 83 00 : 0001 1011 0000 1111 11110000 10000011 0000   CH2
[01] {36} 98 0f f0 83 a0 : 1001 1000 0000 1111 11110000 10000011 1010   CH2
[01] {36} 4b 07 f0 83 90 : 0100 1011 0000 0111 11110000 10000011 1001   CH2
[01] {36} ca 07 f0 83 40 : 1100 1010 0000 0111 11110000 10000011 0100   CH2
[01] {36} 8b 07 f0 83 10 : 1000 1011 0000 0111 11110000 10000011 0001   CH2
[01] {36} da 07 f0 83 50 : 1101 1010 0000 0111 11110000 10000011 0101   CH2
[01] {36} 29 07 f0 83 e0 : 0010 1001 0000 0111 11110000 10000011 1110   CH2
[01] {36} 8b 00 08 43 b0 : 1000 1011 0000 0000 00001000 01000011 1011   CH2
[01] {36} 2b 0f f0 83 30 : 0010 1011 0000 1111 11110000 10000011 0011   CH2
[01] {36} 59 07 f0 83 b0 : 0101 1001 0000 0111 11110000 10000011 1011   CH2
[01] {36} aa 0a 88 63 c0 : 1010 1010 0000 1010 10001000 01100011 1100   CH2
[01] {36} 38 08 08 43 b0 : 0011 1000 0000 1000 00001000 01000011 1011   CH2

[01] {36} ae 08 88 c3 70 : 1010 1110 0000 1000 10001000 11000011 0111   CH3
[01] {36} 4f 0d 08 83 50 : 0100 1111 0000 1101 00001000 10000011 0101   CH3
[01] {36} be 05 08 83 30 : 1011 1110 0000 0101 00001000 10000011 0011   CH3
[01] {36} 3e 01 08 83 90 : 0011 1110 0000 0001 00001000 10000011 1001   CH3
[01] {36} bd 01 08 83 70 : 1011 1101 0000 0001 00001000 10000011 0111   CH3
[01] {36} ac 0e 08 83 b0 : 1010 1100 0000 1110 00001000 10000011 1011   CH3

ortylp, 2015-09-01:
Almost all bits are decoded, checksum is working.
Negative temperature formula is too complex to be the right one, more samples and simplification is needed, the same for humidity.

Please test it with the hardware, compare temperature AND humidity. Humidity usually goes pretty low at negative temperatures.

iotzo, 2015-09-01:
Temp and humidity are ok, including negative temps.
Thank You:)
I'll test channel and battery bits tomorrow and report back in case of some problem.
Here is some output:
2015-09-01 13:52:57 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 1.4 C / 34.5 F, humidity 36%
2015-09-01 13:52:57 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 1.4 C / 34.5 F, humidity 36%
2015-09-01 13:52:57 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 1.4 C / 34.5 F, humidity 36%
2015-09-01 13:52:57 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 1.4 C / 34.5 F, humidity 36%

2015-09-01 13:53:29 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.8 C / 33.4 F, humidity 37%
2015-09-01 13:53:29 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.8 C / 33.4 F, humidity 37%
2015-09-01 13:53:29 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.8 C / 33.4 F, humidity 37%
2015-09-01 13:53:29 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.8 C / 33.4 F, humidity 37%
2015-09-01 13:54:01 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.3 C / 32.5 F, humidity 37%
2015-09-01 13:54:01 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.3 C / 32.5 F, humidity 37%
2015-09-01 13:54:01 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.3 C / 32.5 F, humidity 37%
2015-09-01 13:54:01 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 0.3 C / 32.5 F, humidity 37%
2015-09-01 13:54:33 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.2 C / 31.6 F, humidity 38%
2015-09-01 13:54:33 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.2 C / 31.6 F, humidity 38%
2015-09-01 13:54:33 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.2 C / 31.6 F, humidity 38%
2015-09-01 13:54:33 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.2 C / 31.6 F, humidity 38%
2015-09-01 13:55:05 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.7 C / 30.7 F, humidity 39%
2015-09-01 13:55:05 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.7 C / 30.7 F, humidity 39%
2015-09-01 13:55:05 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.7 C / 30.7 F, humidity 39%
2015-09-01 13:55:05 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -0.7 C / 30.7 F, humidity 39%
2015-09-01 13:55:37 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.1 C / 30.0 F, humidity 40%
2015-09-01 13:55:37 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.1 C / 30.0 F, humidity 40%
2015-09-01 13:55:37 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.1 C / 30.0 F, humidity 40%
2015-09-01 13:55:37 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.1 C / 30.0 F, humidity 40%
2015-09-01 13:56:09 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.5 C / 29.3 F, humidity 40%
2015-09-01 13:56:09 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.5 C / 29.3 F, humidity 40%
2015-09-01 13:56:09 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.5 C / 29.3 F, humidity 40%
2015-09-01 13:56:09 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.5 C / 29.3 F, humidity 40%
2015-09-01 13:56:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.9 C / 28.6 F, humidity 41%
2015-09-01 13:56:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.9 C / 28.6 F, humidity 41%
2015-09-01 13:56:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.9 C / 28.6 F, humidity 41%
2015-09-01 13:56:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -1.9 C / 28.6 F, humidity 41%
2015-09-01 13:57:13 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.3 C / 27.9 F, humidity 42%
2015-09-01 13:57:13 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.3 C / 27.9 F, humidity 42%
2015-09-01 13:57:13 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.3 C / 27.9 F, humidity 42%
2015-09-01 13:57:13 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.3 C / 27.9 F, humidity 42%
2015-09-01 13:57:45 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.6 C / 27.3 F, humidity 43%
2015-09-01 13:57:45 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.6 C / 27.3 F, humidity 43%
2015-09-01 13:57:45 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.6 C / 27.3 F, humidity 43%
2015-09-01 13:57:45 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.6 C / 27.3 F, humidity 43%
2015-09-01 13:58:17 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.9 C / 26.8 F, humidity 41%
2015-09-01 13:58:17 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.9 C / 26.8 F, humidity 41%
2015-09-01 13:58:17 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.9 C / 26.8 F, humidity 41%
2015-09-01 13:58:17 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -2.9 C / 26.8 F, humidity 41%
2015-09-01 13:58:49 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature -3.3 C / 26.1 F, humidity 42%
[... cut ...]
2015-09-01 14:28:09 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 20.6 C / 69.1 F, humidity 52%
2015-09-01 14:28:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 20.7 C / 69.3 F, humidity 51%
2015-09-01 14:28:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 20.7 C / 69.3 F, humidity 51%
2015-09-01 14:28:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 20.7 C / 69.3 F, humidity 51%
2015-09-01 14:28:41 TFA-Twin-Plus-30.3049 Sensor 15: battery OK, channel 1, temperature 20.7 C / 69.3 F, humidity 51%

ortylp, 2015-09-02:
OK, if you are sure it works as supposed, then we can close this issue.

iotzo, 2015-09-02:
The channel and battery bits are ok too!

closed 2015-09-02 by iotzo