merbanan/rtl_433 #149: implementing Esperanza EWS

Alberts00, 2015-08-21:
Hello! 
I would like to thank all developers who have made this tool. Thank you!

A few days ago I remembered that I have a temperature sensor which runs on 433Mhz. I have successfully found the most relevant pieces of information which were given to my using the analyse output (-a flag)
```
*** signal_start = 800122, signal_end = 1062693
signal_len = 262571,  pulses = 266
Iteration 1. t: 152    min: 151 (23)    max: 153 (243)    delta 9
Iteration 2. t: 152    min: 151 (2)    max: 153 (264)    delta 0
Distance coding: Pulse length 152

Short distance: 485, long distance: 962, packet distance: 1922

p_limit: 152
bitbuffer:: Number of rows: 14 
[00] {0} : 
[01] {0} : 
[02] {42} 0b 9b b5 a4 86 00 : 00001011 10011011 10110101 10100100 10000110 00
[03] {0} : 
[04] {42} 0b 9b b5 a4 86 00 : 00001011 10011011 10110101 10100100 10000110 00
[05] {0} : 
[06] {42} 0b 9b b5 a4 86 00 : 00001011 10011011 10110101 10100100 10000110 00
[07] {0} : 
[08] {42} 0b 9b b5 a4 86 00 : 00001011 10011011 10110101 10100100 10000110 00
[09] {0} : 
[10] {42} 0b 9b b5 a4 86 00 : 00001011 10011011 10110101 10100100 10000110 00
[11] {0} : 
[12] {42} 0b 9b b5 a4 86 00 : 00001011 10011011 10110101 10100100 10000110 00
[13] {0} :```
After that I found where and how the humidity, channel and temperature is stored.
These are the settings I'm currently using for my sensor when implementing callback.
```r_device esperanza_ews = {
  .name          = "Esperanza EWS-103",
  .modulation    = OOK_PULSE_PPM_RAW,
  .short_limit   = 152,
  .long_limit    = 2500,
  .reset_limit   = 5000,
  .json_callback = &esperanza_ews_callback,
  .disabled      = 0,
  .demod_arg     = 0,
};```

There are a few issues here:
1) The callback doesn't always get called, even though debug does show the information.
Here's an example
```
Pulse width distribution:
 [ 0] mean:  152 [150;155],	 count: 266
Gap width distribution:
 [ 0] mean: 1923 [1918;1939],	 count:  13
 [ 1] mean:  486 [479;493],	 count: 126
 [ 2] mean:  964 [960;982],	 count: 126
Pulse period distribution:
 [ 0] mean: 2076 [2072;2092],	 count:  13
 [ 1] mean:  639 [632;647],	 count: 126
 [ 2] mean: 1117 [1111;1136],	 count: 126
Guessing modulation: Pulse Position Modulation with fixed pulse width

Detected OOK package
Analyzing pulses...
Total number of pulses: 266
Pulse width distribution:
 [ 0] mean:  153 [151;155],	 count: 266
Gap width distribution:
 [ 0] mean: 1923 [1920;1927],	 count:  13
 [ 1] mean:  486 [479;501],	 count: 120
 [ 2] mean:  964 [959;985],	 count: 132
Pulse period distribution:
 [ 0] mean: 2076 [2072;2081],	 count:  13
 [ 1] mean:  639 [632;654],	 count: 120
 [ 2] mean: 1117 [1112;1139],	 count: 132
Guessing modulation: Pulse Position Modulation with fixed pulse width

Detected OOK package

called 			 bitbuffer length 264

pulse_demod_ppm(): Esperanza EWS-103 
bitbuffer:: Number of rows: 1 
[00] {264} ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
Analyzing pulses...
Total number of pulses: 265
Pulse width distribution:
 [ 0] mean:  153 [151;155],	 count: 264
 [ 1] mean: 1989 [1989;1989],	 count:   1
Gap width distribution:
 [ 0] mean: 1923 [1918;1927],	 count:  13
 [ 1] mean:  486 [479;494],	 count: 120
 [ 2] mean:  964 [908;984],	 count: 130
 [ 3] mean:  302 [302;302],	 count:   1
Pulse period distribution:
 [ 0] mean: 2091 [2072;2291],	 count:  14
 [ 1] mean:  639 [632;648],	 count: 120
 [ 2] mean: 1117 [1060;1139],	 count: 130
Guessing modulation: No clue...
```
2) If it does get called, the bitbuffer doesn't consist of 14 rows, which were seen in the analyse output.

Is it possible to get the same bitbuffer as I have when using -a?
A few samples are downloadable here.
http://vpslv.tirlins.com/data/rtl433/

ortylp, 2015-08-21:
OOK_PULSE_PPM_RAW
short limit = (1800+900)/2

merbanan, 2015-09-02:
Can you send a pull request to the rtl_433_tests repository ?

Alberts00, 2015-09-02:
I will soon, sorry for the delay.
@ortylp Still the same issue.

closed 2015-10-06 by Alberts00