merbanan/rtl_433 #150: rtl_433 -a gives strange output with cresta sensor. Help appreciated!

jcarduino, 2015-08-21:
Can someone please indicate what needs to be done to get the proper  OOK signal from a cresta sensor?
3 raw samples: https://www.dropbox.com/s/eqip9237hdxuitq/filname.raw?dl=0
sample files: -at https://www.dropbox.com/s/tlkv8nflp62vy71/data.zip?dl=0

-a gives me following, which leaves me clueless:

*** signal_start = 8105744, signal_end = 8219150
signal_len = 113406,  pulses = 199
Iteration 1. t: 187    min: 125 (125)    max: 249 (74)    delta 8381
Iteration 2. t: 187    min: 125 (125)    max: 249 (74)    delta 0
Distance coding: Pulse length 187

Short distance: 116, long distance: 238, packet distance: 13679

p_limit: 187
bitbuffer:: Number of rows: 3 
[00] {64} c9 0a 6c 91 06 25 e2 56 
[01] {66} c9 0a 6c 91 06 25 e2 40 80 
[02] {66} c9 0a 6d 11 06 25 e2 04 80 
*** signal_start = 18856812, signal_end = 18970217
signal_len = 113405,  pulses = 204
Iteration 1. t: 187    min: 125 (135)    max: 249 (69)    delta 8545
Iteration 2. t: 187    min: 125 (135)    max: 249 (69)    delta 0
Distance coding: Pulse length 187

Short distance: 116, long distance: 238, packet distance: 13682

p_limit: 187
bitbuffer:: Number of rows: 3 
[00] {67} c9 0a 6c 82 03 12 f0 24 40 
[01] {67} c9 0a 6c 82 03 12 f0 41 40 
[02] {67} c9 0a 6d 02 03 12 f0 10 c0 
*** signal_start = 29607876, signal_end = 29721282
signal_len = 113406,  pulses = 204
Iteration 1. t: 187    min: 125 (135)    max: 249 (69)    delta 8362
Iteration 2. t: 187    min: 125 (135)    max: 249 (69)    delta 0
Distance coding: Pulse length 187

Short distance: 116, long distance: 238, packet distance: 13682

p_limit: 187
bitbuffer:: Number of rows: 3 
[00] {67} c9 0a 6c 82 03 12 f0 24 40 
[01] {67} c9 0a 6c 82 03 12 f0 41 40 
[02] {67} c9 0a 6d 02 03 12 f0 10 c0 
*** signal_start = 40358945, signal_end = 40472346
signal_len = 113401,  pulses = 195
Iteration 1. t: 187    min: 125 (117)    max: 249 (78)    delta 8297
Iteration 2. t: 187    min: 125 (117)    max: 249 (78)    delta 0
Distance coding: Pulse length 187

Short distance: 116, long distance: 238, packet distance: 13681

p_limit: 187
bitbuffer:: Number of rows: 3 
[00] {63} c9 0a 6c c4 06 25 f9 38 
[01] {64} c9 0a 6c c4 06 25 fa 48 
[02] {65} c9 0a 6d 44 06 25 f8 90 00 
*** signal_start = 51110006, signal_end = 51223405
signal_len = 113399,  pulses = 195
Iteration 1. t: 187    min: 125 (117)    max: 249 (78)    delta 8473
Iteration 2. t: 187    min: 125 (117)    max: 249 (78)    delta 0
Distance coding: Pulse length 187

Short distance: 116, long distance: 238, packet distance: 13682

p_limit: 187
bitbuffer:: Number of rows: 3 
[00] {63} c9 0a 6c c4 06 25 f9 38 
[01] {64} c9 0a 6c c4 06 25 fa 48 
[02] {65} c9 0a 6d 44 06 25 f8 90 00 
*** signal_start = 61861065, signal_end = 61974465
signal_len = 113400,  pulses = 195
Iteration 1. t: 186    min: 124 (117)    max: 249 (78)    delta 8109
Iteration 2. t: 186    min: 124 (117)    max: 249 (78)    delta 0
Pulse coding: Short pulse length 124 - Long pulse length 249

Short distance: 116, long distance: 239, packet distance: 13681

p_limit: 186
bitbuffer:: Number of rows: 3 
[00] {64} e1 23 2d 60 42 89 fc 58 
[01] {65} e1 23 2c e0 42 89 fc 48 00 
[02] {66} e1 23 2d 60 42 89 fc 40 00 
*** signal_start = 72612124, signal_end = 72725524
signal_len = 113400,  pulses = 195
Iteration 1. t: 186    min: 124 (117)    max: 249 (78)    delta 8109
Iteration 2. t: 186    min: 124 (117)    max: 249 (78)    delta 0
Pulse coding: Short pulse length 124 - Long pulse length 249

Short distance: 116, long distance: 238, packet distance: 13682

p_limit: 186
bitbuffer:: Number of rows: 3 
[00] {64} e1 23 2d 60 42 89 fc 58 
[01] {65} e1 23 2c e0 42 89 fc 48 00 
[02] {66} e1 23 2d 60 42 89 fc 40 00 
*** signal_start = 83363186, signal_end = 83476585
signal_len = 113399,  pulses = 197
Iteration 1. t: 187    min: 125 (121)    max: 249 (76)    delta 8297
Iteration 2. t: 187    min: 125 (121)    max: 249 (76)    delta 0
Distance coding: Pulse length 187

Short distance: 116, long distance: 238, packet distance: 13682

p_limit: 187
bitbuffer:: Number of rows: 3 
[00] {65} c9 0a 6c a1 06 25 ec a0 00 
[01] {64} c9 0a 6c a1 06 25 ec c4 
[02] {65} c9 0a 6d 21 06 25 ec 06 00

vestom, 2015-08-23:
Hi,

It seems like you are already getting a somewhat proper signal of ~65 bits :-)
I looked at your test files in data.zip. Some of the files only contain a long unmodulated pulse (gfile001, gfile003). However the others (gfile002, 4, 5) contain a nice Pulse Width Modulated signal of 64-67 bits and a single short pulse signifying the end (see below). You could make a device decoder using calibeur.c as a starting point with the values shown below (short_limit, long_limit, ...). Please also push your (valid) test files for inclusion in https://github.com/merbanan/rtl_433_tests, otherwise we cannot test for regressions.

Have fun :-)
/Tommy

$ rtl_433 -A -r cresta/01/gfile002.data 
...
Detected OOK package
Analyzing pulses...
Total number of pulses: 68
Pulse width distribution:
 [ 0] mean:  248 [246;252],	 count:  23
 [ 1] mean:  125 [124;128],	 count:  44
 [ 2] mean:   31 [31;31],	 count:   1
Gap width distribution:
 [ 0] mean:  240 [239;243],	 count:  23
 [ 1] mean:  118 [116;121],	 count:  44
Pulse period distribution:
 [ 0] mean:  489 [487;492],	 count:  10
 [ 1] mean:  366 [363;368],	 count:  26
 [ 2] mean:  244 [241;247],	 count:  31
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 78, long_limit: 186, reset_limit: 244, demod_arg: 0
pulse_demod_pwm_ternary(): Analyzer Device 
bitbuffer:: Number of rows: 2 
[00] {67} e3 03 2d 12 43 09 34 90 00 
[01] {0} :

jcarduino, 2015-08-24:
Thanks... Will take it further from here.

Will also add some testfiles for electov1 this week

jcarduino, 2015-08-25:
Still some fighting here with bitstream...

Next is from some documentation from a arduino program. I am using.

Gives me: Raw Data: 57 11 cb a5 21 c1 f4 66 55 35 8b 00 00 00 while working program reports 7526CEDE24C254FE7A340000 

I am using OOK_PULSE_MANCHESTER_ZEROBIT as it is manchester with a startbit 0..

/* I'll follow CrestaProtocol documentation here. However, I suspect it is inaccurate at some points:
	* - there is no stop-bit after every byte. Instead, there's a start-bit (0) before every byte.
	* - Conversely, there is no start-bit "1" before every byte.
	* - An up-flank is 0, down-flank is 1, at least with both my receivers.
	*
	* However, since the first start-bit 0 is hard to distinguish given the current clock-detecting
	* algorithm, I pretend there *is* a stop-bit 0 instead of start-bit. However, this means the
	* last stop-bit of a package must be ignored, as it simply isn't there.
	*
	* This manchester decoder is based on the principle that short edges indicate the current bit is the
	* same as previous bit, and that long edge indicate that the current bit is the complement of the
	* previous bit.
	*/

Any clues?

jcarduino, 2015-08-26:
Question is... do I need to write a decoder for it????

vestom, 2015-08-27:
You are right. The signal is Manchester and not PWM. The last short bit fools the protocol detection. 
You should use a short_limit of 125, long_limit of 0 (not used) and reset_limit of 375(ish).
From your text above, I would guess you would need to assume each byte is prefixed with a 0 - and this means you will have to write some decoding code to get meaningful data.
Do you have a link to either the protocol documentation or Arduino code you mention?

jcarduino, 2015-09-01:
Uploaded files have arduino code.

ua3nbw, 2015-12-03:
decoder for cresta sensor

/*
 * RST  temperature / humidity sensors
 *
 * Copyright (c) 2015, Valery Rezvyakov, Kostroma, RUSSIA (ua3nbw)
 * 
 * Радиодатчик RST 02510
 * Message Format can be found at: http://members.upc.nl/m.beukelaar/Crestaprotocol.pdf
 * Радиодатчики RST 02510, RST 02500(термометр) подходят для моделей:
 * 
 * RST 01923, RST 01925, RST 01927, RST 02515, RST 02518, RST 02519,RST 02520, RST 02521, RST 02523, RST 02525, 
 * RST 02527, RST 02534,RST 02535, RST 02536, RST 32508 и т.д.
 * -
 * •Возможна работа с моделями:
 * RST 02501, RST 02503, RST 02505, RST 02517, RST 02531, RST 02532, RST 02533, RST 32502, RST 32503, RST 32504,
 * RST 32505, RST 32506, RST 32507, RST 32509 и т.д.,
 * но эти модели не могут отображать влажность, поэтому с ними лучше использовать радиодатчик RST 02500(термометр)
 *  
 * •Возможна работа с некоторыми temperature / humidity sensors фирм Atomic, Meteomaster, Vitek, Hama и т.д.
 * 
 * 
 */
```
 #include "rtl_433.h"
 #include "util.h"

#define RST_BITLEN		91
#define RST_BROWLEN		9
static char time_str[LOCAL_TIME_BUFLEN];

static int rst_weather_callback(bitbuffer_t *bitbuffer) {

	if (bitbuffer->bits_per_row[0] != RST_BITLEN) { 
         return 0; 
     } 

           uint8_t *br = bitbuffer->bb[0];
           time_t time_now;
           uint8_t csum = 0;
 /* OOK_PULSE_MANCHESTER_ZEROBIT as it is manchester with a startbit 0.. */
             int j,i=0;
             for (j = 0; j < RST_BROWLEN+2; j++) {
   /* shift all the bits left 1 to align the fields */   
             for (i = j; i < RST_BROWLEN+1; i++) {
               uint8_t bits1 = br[i] << 1;
               uint8_t bits2 = (br[i+1] & 0x80) >> 7;
               bits1 |= bits2;
               br[i] = bits1;
             }
           }
 
            for (i = 1; i < RST_BROWLEN; i++) {
            /*  all bytes received, make sure checksum is okay */            	
              csum ^= br[i];
  	        /* Reverse all bits in a byte  */   
              br[i] = reverse8( br[i] );
            /* Decrypt raw received data byte */
              br[i] ^= br[i] << 1;
                }

 // fprintf (stdout, " checksum  =  %02x ", csum); 

if(csum !=0){
				return 0;
		}
                                 
  
//fprintf(stderr, "\n!!! ");
//bitbuffer_print(bitbuffer);

    time(&time_now);
    local_time_str(time_now, time_str);
  

 uint8_t channel = br[1] >> 5;
 uint8_t humidity = 10 * (br[6] >> 4) + (br[6] & 0x0f);
 float temperature = 100 * (br[5] & 0x0f) + 10 * (br[4] >> 4) + (br[4] & 0x0f);
	// temp is negative?
	if (!(br[5] & 0x80)) {
		temperature = -temperature;
	}

fprintf(stdout, "%s RST 02510 sensor\n", time_str);
fprintf(stdout, "Channel        = %d\n", channel);
fprintf(stdout, "Temperature    = %.1f C\n", temperature/10);
fprintf(stdout, "Humidity       = %d%%\n", humidity);

	return 1;
}

  .name           = "RST Temperature Sensor",
  .modulation     = OOK_PULSE_MANCHESTER_ZEROBIT,
  .short_limit    = 123,
  .long_limit     = 0,
  .reset_limit    = 375,
  .json_callback = &rst_weather_callback,
  .disabled      = 0,
  .demod_arg     = 0,

merbanan, 2018-12-03:
No signal samples available closing issue.

closed 2018-12-03 by merbanan