merbanan/rtl_433 #153: Auriol (LIDL) support

xes, 2015-08-27:
Do you think it would be possible to include the support for the Auriol (LIDL) sensors?

I have tried the pm-cz's version with an Auriol H13726 (AKA Ventus W155), an LD3091 and Z30914 (doorbell) and it is working pretty well.

H13726A http://www.lidl-service.com/static/31159141/75966_EN_FI_DA_RO.pdf
LD3091 http://www.lidl-service.com/static/15093932/61430_de_fr_it_nl.pdf
Z30914 http://www.lidl-service.com/static/33168761/75276_FR.pdf

I have used:
https://github.com/pm-cz/rtl_433

Auriol protocol for the H13726:
http://www.tfd.hu/tfdhu/files/wsprotocol/auriol_protocol_v20.pdf

Best regards,
Xes

merbanan, 2015-08-27:
If someone cleans up the code and sends a pull request I will accept it.

xes, 2015-08-28:
@merbanan  
Thank you for your reply.
I can't manage to adapt the pm-cz's Auriol sensors definition to the new rtl_433 structure... i hope someone would help.  
If needed, i can provide samples for all the Auriol and Silvercrest sensors previously listed.

vestom, 2015-08-29:
Yes, please add samples for these sensors to https://github.com/merbanan/rtl_433_tests
I might have a look :-)

merbanan, 2015-08-30:
I have added Auriol version 09/2012 test data. The signal is distance coded. AFAIK there is no new distance demodulator.

vestom, 2015-08-30:
The signal distance demodulator is called Pulse Position Modulation (PPM) decoder. It is the pulse_demod_ppm() function or OOK_PULSE_PPM_RAW modulation type.

I will try to make a stub function (rather that than having to convert it later :-). Hopefully someone else will work with decoding the data.

merbanan, 2015-08-30:
@vestom I changed the rubicson protocol to use OOK_PULSE_PPM_RAW and it started working again. It seems that the Auriol samples I've sent use the same protocol, something is just fishy with the timings.

vestom, 2015-08-30:
Ahh, I didn't catch it is the same as rubicson. It seems the timing is off though. These should work with both:
```
	.modulation		= OOK_PULSE_PPM_RAW,
	.short_limit	= (244+485)/2,		// Gaps:  Short 244, Long 485, Sync 1000
	.long_limit		= (485+1000)/2,		// Pulse: 125 (Initial pulse in each package is 97)
	.reset_limit	= 1200,				// Two initial pulses and a gap of 2280 is filtered out
```

vestom, 2015-08-30:
The rubicson driver filters out the data though as the second byte is 0 for Auriol:
```
Test mode active. Reading samples from file: /home/tov/projects/rtl_433_tests/tests/auriol/01/gfile001.data
Detected OOK package
bitbuffer:: Number of rows: 1 
[00] {1} 00 : 0
bitbuffer:: Number of rows: 13 
[00] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[01] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[02] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[03] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[04] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[05] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[06] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[07] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[08] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[09] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[10] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[11] {36} 81 00 f3 f0 00 : 10000001 00000000 11110011 11110000 0000
[12] {0} : 
Analyzing pulses...
Total number of pulses: 447
Pulse width distribution:
 [ 0] mean:  125 [101;127],	 count: 437
 [ 1] mean:   97 [95;100],	 count:  10
Gap width distribution:
 [ 0] mean:  125 [125;125],	 count:   1
 [ 1] mean: 2279 [2279;2279],	 count:   1
 [ 2] mean:  485 [484;487],	 count: 144
 [ 3] mean:  240 [237;243],	 count: 288
 [ 4] mean:  999 [978;1005],	 count:  12
Pulse period distribution:
 [ 0] mean:  245 [245;245],	 count:   1
 [ 1] mean: 2399 [2399;2399],	 count:   1
 [ 2] mean:  608 [580;613],	 count: 144
 [ 3] mean:  366 [362;369],	 count: 288
 [ 4] mean: 1124 [1104;1128],	 count:  12
Guessing modulation: No clue...

Test mode active. Reading samples from file: /home/tov/projects/rtl_433_tests/tests/rubicson/01/gfile004.data
Sensor temperature event:
protocol       = Rubicson/Auriol, 37 bits
rid            = 2c
temp           = 11.9
2d c0 77 79 b8
Detected OOK package
bitbuffer:: Number of rows: 12 
[00] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[01] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[02] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[03] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[04] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[05] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[06] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[07] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[08] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[09] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[10] {36} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 0111
[11] {37} 2d 80 ee f3 70 : 00101101 10000000 11101110 11110011 01110
Analyzing pulses...
Total number of pulses: 445
Pulse width distribution:
 [ 0] mean:  121 [115;126],	 count: 443
 [ 1] mean:  155 [155;155],	 count:   1
 [ 2] mean:   87 [87;87],	 count:   1
Gap width distribution:
 [ 0] mean:  244 [240;251],	 count: 193
 [ 1] mean:  488 [457;494],	 count: 240
 [ 2] mean:  978 [975;983],	 count:  11
Pulse period distribution:
 [ 0] mean:  366 [359;373],	 count: 193
 [ 1] mean:  610 [604;616],	 count: 240
 [ 2] mean: 1100 [1096;1106],	 count:  11
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 104, long_limit: 138, reset_limit: 984, demod_arg: 0
```

xes, 2015-09-01:
Just created a pull request to rtl_433_tests:
Added samples and details for Auriol H13726A,LD3091 and Silvercrest Z30914

xes, 2015-10-14:
Any hope for the LD3091?

merbanan, 2018-12-03:
The only available test signal decodes properly now. Closing issue.

closed 2018-12-03 by merbanan