merbanan/rtl_433 #158: Support for Auriol Z29536-TX

gentoolinux, 2015-08-30:
Please add support for the Auriol Z29536-TX sensor (weather station Z29536-RX, bought from Lidl, available under IAN 37966 and IAN 55982). You can find the manual at http://www.lidl-service.com/static/7928307/55982_EN_FI_SV_DA.pdf and a picture at http://offers.kd2.org/pics/00/7f/007fcebc333555e8256a4887afe288eda83698d4.jpg.

Using https://github.com/pm-cz/rtl_433 I am able to decode the temperature data (frequency 433.7925 MHz determined with Gqrx). The sensor is identified as "Remote Temp Transmitter" (see commit https://github.com/pm-cz/rtl_433/commit/7ea45d7482635c052596687e521d405c7b0142d0).
I have recorded some signals using "rtl_433 -f 433792500 -a -t" and uploaded them to https://github.com/gentoolinux/rtl_433_tests/tree/82de9daf28716d8c9aaa736224cdac3773f3c969/tests/auriol/02.
Most of the decoded signals show the correct temperature, e.g. gfile010.data:

Sensor temperature event:
protocol      = Remote Temp Transmitter
button        = 0
battery       = Low
temp          = 29.2
channel       = 0
rid           = 236
hrid          = ec
6e c1 24 60 00

However, sometimes the wrong protocol is detected. For example in gfile012.data:

Sensor temperature event:
protocol      = Prologue
button        = 0
battery       = Low
temp          = 56.6
humidity      = 0
channel       = 2
id            = 5
rid           = 236
hrid          = ec
5e c1 23 60 00
Sensor temperature event:
protocol      = Remote Temp Transmitter
button        = 0
battery       = Low
temp          = 29.1
channel       = 0
rid           = 236
hrid          = ec
5e c1 23 60 00

Or in gfile014.data:

Sensor temperature event:
protocol      = Prologue
button        = 0
battery       = Low
temp          = -176.5
humidity      = 0
channel       = 1
id            = 5
rid           = 236
hrid          = ec
5e c0 91 b0 00

The weather station also shows air pressure. It would be nice, to be able to decode that data, too.

merbanan, 2015-08-30:
Send me a pull request and I will merge it to the tests repository.

gentoolinux, 2015-08-30:
Done. Thanks for merging.

merbanan, 2015-08-30:
If you try now I think that temperature should be supported.

gentoolinux, 2015-08-31:
Thank you. Unfortunately it's not working. Most of the packets are not detected or wrongly decoded. For example I see messages like

AcuRite Rain Gauge Total Rain is 68.0mm
Raw Message: 02 20 88 00 00

or

Sensor temperature event:
protocol      = Prologue, 28 bits
button        = 0
battery       = Low
temp          = 8.6
humidity      = 0
channel       = 2
id            = 5
rid           = 236
hrid          = ec
Detected Acurite 5n1 sensor, 1 bits
Sensor temperature event:
protocol      = Prologue, 28 bits
button        = 0
battery       = Low
temp          = 8.6
humidity      = 0
channel       = 2
id            = 5
rid           = 236
hrid          = ec
Sensor temperature event:
protocol      = Prologue, 28 bits
button        = 0
battery       = Low
temp          = 8.6
humidity      = 0
channel       = 2
id            = 5
rid           = 236
hrid          = ec
Detected Acurite 5n1 sensor, 1 bits
Detected Acurite 5n1 sensor, 0 bits
Sensor temperature event:
protocol      = Prologue, 28 bits
button        = 0
battery       = Low
temp          = 8.6
humidity      = 0
channel       = 2
id            = 5
rid           = 236
hrid          = ec

Here's some output created by "rtl_433 -f 433792500 -a":

Tuned to 433792500 Hz.
*** signal_start = 5465256, signal_end = 5722084
signal_len = 256828,  pulses = 1159
Iteration 1. t: 123    min: 6 (796)    max: 240 (204)    delta 137
Iteration 2. t: 123    min: 6 (796)    max: 240 (204)    delta 0
Pulse coding: Short pulse length 6 - Long pulse length 240

Short distance: 56, long distance: 864, packet distance: 2103

p_limit: 123
bitbuffer:: Number of rows: 8 
[00] {2} 80 : 10
[01] {142} a0 82 22 00 20 10 18 18 24 03 82 c1 04 10 88 d0 08 00 
[02] {163} c0 09 04 c8 10 22 42 00 48 02 0c 01 04 00 88 40 22 04 01 14 00 
[03] {139} d0 88 08 20 81 05 00 87 00 2c 94 22 03 80 01 08 10 00 
[04] {162} 81 08 24 04 2d 08 02 81 02 01 10 c2 01 10 02 00 10 48 81 10 00 
[05] {124} 81 1d 02 20 84 98 82 8b 11 08 11 22 01 01 04 00 
[06] {125} 89 04 02 10 92 84 a4 08 16 8a 12 00 40 a0 10 60 
[07] {143} 81 00 08 10 41 91 10 46 80 8c 50 80 40 8c 02 80 20 84 
*** signal_start = 14113581, signal_end = 14370407
signal_len = 256826,  pulses = 1148
Iteration 1. t: 123    min: 7 (794)    max: 240 (206)    delta 169
Iteration 2. t: 123    min: 7 (794)    max: 240 (206)    delta 0
Pulse coding: Short pulse length 7 - Long pulse length 240

Short distance: 55, long distance: 876, packet distance: 2159

p_limit: 123
bitbuffer:: Number of rows: 9 
[00] {1} 80 : 1
[01] {129} 94 02 05 90 0e 04 44 20 41 41 08 82 06 88 02 42 00 
[02] {129} 80 88 c0 94 10 48 41 00 48 03 42 95 84 02 08 81 00 
[03] {167} a0 a0 40 40 40 82 a0 08 01 08 44 08 01 21 08 00 40 68 20 20 42 
[04] {137} 88 22 81 14 50 20 14 02 48 42 08 51 10 04 40 a4 08 00 
[05] {130} c8 10 21 00 14 89 00 62 10 24 4a 0c 50 40 81 24 00 
[06] {158} 80 42 08 02 91 10 50 00 80 40 98 10 08 41 10 10 20 91 00 d0 
[07] {143} 90 81 10 42 41 02 20 42 09 34 08 82 24 41 40 82 02 00 
[08] {6} 88 : 100010

merbanan, 2018-12-03:
This should be supported now. Closing issue.

closed 2018-12-03 by merbanan