merbanan/rtl_433 #166: New device (most likely a weather station) at 433MHz

viraptor, 2015-09-05:
I'm trying to receive data from my weather station (Digitech XC0348). Documentation confirms it sends data at 433MHz. From running analyzer for some time, I believe I get consistent results. Skipping duplicate entries (many), here are some unique rows:

Common header: (always 88 pulses)
```
Detected OOK package
Analyzing pulses...
Total number of pulses: 88
Pulse width distribution:
 [ 0] mean:  134 [133;135],	 count:  32
 [ 1] mean:  379 [378;381],	 count:  56
Gap width distribution:
 [ 0] mean:  246 [245;249],	 count:  87
Pulse period distribution:
 [ 0] mean:  381 [380;383],	 count:  31
 [ 1] mean:  625 [624;627],	 count:  56
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 256, long_limit: 250, reset_limit: 250, demod_arg: 0
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 1 
```

Data:
```
[00] {88} ff a8 51 f2 52 00 01 04 5d 06 0e 
[00] {88} ff a8 51 f2 52 00 01 04 5d 02 ca 
[00] {88} ff a8 51 f2 52 00 01 04 5d 00 a8 
[00] {88} ff a8 51 f2 52 01 02 04 5d 02 85 
[00] {88} ff a8 51 f2 52 00 00 04 5d 00 33
```
Current conditions: 9.6C, 83% hum, 0 wind, 0 rain.

viraptor, 2015-09-05:
It is the weather station - I just noticed the panel that came with it blinks the receiving icon exactly when rtl_433 reports new packet.

viraptor, 2015-09-05:
And some early data analysis, all big endian:
```
[00] {88} ff a8 XX XX YY ZZ 01 04 5d UU UU
```
`XX XX`: temperature, likely in 0.1⁰C steps (`51 e7` == 8.7⁰C, `51 ef` == 9.5⁰C)
`YY`: percent in a single byte (for example `54` == 84%)
`ZZ`: wind speed (`00` == 0, `01` == 1.1km/s, ...)
`UU UU`: wind direction: `00 00` is N, `02..` is NE, `04..` is E, etc. up to `0f ff` is seems (highest I got was `0e 08` for NW)

viraptor, 2015-09-26:
Found out the checksum. It's a CRC with polynomial 0x31 and initial value 0xff. Won't be able to update the code for a week or two, so this is just to note for later.

merbanan, 2015-09-26:
While not related to this sensor I have a checksum/crc that I seem to not be able to figure out.

https://github.com/merbanan/rtl_433_tests/blob/master/tests/rubicson/01/readings3.txt

ea 80 b0 f3 e = 17.6
ea 80 b1 fc a = 17.7

This reading is just one bit of difference of the payload. But the last 8 bits differ greatly. Do you have some method or software that can crack this ?

viraptor, 2015-09-27:
Check the project I mentioned (reveng on sourceforge). That should do the
trick.

rct, 2015-10-08:
@viraptor Thanks for the tip on reveng.  It was able to figure out the DSC contact CRC I was struggling with for a while.  PyCRC, (pycrc.org) was also helpful.  The stumbling block was that it was a "little endian" CRC.

@merbanan looks like your CRC might be polynomial 0x93, initial value = 0x4f.

util.c/crc8 doesn't currently take a parameter for the initial value but that's easy enough to change.

rct, 2015-10-08:
@merbanan btw, the reveng cmdline that found that was simply:
reveng -w 8 -s ea80b0f30e ea80b1fc0a

merbanan, 2015-10-09:
@rct I'll have to try that. But when I try with reveng and the same command line I get other values.

merbanan, 2018-12-03:
Decoder merged, closing issue.

closed 2018-12-03 by merbanan