merbanan/rtl_433 #168: Doorbell 13bits pattern ?

fcartegnie, 2015-09-11:
From a simple doorbell.
I'd also like to figure the modulation for tx.

Tuned to 433920000 Hz.
*** signal_start = 737916, signal_end = 1018802
signal_len = 280886,  pulses = 636
Iteration 1. t: 107    min: 70 (343)    max: 145 (293)    delta 18
Iteration 2. t: 107    min: 70 (343)    max: 145 (293)    delta 0
Pulse coding: Short pulse length 70 - Long pulse length 145

Short distance: 78, long distance: 153, packet distance: 2619

p_limit: 107
bitbuffer:: Number of rows: 25 
[00] {13} 03 b8 : 00000011 10111
[01] {13} 03 b8 : 00000011 10111
[02] {13} 03 b8 : 00000011 10111
[03] {13} 03 b8 : 00000011 10111
[04] {13} 03 b8 : 00000011 10111
[05] {13} 03 b8 : 00000011 10111
[06] {13} 03 b8 : 00000011 10111
[07] {13} 03 b8 : 00000011 10111
[08] {13} 03 b8 : 00000011 10111
[09] {13} 03 b8 : 00000011 10111
[10] {13} 03 b8 : 00000011 10111
[11] {13} 03 b8 : 00000011 10111
[12] {13} 03 b8 : 00000011 10111
[13] {13} 03 b8 : 00000011 10111
[14] {13} 03 b8 : 00000011 10111
[15] {13} 03 b8 : 00000011 10111
[16] {13} 03 b8 : 00000011 10111
[17] {13} 03 b8 : 00000011 10111
[18] {13} 03 b8 : 00000011 10111
[19] {13} 03 b8 : 00000011 10111
[20] {13} 03 b8 : 00000011 10111
[21] {13} 03 b8 : 00000011 10111
[22] {13} 03 b8 : 00000011 10111
[23] {13} 03 b8 : 00000011 10111
[24] {324} 03 b8 1d c0 ee 07 70 3b 81 dc 0e e0 77 03 b8 1d c0 ee 07 70 3b 81 dc 0e e0 77 03 b8 1d c0 ee 07 70 3b 81 dc 0e e0 77 03 b0 
*** signal_start = 1850205, signal_end = 2136771
signal_len = 286566,  pulses = 650
Iteration 1. t: 103    min: 66 (350)    max: 141 (300)    delta 50
Iteration 2. t: 103    min: 66 (350)    max: 141 (300)    delta 0
Pulse coding: Short pulse length 66 - Long pulse length 141

Short distance: 82, long distance: 157, packet distance: 2623

p_limit: 103
bitbuffer:: Number of rows: 25 
[00] {13} 03 b8 : 00000011 10111
[01] {13} 03 b8 : 00000011 10111
[02] {13} 03 b8 : 00000011 10111
[03] {13} 03 b8 : 00000011 10111
[04] {13} 03 b8 : 00000011 10111
[05] {13} 03 b8 : 00000011 10111
[06] {13} 03 b8 : 00000011 10111
[07] {13} 03 b8 : 00000011 10111
[08] {13} 03 b8 : 00000011 10111
[09] {13} 03 b8 : 00000011 10111
[10] {13} 03 b8 : 00000011 10111
[11] {13} 03 b8 : 00000011 10111
[12] {13} 03 b8 : 00000011 10111
[13] {13} 03 b8 : 00000011 10111
[14] {13} 03 b8 : 00000011 10111
[15] {13} 03 b8 : 00000011 10111
[16] {13} 03 b8 : 00000011 10111
[17] {13} 03 b8 : 00000011 10111
[18] {13} 03 b8 : 00000011 10111
[19] {13} 03 b8 : 00000011 10111
[20] {13} 03 b8 : 00000011 10111
[21] {13} 03 b8 : 00000011 10111
[22] {13} 03 b8 : 00000011 10111
[23] {13} 03 b8 : 00000011 10111
[24] {338} 03 b8 1d c0 ee 07 70 3b 81 dc 0e e0 77 03 b8 1d c0 ee 07 70 3b 81 dc 0e e0 77 03 b8 1d c0 ee 07 70 3b 81 dc 0e e0 77 03 b8 1d c0 
^CSignal caught, exiting!

merbanan, 2015-09-30:
This is a pwm modulated signal. Multiply the values with 4 to get microsecs for the duration. Pulse coding: Short pulse length 70 - Long pulse length 145

merbanan, 2018-12-03:
These types of devices can be handled by the flex decoder. Closing issue.

closed 2018-12-03 by merbanan