merbanan/rtl_433 #169: Acurite 5-n-1 rain gauge reading incorrect (all 0)

rct, 2015-09-11:
The Acurite 5-n-1 rain gauge decoding isn't working.  I get values of all zeros. 

I made some -a -t recordings during a heavy rain.   The wind speed and direction all seem within plausible range.  

I will start digging into it, but wanted to check if anyone else is seeing incorrect or correct readings.

ygator, 2015-10-06:
I believe I get false AcuRite Rain Gauge messages from my Oregon Scientific THC138 Temperature Sensors.  They are version 1.0 protocol which rtl_433 does not decode.

rct, 2015-11-07:
Closing this issue.  I was incorrect.  Due to signal level differences, my 5n1 rain messages weren't being decoded and I was getting false positives with bizarre results from the  Acurite 896 range gauge device decodings.

@ygator, the acurite 896 rain gauge decoder should be tightened up to not give false reports, but there aren't any test signals in the rtl_433_tests repository to work with.

closed 2015-11-07 by rct