merbanan/rtl_433 #177: VirtualWire library support

emilburzo, 2015-10-06:
Hi,

I've made some DIY arduino wireless temperature modules with some cheap[1] 433 MHz TX.

The library[2] I am using is open source, and you can send any string with it.

Everything is working fine, but I would like to get rid of the RX module and just use this awesome project instead.

Is it possible to intercept variable length strings and print them to console?

[1] https://eud.dx.com/product/433mhz-wireless-transmitter-module-superregeneration-for-arduino-green-844149254
[2] http://www.airspayce.com/mikem/arduino/VirtualWire/

merbanan, 2015-10-06:
Try running rtl_433 -a or rtl_433 -A. That should invoke analyse mode that might be able to decode the data automatically. After that either hack it or implement a protocol decoder.

emilburzo, 2015-10-06:
I have tried running analyse mode without much luck.

I think this is the relevant encoding code from the library.
```
// Wait until transmitter is available and encode and queue the message
// into vw_tx_buf
// The message is raw bytes, with no packet structure imposed
// It is transmitted preceded a byte count and followed by 2 FCS bytes
uint8_t vw_send(uint8_t* buf, uint8_t len)
{
    uint8_t i;
    uint8_t index = 0;
    uint16_t crc = 0xffff;
    uint8_t *p = vw_tx_buf + VW_HEADER_LEN; // start of the message area
    uint8_t count = len + 3; // Added byte count and FCS to get total number of bytes

    if (len > VW_MAX_PAYLOAD)
	return false;

    // Wait for transmitter to become available
    vw_wait_tx();

    // Encode the message length
    crc = _crc_ccitt_update(crc, count);
    p[index++] = symbols[count >> 4];
    p[index++] = symbols[count & 0xf];

    // Encode the message into 6 bit symbols. Each byte is converted into 
    // 2 6-bit symbols, high nybble first, low nybble second
    for (i = 0; i < len; i++)
    {
	crc = _crc_ccitt_update(crc, buf[i]);
	p[index++] = symbols[buf[i] >> 4];
	p[index++] = symbols[buf[i] & 0xf];
    }

    // Append the fcs, 16 bits before encoding (4 6-bit symbols after encoding)
    // Caution: VW expects the _ones_complement_ of the CCITT CRC-16 as the FCS
    // VW sends FCS as low byte then hi byte
    crc = ~crc;
    p[index++] = symbols[(crc >> 4)  & 0xf];
    p[index++] = symbols[crc & 0xf];
    p[index++] = symbols[(crc >> 12) & 0xf];
    p[index++] = symbols[(crc >> 8)  & 0xf];

    // Total number of 6-bit symbols to send
    vw_tx_len = index + VW_HEADER_LEN;

    // Start the low level interrupt handler sending symbols
    vw_tx_start();

    return true;
}
```

But it's a bit over my head.

merbanan, 2015-10-06:
I don't get the modulation used. But try with lowering the detection limit. Use -l 4000 or so. You should be able to detect signals being transmitted.

emilburzo, 2015-10-06:
OK, I got some samples:

**1,21.70,68.50,4344,0**

```
[00] {21} 00 00 60 : 00000000 00000000 01100000 
[01] {89} 1e 08 f9 07 08 3c 41 c2 00 42 60 00 : 00011110 00001000 11111001 00000111 00001000 00111100 01000001 11000010 00000000 01000010 01100000 00000000 
[02] {7} 80 : 10000000 
```

**2,21.90,66.10**
```
[00] {17} 00 00 00 : 00000000 00000000 00000000 
[01] {62} 94 58 23 e1 0e 10 9c c0 : 10010100 01011000 00100011 11100001 00001110 00010000 10011100 11000000 
[02] {7} 34 : 00110100 
```

**1,21.70,68.20,4344,0**
```
[00] {22} 00 00 24 : 00000000 00000000 00100100 
[01] {98} 88 00 44 80 00 12 00 00 00 00 00 02 00 : 10001000 00000000 01000100 10000000 00000000 00010010 00000000 00000000 00000000 00000000 00000000 00000010 00000000 
```

**2,21.90,65.70**
```
[00] {17} 00 00 00 : 00000000 00000000 00000000 
[01] {56} 94 58 23 e1 0e 10 5c 00 : 10010100 01011000 00100011 11100001 00001110 00010000 01011100 00000000 
[02] {5} 00 : 00000000 
[03] {10} 00 40 : 00000000 01000000 
```

**1,21.70,68.30,4344,0**
```
[00] {21} 00 00 60 : 00000000 00000000 01100000 
[01] {98} 1e 08 f9 07 08 3c 00 e1 00 21 30 26 40 : 00011110 00001000 11111001 00000111 00001000 00111100 00000000 11100001 00000000 00100001 00110000 00100110 01000000 
```

merbanan, 2015-10-06:
Ok, rtl_433 can see the signal. But you would need to write your own protocol decoder.

faspina, 2018-08-29:
I know this thread is old I am interested in this exact same thing. Mainly to consolidate some 433 rx devies to one device.  Did you get anywhere

zuckschwerdt, 2018-08-29:
Can you record some samples? Shouldn't be too hard to implement a decoder.

faspina, 2018-08-29:
I can. The thing is I wrote both side of the virualwire code. The send and the receive on the other end. I want to replace the receive on the other end.  I can change the send side to be whatever I can program the arduino to use that will work with RTL_433

zuckschwerdt, 2018-08-29:
Is "VirtualWire" a universal protocol with other applications (i.e. is it used anywhere else)?
Otherwise maybe look into the RadioHead protocol. We have that already, see https://github.com/merbanan/rtl_433/blob/master/src/devices/radiohead_ask.c

faspina, 2018-08-29:
I can do that. THanks!!!

faspina, 2018-08-29:
Quick update .Hooked up my rx and manage to capture this from my virtualwire device.  I know it was it because it repeated 3 time which is what I programmed it to do. So its reading virtualwire from tx as radiohead 

2018-08-29 16:31:22 : RadioHead ASK
        Data len:  4
        To:  48
        From:  49
        Id:  58
        Flags:  32
        Payload:  [56, 53, 46, 51]
        Integrity:  CRC

56,53,46,51  ascii char are "85.1" which is what the temp reading is.

zuckschwerdt, 2018-08-31:
Ok, as far as I can see VirtualWire is the predecessor of RadioHead and obsolete now. https://www.airspayce.com/mikem/arduino/VirtualWire/
If RadioHead works for you (changes should be minimal) then there is no need to care about VirtualWire anymore.

merbanan, 2018-12-03:
Closing issue. The Flex decoder should be possible to use for cases lite this. Closing issue.

closed 2018-12-03 by merbanan