merbanan/rtl_433 #182: Oregon Scientific Protocol V1.0 Temperature Sensor

ygator, 2015-10-07:
Can someone help with adding this Protocol to rtl_433?  Thanks.

I believe this is the capture of the signal.
https://drive.google.com/file/d/0B5D4Lm5bUBiQd01KXzg1T1NfMlU/view

The transmission is repeated.

0198821b

Nibble 0 = CCBS
CC = Channel - 1
B = Low Battery
S = Temperature Sign 0=+ 1=-

Nibble 1 Nibble 4 = Power On ID

Nibble 5 Nibble 2 . Nibble 3 = BCD Temperature in Celsius

Nibble 6 Nibble 7 = Byte 3 = Checksum
Checksum equals Sum of Bytes 0 1 2

So this transmission is
Channel 1
Battery OK
Power On ID = 18
Temperature +29.8C

ygator, 2015-10-08:
-A gives

Test mode active. Reading samples from file: /tmp/0198821b.data
Detected OOK package
wt450_callback: wrong size of bit per row 0
Analyzing pulses...
Total number of pulses: 40
Pulse width distribution:
 [ 0] mean:  431 [429;433],      count:  34
 [ 1] mean: 1439 [1439;1439],    count:   1
 [ 2] mean:  798 [798;799],      count:   5
Gap width distribution:
 [ 0] mean:  300 [299;303],      count:  31
 [ 1] mean: 1182 [1058;1307],    count:   2
 [ 2] mean:  667 [666;669],      count:   6
Pulse period distribution:
 [ 0] mean:  732 [730;735],      count:  28
 [ 1] mean: 1472 [1465;1487],    count:   3
 [ 2] mean: 2746 [2746;2746],    count:   1
 [ 3] mean: 1099 [1098;1100],    count:   7
Guessing modulation: Pulse Width Modulation with startbit/delimiter
Attempting demodulation... short_limit: 614, long_limit: 1118, reset_limit: 1308, demod_arg: 2
pulse_demod_pwm_ternary(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {12} 00 00 : 00000000 0000
[01] {27} 00 26 12 00 : 00000000 00100110 00010010 000

merbanan, 2015-11-24:
Add some samples via a pull request to the rtl_433_tests repository.

ygator, 2015-12-12:
Sample added

enavarro222, 2015-12-12:
Here is a pretty good description of V1 (and 2 and 3) protocol.
http://wmrx00.sourceforge.net/Arduino/OregonScientific-RF-Protocols.pdf

If it may help...

ygator, 2016-02-09:
I'm closing this as I have this done.  I will commit within a week or two.

closed 2016-02-09 by ygator