merbanan/rtl_433 #195: Support for EverFlourish EMW203T

MartijnBraam, 2015-10-28:
I have a EverFlourish EMW203T lighting remote. It is a simple 6 button remote that controls 3 outlet receivers.

If I set the correct frequency (it says 433.92Mhz on the back of the remote, It actually sends at 433.966) then I can receive 2 of the buttons with the Brennstuhl RCS 2044 module (with the wrong labeling)

If i run with `rtl_433 -f 433950000 -a` then I can receive all buttons fine. Every keypress creates at least 6 rows and repeats until released. The buttons are labeled 1, 2 and 3 (on and off) and on the back I can choose channel A, B, C or D.

A few example rows:
```
00010101 00010101 01010111 0 - A 1 On
00010101 00010101 01010100 0 - A 1 Off
00010101 01000101 01010111 0 - A 2 On
00010101 01000101 01010100 0 - A 2 Off
00010101 01010001 01010111 0 - A 3 On
00010101 01010001 01010100 0 - A 3 Off

00010101 00010101 01010111 0 - A 1 On
01000101 00010101 01010111 0 - B 1 On
01010001 00010101 01010111 0 - C 1 On
01010100 00010101 01010111 0 - D 1 On
```

merbanan, 2015-11-01:
Add some samples to rtl_433_tests.

veurits, 2018-08-25:
I got the older everflourish EMW200T , protocol seems to be the same.

Found more info about the signal in : https://arduinodiy.wordpress.com/tag/everflourish/

zuckschwerdt, 2018-08-25:
with the 24 bits it looks like some x1527 type. Use the flex decoder – that's the supported way for push-button remotes and alike.

closed 2018-08-25 by zuckschwerdt