merbanan/rtl_433 #202: Add support for Acurite 986 Refrigerator/Freezer Thermometer

rct, 2015-11-21:
Manufacturer link: http://www.acurite.com/kitchen-gadgets/digital-freezer-refrigerator-thermometer-with-temperature-alarms-00986.html

Two temperature sensor transmitters includes a receiver display.

rct, 2015-11-21:
PR #203 implements this, however, the existing PPM decoder looses the last bit in the CRC byte.  The data is sent twice.  If the CRC is even both copies will decode correctly.  If odd, one copy will fail.

This could be improved by implementing PPM demodulators that are similar to the newer PWM demods, ternary and precise.   There are some short and long start/sync pulses that should ideally not be included in the data (should be in separate rows in the bitbuffer.)

This is working well enough for now. If I get time I'll look at adding the new PPM demods.

Cabalist, 2018-02-28:
I realize this issue is very old but I was curious if the new PPM demods were implemented?

zuckschwerdt, 2018-02-28:
PWM has be updated and PPM should be next. I'll look into it.

Cabalist, 2018-02-28:
Thanks!  Do you have a donate button or anything I can send a beer to?

zuckschwerdt, 2018-02-28:
Thanks for the kind offer. I guess it wouldn't be fair to all the other contributors.
But perhaps we'll do a fundrasing for some bigger feature some day.
Cheers!

merbanan, 2018-02-28:
Well it would be nice to gather a war chest for bigger feature requests.

Cabalist, 2018-02-28:
Let me know!  This is neat work and I’d like to show some appreciation. :)

rct, 2018-03-01:
I can't believe it's been over two years.  I never got back to this because the Acurite 986 has been working well enough for me as is even with the occasional lost bit.  I think at the time vestom was doing some work on the demodulators.  I've been meaning to convert it to the structured output (JSON, etc.).

trlafleur, 2018-03-31:
I would also like to see this fix, but its not high priority...

trlafleur, 2018-04-05:
I have done some work on decode this sensor....

This is the data that you will see in the capture below. (Note: bits are LSB first)
0Xd2, 0X96, 0X1e, 0X00, 0Xb4  -->    11010010, 10010110, 11110, 0, 10110100
Acurite 986 sensor: 0X6978 - 1R: 75 F

1) Sometime it sends one block of data, sometime two
2) 1st sync pulse is ~1700us most of the time, but it stretches to ~2300us on occasion.
3) Depending on last bit in CRC, if the block is send twice, the last bit is missing only in the 2nd block 
[image]
(see timing capture)
4) device is NOT sending pre-bit to set AGC in receiver as noted in source code

thanks for a great product...

trlafleur, 2018-04-06:
> Where is the logic analyzer connected -- to the 986 transmitter or the
> receiver that is your "standalone"?
>
> So if I'm understanding you correctly, you are saying the problem with the
> missing bit that makes the CRC fail is with the device itself and not the
> demodulator?

logic analyzer is connected to the receiver bitstream...

Yes, looks to me that the device is the issue... 1st block is ALWAYS
correct, 2nd block fails with the end of the last bit missing.....

I have ch2 connected to a bit on the processor to show me if I have a
valid sync, ch3, to edge detect decoder, ch4, to end of valid frame..

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

rct, 2018-04-22:
I think PR #709 may things well enough for now. The 986 is now doing JSON/using data_make.

I've added my hack that retries the CRC with the missing one bit - if the normal CRC check fails.   I was running this for a while on an old branch that I had saved with the pre-data_make decoders that matched my scripts.

There are some other sanity checks, so I think the chance of letting a corrupt message through is pretty low.

If someone does want to look further into whether it is a device or demodulator issue, I believe both of the samples in rtl_433_tests are missing the final bit on the second repeat.  After adding in the hack both produce two copies correctly.

merbanan, 2018-04-23:
Merged the code, closing.

closed 2018-04-23 by merbanan

trlafleur, 2018-04-23:
I have tested and it works just fine.... Thanks

lagomorph, 2022-12-29:
Just curious if anyone here would have some insight... I'm using a flipperzero to send a manufactured signal to an Acurite-986 and rtl_433 receives and decodes it just fine, even from outside the house. I can never get the 986 display to update with my signal though even though I can't detect any differences between from the ones sent by the sensors (except mine is a little cleaner). :)

zuckschwerdt, 2022-12-29:
Upload a capture of your synthetic signal as zip here and we'll take a look ;) Do you include the longer warmup/sync?

gdt, 2022-12-29:
I have a 986 and it mostly decodes with rtl_433, FWIW.

lagomorph, 2022-12-29:
@zuckschwerdt You mean the two short pulses and 4 long pulses? If so, yeah. I used the original received transmission from an actual 986. When sending that didn't update the display I cleaned it up using the timings in the source here. That made it a tiny bit longer than the real device transmission but it looks very close to me. I've decoded a bunch of these manually and recomputed the CRCs.

[signalcap.zip](https://github.com/merbanan/rtl_433/files/10320984/signalcap.zip)

lagomorph, 2022-12-29:
> I have a 986 and it mostly decodes with rtl_433, FWIW.

rtl_433 is very reliable at decoding my real sensors too. They each appear to transmit every two minutes.