merbanan/rtl_433 #211: Getting values only once and then exit?

hanez, 2015-11-27:
Hi,

is there a way to get the values only once and then exit the program? I ask because I want to use your program within a shell script and grep some data after execution for later usage in the script. I tried a lot of stuff and read parts of the code but it seems that your program wants to run forever until killing it using Ctrl-C... am I right? If it is not possible could you give me a hint where in the code this could be implemented? I will then try to write a patch...

Anyway, thank you for this wonderful piece of software! I have a lot of fun reading sensor values from all devices from all my neighbours... ;)

Regards
Johannes

pandysong, 2015-11-28:
You may tell which part of the data your script needs to grep.
If for example, you want to grep the data from "rtl_433.exe -a", you may want to add exit(0) in pwm_analyze(...)

merbanan, 2018-12-06:
We now have the -E option that may be what you asked for. Closing issue.

closed 2018-12-06 by merbanan