merbanan/rtl_433 #218: Watchman Sonic / Apollo Ultrasonic oil monitor

dwmw2, 2015-12-04:
These two are basically the same device. The receiver uses a Si4320, the same as the popular RFM01/RFM11/RFM12/etc FSK transceivers. The framing, FSK and Manchester encoding are all handled by the chip and so probably the same as plenty of other devices.

I've got a proof-of-concept hack which takes output from rtl_fm and decodes the packets: http://david.woodhou.se/watchman-demod.c

Obviously having got this far, I need to replace my horrid hack with some proper signal processing, and adding it as a supported device to rtl_433 seems like the right approach. I could do with some guidance with that task though, please.

I have two devices to test with. One appears to transmit at 433906MHz ± 34kHz, while the other is 433896MHz ± 29kHz. Looking at the Si4320/RFC01 datasheets, it seems the bandwidth is supposed to be 67kHz.

Here's the waveform I get from rtl_fm: http://david.woodhou.se/watchman.png

There are three high periods, followed by three low periods. After that, a 0 bit is indicated by high-then-low, while 1 is indicated by low-then-high. After the 64th bit, there are two more periods remaining at the same level (so remaining low after a final 0 bit, or remaining high after a final 1 bit).

The period is about 1ms, and the complete transmission 136ms (6ms preamble, 128ms data, 2ms postamble).

This results in 8 bytes of data. The final byte is a 1Wire CRC8, and the actual sensor data (a depth reading from the ultrasonic data) is in the byte before that.

cf. http://openenergymonitor.org/emon/node/10107

So, where do I start? :)

dwmw2, 2015-12-04:
Test capture from rtl_433 uploaded to http://david.woodhou.se/rtl433_testcap_watchman.cap

merbanan, 2015-12-04:
Nice domain :) Anyway pretty please document the device(s), take some signal samples (if they actually are useful, in your case we might need a higher sample rate) and send a pull request to https://github.com/merbanan/rtl_433_tests.

Read the rtl_433 mailing list the thread "Lutron Motion Detection". Hopefully everything will be clear after that. Most likely the actual fsk demodulation should be supported and that you only need to write the actual packet parsing code. But start with this and raise issues on the bug tracker or the mailing list.

dwmw2, 2015-12-05:
For a packet which ends with a 1 bit, I end up with this:

```
Breakpoint 1, pulse_demod_manchester_framed (pulses=0x7ffff554a0dc, device=0x7ffffffb7dc0) at /home/dwmw2/git/rtl_433/src/pulse_demod.c:282
282		int events = 0;
(gdb) p *pulses
$8 = {num_pulses = 53, pulse = {766, 258, 258, 518, 519, 258, 258, 262, 518, 258, 258, 262, 261, 524, 519, 258, 519, 258, 257, 258, 258, 257, 259, 262, 261, 262, 266, 
    519, 258, 258, 258, 258, 258, 258, 258, 519, 258, 258, 518, 259, 258, 258, 258, 262, 262, 519, 266, 262, 261, 519, 519, 262, 779, 0 <repeats 947 times>}, gap = {
    774, 259, 515, 516, 258, 259, 520, 256, 259, 258, 516, 255, 256, 515, 259, 515, 258, 259, 264, 259, 259, 259, 514, 256, 256, 255, 257, 258, 258, 259, 259, 259, 
    258, 263, 516, 258, 259, 515, 259, 263, 258, 259, 515, 255, 256, 515, 256, 256, 256, 515, 515, 255, 268, 0 <repeats 947 times>}}
```

This seems entirely reasonable. I can cope with copying the existing pulse_demod_manchester_zerobit() function and making it handle the preamble and postamble.

For a packet which ends with a 0 bit, however, I end up with something less reasonable:
```
$10 = {num_pulses = 53, pulse = {768, 258, 258, 519, 518, 258, 258, 261, 519, 259, 257, 261, 262, 523, 519, 258, 519, 257, 258, 258, 258, 258, 258, 261, 262, 262, 267, 
    519, 258, 259, 258, 258, 257, 258, 258, 519, 257, 259, 518, 258, 258, 519, 257, 258, 519, 265, 262, 261, 262, 519, 258, 518, 6, 0 <repeats 947 times>}, gap = {774, 
    258, 516, 515, 259, 259, 520, 256, 258, 258, 516, 256, 256, 515, 259, 515, 259, 259, 264, 258, 259, 258, 516, 256, 255, 256, 255, 258, 259, 258, 259, 258, 260, 
    263, 515, 259, 259, 515, 259, 263, 515, 259, 260, 515, 515, 257, 255, 257, 255, 259, 515, 771, 266, 0 <repeats 947 times>}}
```
I *consistently* seem to have a spurious tiny pulse reported at the end of the packet. Is that something I should expect and deal with, or is it indicative of a problem elsewhere?

Currently playing with something along these lines...
```
--- a/src/pulse_detect.c
+++ b/src/pulse_detect.c
@@ -400,7 +400,18 @@ void pulse_analyzer(pulse_data_t *data)
                device.short_limit      = min(hist_pulses.bins[0].mean, hist_pulses.bins[1].mean);              // Assume shortest pulse is half period
                device.long_limit       = 0; // Not used
                device.reset_limit      = hist_gaps.bins[hist_gaps.bins_count-1].max + 1;                               // Set limit above biggest gap
+       } else if(hist_pulses.bins_count == 3 && hist_pulses.bins[2].count == 2 &&
+                 hist_pulses.bins[2].mean > hist_pulses.bins[1].mean && hist_pulses.bins[2].mean > hist_pulses.bins[0].mean &&
+                 hist_gaps.bins_count == 3 && hist_gaps.bins[2].count == 1 &&
+                 hist_gaps.bins[2].mean > hist_gaps.bins[1].mean && hist_gaps.bins[2].mean > hist_gaps.bins[0].mean) {
+       rfm:
+               fprintf(stderr, "RFM Framed Manchester coding\n");
+               device.modulation       = OOK_PULSE_MANCHESTER_FRAMED;
+               device.short_limit      = min(hist_pulses.bins[0].mean, hist_pulses.bins[1].mean);              // Assume shortest pulse is half period
+               device.long_limit       = 0; // Not used
+               device.reset_limit      = max(hist_gaps.bins[0].max, hist_pulses.bins[0].max) + 1;                              // Set limit above biggest gap
        } else if(hist_pulses.bins_count == 3) {
+               *(int *)0=0;
                fprintf(stderr, "Pulse Width Modulation with startbit/delimiter\n");
                device.modulation       = OOK_PULSE_PWM_TERNARY;
                device.short_limit      = (hist_pulses.bins[0].mean + hist_pulses.bins[1].mean) / 2;    // Set limit between two lowest pulse widths
```

merbanan, 2015-12-05:
I suspect that you need a higher sample rate for the demodulator to work properly. Read this and post a screen shot of the captured signal.

https://groups.google.com/forum/m/#!topic/rtl_433/1v-s2KXmuTs

merbanan, 2015-12-05:
Demodulated signal to be exact.

dwmw2, 2015-12-05:
I think the sample rate is fine — it's already 250 samples per half-bit, which is an order of magnitude higher than the sample rate that was working with my proof-of-concept decoder hack.

I'm working on a pull request for rtl_433_tests now. I have a set of gfile-xxx.data files. I ran each one through 'rtl_433 -m 2' to get an equivalent fm-demodulated version, and *(after hacking my decoder to expect ten times the hardcoded sample rate — did I say it was a dirty hack?)* I can decode them just fine.

merbanan, 2015-12-05:
Ok, if the modulation is fine then you need to get a proper pulse table, then apply the manchester decoding on that and after that send it to a protocol decoder. As simple as that :)

dwmw2, 2015-12-05:
Ok, so we can look at the data now after pull request 50.

If we start with gfile011.data, I end up with this. Each period is 250 samples. The signal starts with three periods high, three periods low, before going into the Manchester encoding. And ends with three periods high, because the last data bit was a 1 *(transmitted as low then high, and repeating that last 'high' for a further two half-bit periods is the end-of-frame marker)*.
```
Test mode active. Reading samples from file: ../rtl_433_tests/tests/oil_watchman/01/gfile011.data
Detected FSK package
Analyzing pulses...
Total number of pulses: 52
Pulse width distribution:
 [ 0] mean:  772 [767;778],	 count:   2
 [ 1] mean:  259 [257;267],	 count:  37
 [ 2] mean:  519 [518;524],	 count:  13
Gap width distribution:
 [ 0] mean:  775 [775;775],	 count:   1
 [ 1] mean:  258 [255;264],	 count:  36
 [ 2] mean:  515 [515;519],	 count:  14
Pulse period distribution:
 [ 0] mean: 1542 [1542;1542],	 count:   1
 [ 1] mean:  517 [516;522],	 count:  27
 [ 2] mean:  775 [773;782],	 count:  19
 [ 3] mean: 1035 [1034;1039],	 count:   4
```
```
$1 = {num_pulses = 52, pulse = {767, 258, 258, 519, 519, 257, 259, 262, 518, 258, 258, 262, 262, 524, 518, 257, 518, 258, 257, 257, 258, 258, 258, 262, 261, 262, 267, 
    519, 258, 259, 258, 258, 258, 258, 258, 519, 257, 258, 519, 257, 258, 257, 261, 262, 518, 518, 258, 519, 257, 519, 262, 778, 0 <repeats 948 times>}, gap = {775, 
    258, 515, 515, 259, 259, 519, 256, 259, 259, 515, 255, 255, 515, 260, 516, 259, 259, 264, 260, 258, 259, 515, 256, 256, 255, 255, 259, 258, 258, 259, 258, 259, 
    263, 515, 259, 259, 515, 259, 264, 259, 516, 256, 256, 516, 264, 515, 259, 516, 515, 255, 269, 0 <repeats 948 times>}}
```
If I convert gfile011.data to FM and then load it into Audacity, it looks like the attached (hopefully) fm-011.png. All is well.
[image]

dwmw2, 2015-12-05:
For contrast, here's a packet which ends with a 0 bit, from gfile012.data. Note the spurious 6-sample "pulse" which appears to have been detected:
```
Test mode active. Reading samples from file: ../rtl_433_tests/tests/oil_watchman/01/gfile012.data
Detected FSK package
Analyzing pulses...
Total number of pulses: 53
Pulse width distribution:
 [ 0] mean:  766 [766;766],	 count:   1
 [ 1] mean:  259 [257;267],	 count:  38
 [ 2] mean:  518 [518;524],	 count:  13
 [ 3] mean:    6 [6;6],	 count:   1
Gap width distribution:
 [ 0] mean:  773 [771;775],	 count:   2
 [ 1] mean:  258 [255;264],	 count:  37
 [ 2] mean:  515 [514;520],	 count:  13
Pulse period distribution:
 [ 0] mean: 1415 [1289;1541],	 count:   2
 [ 1] mean:  517 [516;522],	 count:  28
 [ 2] mean:  775 [773;778],	 count:  19
 [ 3] mean: 1035 [1034;1038],	 count:   3
```
```
(gdb) p *pulses
$1 = {num_pulses = 53, pulse = {766, 257, 259, 518, 518, 258, 258, 261, 519, 259, 257, 262, 261, 524, 519, 257, 519, 258, 257, 258, 257, 258, 258, 262, 261, 262, 266, 
    519, 257, 257, 257, 258, 258, 258, 259, 519, 258, 257, 518, 258, 258, 519, 258, 257, 518, 267, 262, 262, 262, 518, 257, 518, 6, 0 <repeats 947 times>}, gap = {775, 
    259, 515, 516, 259, 259, 520, 256, 258, 258, 516, 255, 257, 514, 259, 516, 258, 259, 264, 259, 259, 259, 515, 256, 256, 255, 256, 259, 260, 259, 260, 258, 259, 
    263, 514, 259, 259, 516, 259, 264, 515, 258, 259, 516, 516, 255, 255, 255, 256, 260, 516, 771, 266, 0 <repeats 947 times>}}

```

Here's the screenshot from Audacity again:
[image]

Now, I can cope with producing a variant of the existing pulse_demod_manchester_zerobit() function, which handles the start-of-frame and end-of-frame stuff and will work packets that end with a 1 bit, like gfile011.data. But this spurious pulse that's being detected at the end of the frame when the last data bit is a zero? Should I really have to cope with that or is it a bug elsewhere? It doesn't seem to be just noise; there is *consistently* a 5-or-6 sample unwanted pulse in the data, for every packet that ends on a low frequency.

merbanan, 2015-12-05:
That is possibly a bug, but as I did not write the code I wont be able to fix that quickly. But if your description is correct:

"The period is about 1ms, and the complete transmission 136ms (6ms preamble, 128ms data, 2ms postamble)."

Then add that as protocol struct parameter so you end parsing before this spurious pulse is detected.

Something like this:

r_device oil_meter = {
[...]
    .preamble = 6ms in samples
    .postamble = 2 ms in samples
};

And then let the fsk manchester packet decoder verify and skip the preamble and also detect the postamble. Now it should be easy to stop and only send a valid bit buffer to the protocol decoder.

merbanan, 2015-12-05:
I think this "bug" is caused by going from the lower level to no carrier present. When that happens the signal will go towards zero and fluctuate. One of those fluctuations will probably go above the zero level and trigger this mini pulse. So in this case frame the bits by the post amble. But even if you didn't have any post amble you could just decode the signal to a bit buffer and send it away to protocol decoding. This would work fine as the packet format is fixed, and there is even a crc to verify the message with.

dwmw2, 2015-12-05:
Hm, not sure I follow; sorry. I was just trying to work on the decoding part first, before I look at the specific watchman protocol. These Si4320 / RFM01 modules are fairly common, it seems.

I've pushed a WIP at https://github.com/dwmw2/rtl_433/commit/acc1f6b80337cc646a9 which is mostly working for gfile011.data (I need to fix up the end-of-frame handling):
```
Detected FSK package
Analyzing pulses...
Total number of pulses: 52
Pulse width distribution:
 [ 0] mean:  772 [767;778],	 count:   2
 [ 1] mean:  259 [257;267],	 count:  37
 [ 2] mean:  519 [518;524],	 count:  13
Gap width distribution:
 [ 0] mean:  775 [775;775],	 count:   1
 [ 1] mean:  258 [255;264],	 count:  36
 [ 2] mean:  515 [515;519],	 count:  14
Pulse period distribution:
 [ 0] mean: 1542 [1542;1542],	 count:   1
 [ 1] mean:  517 [516;522],	 count:  27
 [ 2] mean:  775 [773;782],	 count:  19
 [ 3] mean: 1035 [1034;1039],	 count:   4
Guessing modulation: RFM Framed Manchester coding
Attempting demodulation... short_limit: 259, long_limit: 0, reset_limit: 779, demod_arg: 0
pulse_demod_manchester_zerobit(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] {65} 28 c7 48 0f 80 44 3a 4b 00 
```

This bears a striking resemblance to the data I was expecting :)

Now, I can cope with discarding the spurious pulse in my pulse_demod_manchester_framed() function. Once I actually implement the end-of-frame handling, that is.

But the really painful part is in pulse_analyser(). The logic for "three bins of pulse length and three bins of gap length and there must be 2 pulses with the least-common length, and that length must be longer than the other two bins, and there must be.... 

Hell, it's horrid enough as it is, and that's without even fixing it *(actually hist_pulses.bins[2].count==1&&hist_gaps.bins[2].count==2 is fine too)*. But extending it to cope with the odd extra bin... seems wrong. Am I missing something here?

merbanan, 2015-12-05:
Sorry that analyzer code is not mine can't help you with that, you could try asking vestrom (Tommy Vestermark) for help. It's not so easy to make something that you can feed random stuff and get it to produce sane things with. You might be able to do something by taking this:

Pulse width distribution:
 [ 1] mean:  259 [257;267],  count:  37
 [ 2] mean:  519 [518;524],  count:  13
 [ 0] mean:  772 [767;778],  count:   2
Gap width distribution:
 [ 1] mean:  258 [255;264],  count:  36
 [ 2] mean:  515 [515;519],  count:  14
 [ 0] mean:  775 [775;775],  count:   1

Discard the out liners with a count limit of 3 (maybe), then in the manchester decoder code you check the length and the gap of the data and if it matches one of these 2 values we know that this is data that belongs to the payload. The guessing logic will then be that there must be 2 "real" bins and 1 or more out liner bins.

dwmw2, 2015-12-05:
Actually, in the example you chose the outliers are *good* — those are the start-of-frame and end-of-frame markers on a frame which ends with a 1 bit. In `pulse_detect()` I was *checking* for those.

The problematic outlier is this one, in the frames which end with a 0-bit:

`Pulse width distribution:`
` [ 0] mean:  766 [766;766],  count:   1`
` [ 1] mean:  259 [257;267],  count:  38`
` [ 2] mean:  518 [518;524],  count:  13`
**`  [ 3] mean:    6 [6;6],  count:   1`**
`Gap width distribution:`
` [ 0] mean:  773 [771;775],  count:   2`
` [ 1] mean:  258 [255;264],  count:  37`
` [ 2] mean:  515 [514;520],  count:  13`

But yes, I can deal with that. Thanks for confirming that I *should* deal with it and not expect to (immediately) make it go away.

dwmw2, 2015-12-07:
The decode works fine with actual device support for the Watchman added. The heuristics for making `pulse_analyzer()` actually guess that it's this kind of framed FSK are somewhat hairier, so I've put them into a separate commit for ease of review. But I also understand now why that part doesn't matter so much as long as the decoding works right — which it does.

closed 2015-12-07 by dwmw2