merbanan/rtl_433 #223: FSK decoding issues at range

dwmw2, 2015-12-07:
I added support for the Watchman Sonic (issue #218) because I didn't want to live with the nasty hack I had originally wrapped around rtl_fm. I wanted to use proper signal processing code.

Unfortunately, now I've got it all working and I've put the transmitter back out on the oil tank where it belongs, I find it doesn't work very well. Where reception is poor, the rtl_433 code is doing a much worse job of extracting the FSK signal than the original using rtl_fm did.

At http://david.woodhou.se/goaway2.cap.gz there is a capture taken while I walked away from the computer, with the transmitter in my hand. When the signal level decreases, the FM demodulation gets very poor. Here's that sample, shown in Audacity *(imported as 25kHz sample rate)* alongside the FM demodulation that rtl_433 can also emit.

[image]

Note that the amplitude of the FM demodulation appears to decrease as the amplitude of the input decreases. Which is strange, because the frequencies surely don't get closer together? By the end of the capture, rtl_433 is unable to make anything out.

On the other hand, if I leave the transmitter out there and run rtl_fm to capture a signal, I get something that my original hack can still decode:
[image]

merbanan, 2015-12-07:
I must say that the fm demodulator in rtl_fm is better, the rtl_433 one is really minimalistic (like 20 lines of code). The options available are:

1. make the rtl_433 fm demodulator better
2. get a better source signal from the rtl-sdr dongle (increase the sample rate, etc...)
3. make if possible to feed rtl_433 predemodulated data (like what you did with your "hack")

Option 3 would guarantee success in your case.

dwmw2, 2015-12-07:
Heh, so much for *"I don't want to have to do signal processing code so I'll just add this protocol to something that already exists"* then :)

On the other hand, if I fix the FM demod in rtl_433 then at least that's not a single-purpose hack, and it might actually be more widely useful. And I can potentially just start lifting code from rtl_fm. I'll take a look...

vestom, 2015-12-12:
Hi,
The FSK decoder is still very young :-) You are right, that the FM demodulation is currently very simple and the frequency output depends on the amplitude as well. A simple atan2 approximation may fix that.

However, as the FSK decoder adapts to (slowish) changes of the frequency levels it should not really be a problem. A bigger problem, is that the carrier detection (OOK) is non-adaptive and will declare end-of-package before the signal is gone.

It is interesting with some more signals to challenge the decoder. I will have a look tomorrow and experiment a bit.

vestom, 2015-12-13:
Here a screenshot showing the simulated FSK demodulation, where the adaptive frequency estimators are shown. There is some noise in the low-amplitude part, but a simulation with a true atan2 FM demodulation showed similar noise. Maybe rtl_fm employs more heavy filtering?
I had to lower the OOK threshold though, so we should probably make that adaptive too.
[image]

dwmw2, 2015-12-13:
To make sure we're doing a proper comparison, I hacked rtl_fm to take input from a file, : http://david.woodhou.se/rtl-fm-infile-hack.patch
The output for that capture looks like this:
[image]
It's noisy at the end, but my fskdemod hack can actually decode it *(if I manually add a DC bias since it was originally just looking for negative vs. positive levels)*:
```
At 15.411625: 28 4e cc 8c 90 2c 00  (crc ok 14)
    Unit ID: 0x284ecc8c flags 0x902c height 0
```
In fact, both rtl_433 and rtl_fm+fskdemod both manage to get about 30 signals out of my 'goaway2.cap' capture. But a number of them are *different*. In general, rtl_433 does better on the earlier signals *(which were closer)* — I'd expect that. Near the end of the capture, however, my hack gets a few more than rtl_433 does. Including the one we're looking at here.

Admittedly, this specific one isn't a perfect example. Normally we do expect oil tanks to remain relatively still, and we wouldn't be walking around with the sensor. But in general it *would* be good to improve performance on the lower-strength signals.

I'll attempt to provide some realistic captures from the tank actually in situ — I've already added some to rtl_433_tests, which are mostly decodable with a combination of reducing the OOK detection limit *(as you noted)*, and reducing `FSK_DEFAULT_FM_DELTA`. But I've been kind, and those are from my laptop slightly closer to the oil tank; I'll do some from the location I *actually* wanted to use rtl_433 in :)

dwmw2, 2015-12-13:
There is a lot of scope for layering violations here, if we know what we're looking for. By the time we get to the noisy part of the signal here, for example, we should *know* we're looking for signals at 433867 kHz and 433925 kHz — so the FM demodulator could be basically making a binary decision, and the results of the noise could be filtered out a little more effectively.

There are other things we could do along those lines too. Some of those feedback options are quite hard to implement, especially given the number of devices we support and the diversity of signals. But some might be feasible, so we should keep an eye out for them.

vestom, 2015-12-13:
I do not think it is necessary nor desirable to use absolute frequency offsets. That would add much complexity to the code and make the decoder very sensitive to changes in tuner frequency and sampling frequency.

I have been toying with improving the current detector with the following:
* Adding automatic OOK / carrier level detection with adaptive threshold estimation
* Adding spurious short pulse suppression in the initial OOK pulse and in the FSK pulses.

It seems to work pretty good with the signal above
```
$ ./rtl_433 -r goaway2_cut.cap -l 0 -A
Registering protocol "CurrentCost Current Sensor"
Test mode active. Reading samples from file: goaway2_cut.cap
Detected FSK package
2015-12-13 22:19:48 Oil Watchman 284ecc8c 90 11 0 0
Analyzing pulses...
Total number of pulses: 52
Pulse width distribution:
 [ 0] mean:  764 [764;764],	 count:   1
 [ 1] mean:  255 [253;260],	 count:  36
 [ 2] mean:  514 [513;515],	 count:  14
 [ 3] mean:    3 [3;3],	 count:   1
Gap width distribution:
 [ 0] mean:  768 [768;769],	 count:   2
 [ 1] mean:  257 [253;264],	 count:  35
 [ 2] mean:  512 [510;516],	 count:  14
Pulse period distribution:
 [ 0] mean: 1533 [1533;1533],	 count:   1
 [ 1] mean:  513 [510;519],	 count:  25
 [ 2] mean:  769 [766;772],	 count:  20
 [ 3] mean: 1026 [1022;1031],	 count:   5
Level estimates: 556 / 4540
Guessing modulation: No clue...
```
I have not tested it on the complete test corpus yet, but you can try it out here:
https://github.com/vestom/rtl_433/tree/experimental
Automatic level detection is enabled by `-l 0`

dwmw2, 2015-12-14:
Thanks; this seems to be an improvement. Doesn't quite seem to work on
https://github.com/merbanan/rtl_433_tests/blob/master/tests/oil_watchman/01/gfile024.data though — for that I still need an explicit `-l 1500` and also to adjust `FSK_DEFAULT_FM_DELTA` down to 4000.

itretail, 2015-12-15:
I have been trying to follow this thread, although most of it is beyond me, but I am trying to establish if this device can produce json or csv  output , if so could you advise the correct command line to use with rtl_433

itretail, 2015-12-15:
by this device  I mean the watchman sonic if that wasn't apparent

merbanan, 2015-12-15:
@itretail It can do both. Add -F json|csv to select between json and csv.

itretail, 2015-12-16:
I already tried that with

rtl_433 -R 42 -F csv output.csv

this only resulted in what looks like the data stream being written to output.csv

dwmw2, 2015-12-16:
Let's start with it simple — can you bring the device close to your receiver and stick a magnet by its binding spot (briefly)? That'll make it stream about three reports a second, instead of one every 35 minutes.

Then take it back out and stick it on the oil tank. Touching a magnet to it again will make it stop streaming. Only if you hold the magnet there for ten seconds or so will it make a new unit ID and need rebinding to the existing receiver.

I don't think you need 'output.csv' on the command line. You get the CSV output on stdout.

FWIW here's how I've hooked it up to Domoticz: http://www.domoticz.com/forum/viewtopic.php?f=17&t=4390&p=65711#p65711 
And to turn the depth reading into a percentage and volume, http://www.domoticz.com/wiki/Lua_-_Oil_Tank_Monitor

itretail, 2015-12-16:
Thanks for that, I might have been  bit inpatient, I didn't realise it only reported every 35 minutes. I'll try your magnet test tomorrow.

dwmw2, 2015-12-21:
Any joy with retesting it?

itretail, 2015-12-22:
not yet, turns out my transmitter is faulty, and now have to wait for a new one to be shipped after xmas. will post results when I get chance to try again.

itretail, 2016-02-09:
finally got the device working, turns out just flat battery but if you open the transmitter up to check the battery it voids the warranty. Anyway now all working and RTL_433 is receiving data about every 45 minutes. Next step is to extract the depth from the output and work out how many litres that represents.
Thanks everyone for the help !!!

AJP4, 2016-10-02:
How did it go?

dwmw2, 2017-02-08:
> Next step is to extract the depth from the output and work out how many litres that represents.

http://www.domoticz.com/wiki/Lua_-_Oil_Tank_Monitor

merbanan, 2018-12-05:
Closing issue due to inactivity.

closed 2018-12-05 by merbanan