merbanan/rtl_433 #224: CurrentCost current sensor & generic RFM01 decoding

enavarro222, 2015-12-09:
The next not-yet-decoded-by-rtl_433 device at home is a CurrentCost (CC) current sensor (http://www.currentcost.com/product-envi.html). (Next and last for now, will be the door bell).

I push to rtl_433_test some samples + a audacity project with data analysed with tag.
https://github.com/merbanan/rtl_433_tests/pull/53

I push a quick and dirty (but mostly working) decoding in my fork:
https://github.com/enavarro222/rtl_433/tree/currentcost

I do not propose a PR yet because, for now, I just savagely patch `pulse_demod_manchester_framed` proposed by @dwmw2. The thing is CC seems to use RFM01... However the msg's sync&init seems somehow different. My "patch" no more work for "Oil Watchman" sensors.

So the ideal PR will provide a decoding that work for both devices.

I didn't dig further for now. But maybe there is someone here with a stronger knowledge of RFM01 who can help !

I may tear-down my sensor to take pics and check it is actually a RFM01.

dwmw2, 2015-12-09:
Hm, I'm looking back at the RFM01 documentation and I see this:

> BR is the bit rate set in the receiver and ΔBR is bit rate difference between the transmitter and the receiver. N is the maximal number of consecutive ones or zeros in the data stream. It is recommended for long data packets to include enough 1/0 and 0/1 transitions, and be careful to use the same division ratio in the receiver and in the transmitter

This implies that perhaps the framing and the Manchester encoding *aren't* done by the hardware, and it's actually part of the higher-level protocol. Perhaps we should parameterise the decoder in a fashion similar to `pulse_demod_pwm_precise()` so that we can pass in the details of the pre/postamble we expect on different devices?

merbanan, 2015-12-09:
Sounds like a good idea. I remember writing something in the previous ticket how it could be done.

dwmw2, 2015-12-09:
Hm, this is not as simple as a different length of preamble and postamble. They all seem to start like this...
[image]
In rtl_433's analysis, the low frequency is 'pulse' and the high is 'gap':
```
[  1] Pulse: 8279, Gap:  126, Period: 8405
[  2] Pulse:  126, Gap:  125, Period:  251
[  3] Pulse:  124, Gap:  126, Period:  250
[  4] Pulse:  125, Gap:  125, Period:  250
[  5] Pulse:  125, Gap:  125, Period:  250
[  6] Pulse:  125, Gap:  126, Period:  251
[  7] Pulse:  125, Gap:  125, Period:  250
[  8] Pulse:  125, Gap:  125, Period:  250
[  9] Pulse:  188, Gap:   62, Period:  250
[ 10] Pulse:   63, Gap:  125, Period:  188
[ 11] Pulse:   62, Gap:  188, Period:  250
[ 12] Pulse:   63, Gap:   62, Period:  125
[ 13] Pulse:   63, Gap:   62, Period:  125
[ 14] Pulse:  188, Gap:   63, Period:  251
[ 15] Pulse:   62, Gap:   63, Period:  125
[ 16] Pulse:   62, Gap:   63, Period:  125
[ 17] Pulse:   63, Gap:  124, Period:  187
```
The base clock is about 62½ samples. You start with a *very* long preamble of about 132 clocks, which seems a little excessive; maybe it's transmitting while its controller is actually obtaining the reading? But then when you get to the data, there are 1-clock, 2-clock *and* 3-clock periods at the same level. If it's a simple Manchester encoding, there *should* be a transition at least every other clock, so you should never see it remain at the same level for 3 clocks except as part of the framing. So what are those 188-sample pulses and gaps doing there?

After the piece I've quoted, it looks like an understandable Manchester encoding, ending with two more clocks at the last frequency to end the frame. But what's happening at the beginning?

dwmw2, 2015-12-09:
Perhaps the answer is for the pulse decoder *not* to handle the Manchester encoding at all. Just return the raw series of levels *(in the watchman case, 1110001010...01..10...0111)* and let the device code handle the framing and do the Manchester decoding for itself *(we can provide a library function)* on the contents of the bitbuffer?

Then I get to *not care* about the answers to my questions above :)

merbanan, 2015-12-09:
"do the Manchester decoding for itself (we can provide a library function)" sounds nice.

enavarro222, 2015-12-09:
:+1: for this solution.
So the decoder is just a `pulse_demod_pcm` ?

enavarro222, 2015-12-09:
Seems ok with pulse_demod_pcm,
I'll try to with a generic manechester decoding, and sync&init seeker.

enavarro222, 2015-12-09:
Ok here is a try with  `pulse_demod_pcm` for Current Cost  sensor:
https://github.com/enavarro222/rtl_433/commit/703a0258dc73d927ed65f6945e7804d22483cdd8

@dwmw2 can you have a look ? Do you think it could work for your sensor ?

If ok we need to mv the two fct `_seek_init_pattern` and `_decode_manchester` else where. utils.c ? bitbuffer.c ? somewhere else ?

dwmw2, 2015-12-10:
I can probably make it work for my sensor, but it doesn't seem to work for yours. Your 'currentcost' branch (https://github.com/enavarro222/rtl_433/commit/9fe73edb855fc0fa780578c05a189e0e294d94c8) works on the test data you provided, but your 'rfm01' branch (https://github.com/enavarro222/rtl_433/commit/703a0258dc73d927ed65f6945e7804d22483cdd8) doesn't:
```
[dwoodhou@i7 rtl_433]$ git reset --hard 9fe73edb855fc0fa780578c05a189e0e294d94c8
HEAD is now at 9fe73ed Change pulse_demod_manchester_framed for CC
[dwoodhou@i7 rtl_433]$ make &> /dev/null
[dwoodhou@i7 rtl_433]$ src/rtl_433 -r ../rtl_433_tests/tests/current_cost/01/cc_2542watt.data -q
2015-12-10 08:04:43 CurrentCost TX: Power 0 2542, Power 1 72, Power 2 72
[dwoodhou@i7 rtl_433]$ git reset --hard 703a0258dc73d927ed65f6945e7804d22483cdd8
HEAD is now at 703a025 Manage manchester decoding in CC callback
[dwoodhou@i7 rtl_433]$ make &> /dev/null
[dwoodhou@i7 rtl_433]$ src/rtl_433 -r ../rtl_433_tests/tests/current_cost/01/cc_2542watt.data -q
[dwoodhou@i7 rtl_433]$ 
```

enavarro222, 2015-12-10:
oh thanks (and sory) ! it's fixed.
I forget to RAZ a buffer so, some time it did work... now it is ok with test data on rtl_433_test.

dwmw2, 2015-12-10:
OK. See pull request https://github.com/merbanan/rtl_433/pull/226 
All that's left is to convert the CurrentCost sensor to use the same facilities, which I've done at https://github.com/dwmw2/rtl_433/commit/748136321f3f4b4764ff2d8cb7304b49c89a1910 (not included in  the pull request).

Is there really no CRC in the packet, or check of any kind?

enavarro222, 2015-12-10:
Great ! I'll propose a PR for CurrentCost.

To my knowledge, nop no check.
Just manchester encoding that may help detect a error..
For sure the last 6 bytes are for measurements.
The first one may be for sensor id, the second one may be for a rolling code. (I have to unplug my sensor and remove the battery to check that)

It could be great if someone else have current cost sensor had could send some test data !

PR #229 (2015-12-10): Add CurrentCost tx device
Here is the PL that should close #224 
Thanks @dwmw2 !

I'll continue to work on this later, at least for that i guess to be a rolling code... some of the packet bits still seems mysterious to me.
closed 2015-12-10 by merbanan