merbanan/rtl_433 #227: Disable / deprioritise devices without CRC support

dwmw2, 2015-12-10:
Watching my oil monitors, I occasionally get spurious reports for invalid devices:
```
2015-12-10 14:51:16 Oil Watchman c7724c 60 19 0 0
2015-12-10 15:23:17 Oil Watchman 4ecc8c 80 20 0 58

Sensor        = Temperature event
Device        = 0
Battery?      = 00
Temp          = 0.000000
Model         = Generic temperature sensor 1
Received Data = 00 00 00
2015-12-10 15:26:20 Oil Watchman c7724c 80 20 0 0
2015-12-10 15:58:16 Oil Watchman 4ecc8c 80 20 0 58
2015-12-10 16:02:03 Oil Watchman c7724c 80 19 0 0
2015-12-10 16:33:15 Oil Watchman 4ecc8c 80 20 0 59
2015-12-10 16:37:46 Oil Watchman c7724c 80 19 0 0
Danfoss CFR Thermostat:
bitbuffer:: Number of rows: 1 
[00] {258} f8 3f f0 03 ff 07 c1 f0 7c 1f 00 3f f0 03 e0 f8 3e 0f 83 e0 f8 3f f0 7c 00 f8 3f f0 7c 1f 07 c1 c0 

2015-12-10 17:13:29 Oil Watchman c7724c 80 19 0 0
 
Sensor        = Temperature event
Device        = 0
Battery?      = 00
Temp          = 0.000000
Model         = Generic temperature sensor 1
Received Data = 00 00 00
2015-12-10 17:49:12 Oil Watchman c7724c 80 18 0 0
```
Unfortunately I didn't capture the input waveform so can't reproduce. But that definitely should have been one of the transmissions from unit 4ecc8c. It got eaten by the Danfoss driver which just dumps the bitbuffer without doing any checks. When the Danfoss callback returned '1' did that mean the pulses weren't even offered to the oil_watchman callback to be interpreted?

Should I have put oil_watchman earlier in the device list, as has a decent CRC and is *very* unlikely to claim events that aren't really its own? And can we have a mode which disables *all* device support that doesn't have reasonable confidence? Perhaps an estimate *in* the `r_device` structure of the probability of false positives would be a good idea? And would also help to focus the submitters' minds on such things :)

merbanan, 2015-12-10:
Hi, run with -t for a while. It will save all detected signals. That way we can debug later.

dwmw2, 2015-12-10:
It doesn't seem to be. Running with `-t -a` saves signals. Is there a mode which will actually run the real device decoder as well as saving signals to separate files?

dwmw2, 2015-12-10:
Running with this for now; we'll see what it catches:
```
--- a/src/rtl_433.c
+++ b/src/rtl_433.c
@@ -619,7 +619,7 @@ static void rtlsdr_callback(unsigned char *iq_buf, uint32_t len, void *ctx) {
 
        if (demod->analyze || (demod->out_file == stdout)) {    // We don't want to decode devices when outputting to stdout
                pwm_analyze(demod, demod->am_buf, len / 2);
-       } else {
+       } /* else */ {
                // Loop through all demodulators for all samples (CPU intensive!)
                for (i = 0; i < demod->r_dev_num; i++) {
                        switch (demod->r_devs[i]->modulation) {
```

dwmw2, 2015-12-10:
Hm, this one doesn't get decoded at all. If I let rtl_433 do the FM decoding and feed it to my own original hack decoder, it does work.

dwmw2, 2015-12-10:
Oh, that one does work with `-l 4716`.

dwmw2, 2015-12-10:
Perhaps for FM decoding, a lower default detection level should apply than for AM? Isn't that half the point in using FM? If you look at the FM decode in this case, it's not as if it's at all ambiguous...
[image]

merbanan, 2015-12-11:
If you find a better limit set that as default.

merbanan, 2015-12-11:
And ideally there should be some kind of prioritization for packet decoding so signals with crc are tried before those without.

merbanan, 2015-12-11:
Some signal formats are very simple. Those are hard to not get conflicts with.

merbanan, 2018-12-03:
There has been significant rework of the code so bogus signals should be less common. It is also possible to disable decoders if the application only need to look for one device. And hopefully the signal recorder will get smarter so it is possible to run the regular decode process but still record signals for later analysis. Closing issue.

closed 2018-12-03 by merbanan