merbanan/rtl_433 #228: Oregon Scientific Output issues

ygator, 2015-12-10:
THGR122N

Incorrectly reports channel 3 as channel 4.

Inconsistent output.  In json called "id" and in kv called "House Code".

Incorrect temperature output in json and csv.
20.299999
17.700001
17.200001

enavarro222, 2015-12-10:
I notice the bug for channel number, I'll check that.

We may start a wiki page to indicate the standard output ? a kind of guideline on how to names devices output fields... 

For this "id"/"house code", for me the term "rolling code" is more clear.

The temperature output may have this form on JSON and CSV output as it is float numbers.

ygator, 2015-12-10:
"id" is my okay or "pid" as I see it as a "power on id" since it only changes when the sensor is powered on.

Do any of the sensors report floating point numbers or do they actually report digits with an implied decimal point?  It might be better to represent these as an integer and decimal place.  The THGR122N sensor actually reports 3 digits and a decimal place of 1.  So 20.3 should be reported as 20.3 and not 20.299999 as json and csv report.  It should also not be reported as 20.30 as kv does as this would indicate another digit of precision.

merbanan, 2015-12-10:
The sensors are only somewhat self calibrated. Ie 21.0 is 1 degree less then 22.0. But the actual real temperature error is ca +-1. So while I'd love to have perfect representation of things I think we have to take what we have now. Reworking it would not gain anything as doing any processing with the data in a later step would most likely induce the same type of rounding.

ygator, 2015-12-11:
It could even be simpler.  Just treat it as a string.

closed 2015-12-11 by merbanan