merbanan/rtl_433 #239: Imagintronix Temperature & Humidity Sensor - WH5

mpmc, 2015-12-15:
I've bought another sensor (I've got three now :p), this one is the Imagintronix Temperature & Humidity Sensor - WH5 apparently  it uses the same protocol as the WH2 (see https://www.pitt-pladdy.com/blog/_20140102-173814_0000_Watson_Temperature_Humidity_Sesnor_Protocol_8683-HTT_WH2_for_W-8683_/) which rtl_433 finds. It unfortunately doesn't find the WH5.

Here's a sample of the output received by rtl_433.
```
[00] {46} 00 00 00 00 00 00 : 00000000 00000000 00000000 00000000 00000000 000000
Analyzing pulses...
Total number of pulses: 47
Pulse width distribution:
 [ 0] mean:  111 [109;115],      count:  27
 [ 1] mean:  356 [354;358],      count:  20
Gap width distribution:
 [ 0] mean:  254 [252;257],      count:  46
Pulse period distribution:
 [ 0] mean:  366 [363;369],      count:  26
 [ 1] mean:  610 [609;612],      count:  20
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_limit: 233, long_limit: 258, reset_limit: 258, demod_arg: 0
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {47} fe 80 64 fc 6b ae : 11111110 10000000 01100100 11111100 01101011 1010111
```

Any help understanding this would be awesome!

merbanan, 2015-12-15:
First you must document the device as well as you can. Look into the readme on how to do that.

dh-harald, 2017-06-30:
Hi @merbanan ,

Is any chance to get this sensor working? Eventually, this sensor is still on the market and it's quiet cheap.
As far as I know, @mpmc has already uploaded the test data to move forward... Could you take a look on it please?

zuckschwerdt, 2017-06-30:
Looks like straight forward PWM coding. I'll take a peek an see what I can make of the payload. Do you already own one? Can you test and confirm readings?

mpmc, 2017-06-30:
I still have this sensor but since rtl_433 didn't support it, I switched to something else. It can however be dug out of the "crap drawer" & tested again if needed.

dh-harald, 2017-06-30:
Yes, I've two sensors: WH5 & RTGN318, so I can compare the results, if needs

zuckschwerdt, 2017-06-30:
Ah well, really looks like it's the same protocol as the WH2. There is just one leading sync bit missing. I'll fix that, stay tuned.

zuckschwerdt, 2017-06-30:
Please try: https://github.com/zuckschwerdt/rtl_433/tree/feat-wh5, i.e.
```
git clone -b feat-wh5 https://github.com/zuckschwerdt/rtl_433.git
cd rtl_433
mkdir build
cd build
cmake ..
make
```
and test :)
The CRC and decoding works. But the temperature value seems wrong. It reads a value around 60 with the rtl_433_tests examples. Could be Fahrenheit instead of Celsius, maybe.

dh-harald, 2017-06-30:
@zuckschwerdt: From the page, that linked before:
"The only difference is that the temperature is represented as T + 40 in the WH5 where it's 2's Compliment in the WH2. Like the WH15B this device is fast responding, but initial accuracy compared to other sensors is not as good as the WH2 new, but more in line with the WH2 now that it's aged a bit."
is it helps you?

zuckschwerdt, 2017-06-30:
Mystery solved, yes. I've updated the branch. Remove the cloned repo and re-clone as outlined above. Should work now.

dh-harald, 2017-06-30:
I came home, start the sensor, and it looks not working.... It can't parse data.
It looks, my sensor provides different data, so I'll collect gfiles, and I'll upload it soon... Stay tuned.

mpmc, 2017-07-04:
@zuckschwerdt Just tried your changes & can confirm it appears to be working and correct!

```
2017-07-04 19:27:45 :	Fine Offset Electronics, WH2 Temperature/Humidity sensor
	ID:	 45
	Temperature:	 21.6 C
	Humidity:	 54 %
2017-07-04 19:28:20 :	Fine Offset Electronics, WH2 Temperature/Humidity sensor
	ID:	 253
	Temperature:	 26.8 C
	Humidity:	 41 %
2017-07-04 19:28:32 :	Fine Offset Electronics, WH2 Temperature/Humidity sensor
	ID:	 45
	Temperature:	 21.6 C
	Humidity:	 54 %
2017-07-04 19:29:07 :	Fine Offset Electronics, WH2 Temperature/Humidity sensor
	ID:	 253
	Temperature:	 26.9 C
	Humidity:	 41 %
2017-07-04 19:29:20 :	Fine Offset Electronics, WH2 Temperature/Humidity sensor
	ID:	 45
	Temperature:	 21.6 C
	Humidity:	 54 %
```
ID: 45 is the one outside [WH5]
ID: 253 is the one on my desk in front of a fan [WH2 / Imagintronix]

The screen for the WH5 says the in temp is 26.3. 

Looks like the battery in the other [third] sensor has died :/

zuckschwerdt, 2017-07-05:
Great! I'll clean the changes up and add an output with `WH5` in the model. Expect this to take a few days though.

zuckschwerdt, 2017-07-05:
@dh-harald your samples get detected as `RTGN318` and the decoding looks sensible:
```
gfile001.data
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 22.900, "humidity" : 51}
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 22.900, "humidity" : 51}

gfile003.data
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 22.900, "humidity" : 51}
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 22.900, "humidity" : 51}

gfile005.data
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.000, "humidity" : 51}
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.000, "humidity" : 51}

gfile007.data
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.000, "humidity" : 51}
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.000, "humidity" : 51}

gfile013.data
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.100, "humidity" : 51}
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.100, "humidity" : 51}

gfile015.data
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.100, "humidity" : 51}
{"time" : "@0.262144s", "brand" : "OS", "model" : "RTGN318", "id" : 173, "channel" : 1, "battery" : "OK", "temperature_C" : 23.100, "humidity" : 51}
```
(`rtl_433 -q -F json -r gfile0XX.data`)

mpmc, 2017-07-05:
@zuckschwerdt Awesome - WH5 is the correct model for the Imagintronix sensor, I got them mixed up. I wonder if this'll also work [correctly] with the cheap black sensor I have [as noted in #103].

zuckschwerdt, 2017-07-05:
@dh-harald sorry, the samples you linked are distorted. Looks like the reciever gain is to high, i.e. the device is too close to the antenna. The RTGN318's are fine. Try to add some distance or maybe unplug the antenna and put the device at 5-10 cm distance. Verify the sample in Audacity (Import Raw, Unsigned 8-bit, 2 channels). Perhaps the AM decoding plugin can help too: http://triq.net/articles/sdr-nyquist-plugin-am-decode-iq-signals -- or even better the WH5 decoding will work right away then :)

merbanan, 2018-12-07:
The issue seems to have been solved. Closing issue.

closed 2018-12-07 by merbanan