merbanan/rtl_433 #242: Patch - ESA 1000 / 2000 Energy Cost Monitor

TylerDurden23, 2015-12-16:
[esa-monitor.diff.txt](https://github.com/merbanan/rtl_433/files/63420/esa-monitor.diff.txt)
[image]

merbanan, 2015-12-17:
Can you send a github pull request of this. Together with some samples for the rtl_433_tests repository ?

TylerDurden23, 2015-12-19:
The recording is attached as part of the post (the record signal gif, github is not accepting raw file uploads).

merbanan, 2015-12-19:
Well look at the rtl_433_tests repository and you will find out how to submit samples. I'm not sure what you did but it is not working for me.

TylerDurden23, 2017-10-31:
Upgrade for compatibility with the latest version.

[esa.zip](https://github.com/merbanan/rtl_433/files/1431093/esa.zip)

zuckschwerdt, 2017-10-31:
- Fork this repo (use the "Fork" button)
- Clone that to your computer (`git clone https://github.com/TylerDurden23/rtl_433.git`)
- Create a branch (`git checkout -b feat-esa`)
- Make changes to the source
- Commit (`git add ...`, `git commit`), Push (`git push`)
- Create a pull request (buttons on github)

merbanan, 2018-12-09:
@TylerDurden23 the code has been committed in a modified form. If you have more samples please submit them. Closing issue.

closed 2018-12-09 by merbanan