merbanan/rtl_433 #247: Current Cost device do not work

jacopot3, 2016-01-09:
Hello, 

I tried multiple Current Cost trasmitter associated with ENVIR (http://www.currentcost.com/product-envir.html) and at the moment none of them work with latest version of rtl_433.

Please find on the link below the technical details on how the RF Current Cost protocol works:

https://github.com/JackKelly/rfm_edf_ecomanager/wiki/Technical-details-of-Current-Cost-RF-protocol

If this could get fixed, it would be much appreciated!

Many thanks!

enavarro222, 2016-01-10:
Hi,

Can you push raw samples of your sensors data on the rtl4_33_test repo (https://github.com/merbanan/rtl_433_tests) ?

Thanks for the link.
After a quick lock, It seems to fit with what I coded in rtl_433.
Except "other data" (https://github.com/JackKelly/rfm_edf_ecomanager/wiki/Technical-details-of-Current-Cost-RF-protocol#other-data) that are not handle.

+

davidsheldon, 2016-02-08:
The problem is with the first byte check. I suspect even my recent pull request wont solve it for these devices if they set the data flag in bit 7 of the first byte.

MarcoMerlo, 2018-01-29:
Hi,
great project, saved me from huge expenses allowing re-use of some sensors!

I'm now in a trouble with the EnviR not recognized by the latest build.

I'm not an expert on C, I'm a java enterprise dev, to me it's not fully clear the process to decode OOK messages, but if I can help to solve this issue just let me know, also because I need it...

Many thanks!

zuckschwerdt, 2018-01-30:
Get samples of different states with `-a -t`. Verify the samples with a spectrogram (http://triq.net/iqs) and analyze the samples with e.g. Audacity. Determine the coding and timings. Report what you found and ask if you get stuck.

merbanan, 2018-12-06:
No sample signals referenced, closing issue. If someone can provide test signals this should be easily implementable.

merbanan, 2018-12-06:
Looks like it actually was implemented.

closed 2018-12-06 by merbanan