merbanan/rtl_433 #254: Cannot find a signal in an obvious spike

viraptor, 2016-01-16:
There's an obvious spike I see every 36 seconds, but rtl_433 doesn't seem to recognize / register it at all. I'm running:

    rtl_433 -m 3 -r gqrx_20160116_014247_433985000_240000_fc.raw -A

The signal quality is not perfect, but the jump is large enough to see clearly.

[signal.zip](https://github.com/merbanan/rtl_433/files/92752/signal.zip)

merbanan, 2016-01-16:
I opened the file in audacity as raw 32bit float little endian. There was only low level noise. You have to use rtl_433  -a -t to grab signal data if you want help analyzing it.

closed 2016-01-16 by merbanan

viraptor, 2016-01-16:
Sorry, that was a bad file :/ Here's one I actually checked. (uint8, iq, 250kHz) It should be decoded by dev 32 (digitech weather station). The signal is there, gets to almost 0.5 in audacity.

Yet, rtl_433 seems to ignore it mostly. With -a and -A it notices the signal if I lower the level to -l 1000. But even then with -R 32 -l 1000, I don't get the data back. -A gets the correct data, -a doesn't.

I created a simple OOK processor in gnuradio and it has no problems decoding that signal. (I'm using 10k low-pass first) It seems like rtl_433 could be more sensitive.
[weather_station.zip](https://github.com/merbanan/rtl_433/files/92890/weather_station.zip)

merbanan, 2016-01-17:
This looks better. And if -A works then the demodulator is ok. What is left then is bugs in the code. And those are fixed now if you retest. Running with -l 0 also works.

merbanan, 2016-01-17:
And I see that this decoder is not yet ported to the new json output. Feel free to implement that.

merbanan, 2016-01-17:
Temperature event:
protocol      = Digitech XC0348 weather station
temp          = 16.7°C
humidity      = 44%
direction     = S
speed         = 0.0 km/h
unknown       = 00 07 ca

For the record this is the output from the supplied file.

viraptor, 2016-01-18:
awesome, thanks!