merbanan/rtl_433 #259: Is the test suite supposed to succeed with current master?

zuckschwerdt, 2016-01-16:
With current master @ccf7f7d many tests fail for me. Ignoring the time (fixed dates vs @ sample_file_pos) I get
```
35 records tested, 29 have fail
```
Specifically:
```
## Fail with 'tests/hideki/01/ch1_-440_38.data':
 iterable_item_added
  * root: [{u'battery': u'OK', u'humidity': 38, u'temperature_C': -4.4, u'rc': 3, u'model': u'HIDEKI TS04 sensor', u'channel': 1}, {u'battery': u'OK', u'humidit
y': 38, u'temperature_C': -4.4, u'rc': 3, u'model': u'HIDEKI TS04 sensor', u'channel': 1}]
## Fail with 'tests/oil_watchman/01/gfile001.data':
 values_changed
  * root[0]['id']: 13060111 ===> 684148751
  * root[1]['id']: 13060111 ===> 684148751
[17 more]
```
note:
````
684148751 is 101000110001110100100000001111
13060111 is                110001110100100000001111
```

and current code seems to remove duplicates, as I get
```
## Fail with 'tests/oregon_scientific/04/gfile011.data':
 iterable_item_added
  * root: [{u'battery': u'OK', u'humidity': 47, u'temperature_C': 25.4, u'model': u'Thermo Hygro RF Clock Sensor RTGN318', u'id': 179, u'channel': 2}]
[7 more]
``

Can someone report what the supposed output is?

enavarro222, 2016-01-16:
For "oil_watchman", id changed... I don't know if if it's a regression or if recorded json was false...

For Hideki, you need to had `--first-line` option (see makefile test target). It's a shortcut for now. Hideki decoder should outputs one line, however it outputs 3. This should be fix in rtl_433 rather than in the test... I had no time to fix it yet.

zuckschwerdt, 2016-01-16:
My bad, sorry. I needed `--rtl433-cmd` and wasn't using the makefile. Works as supposed. Thanks!

closed 2016-01-16 by zuckschwerdt