merbanan/rtl_433 #260: Temperature in Celsius, Fahrenheit or even both?

zuckschwerdt, 2016-01-16:
When using or converting to the new data printer format, do we give temperature readings in Celsius, Fahrenheit or even both?
I'd like to see some easily processable output (with JSON). I don't think giving both would add any value, only make things more verbose. But that might be just me?
Are there historical requirements and what are the opinions on this?

Quick census on `.field` keys:
* temperature_F
** ambient_weather.c
* temperature_C
** alecto.c
** hideki.c
** oil_watchman.c
** oregon_scientific.c
** prologue.c
** wt450.c

rct, 2016-01-16:
The decoder should output what ever units are "native" to the device/message.

If you wind up with multiple conversions, you may loose some precision due to rounding floats for output to ascii.  

You may also obscure small changes which could be jitter or could be useful data.

The LaCrosse TX 433 mhz units I've seen use celcius natively (just offseting to avoid the need for a sign) so they should output Celsius.     

(I say that even though I'm from one of the few/only countries that uses Fahrenheit, so the output will no longer be directly meaningful to me. ;-)

zuckschwerdt, 2016-01-16:
Entirely reasonable, output all data as native as possible. (I'd even avoid floats as most devices encode a decimal and dividing by 10 isn't all perfect in binary.)
My current metric logging (graphite) doesn't honor units, it would be great to get a standardized output if at all possible.
Perhaps an option to convert values (native, C, F) in the data printer would be ok?

merbanan, 2016-01-16:
A normalizing step to select Celsius or Fahrenheit output, would be welcome.

rct, 2016-01-17:
A normalizing step would be welcome only if raw output was still accessible.

zuckschwerdt, 2016-01-17:
I was thinking about a command line option, e.g. `--c-to-f` & `--f-to-c` which would trigger code in `data_acquired_handler`:
* if `--c-to-f ` and the type is DATA_DOUBLE and the key is temperature_C then convert
* if `--f-to-c ` and the type is DATA_DOUBLE and the key is temperature_F then convert

And maybe, if there is a trailing `" C"` (`" F"`) in the format string switch that.
Also some sensors are opinionated and have build-in conversion. I'd remove that.

Current practise:
```
alecto.c:
  "temperature_C", "Temperature", DATA_FORMAT, "%.02f C", DATA_DOUBLE, (float) temp / 10.0F,
ambient_weather.c:
  "temperature_F", "", DATA_FORMAT, "%.1f", DATA_DOUBLE, temperature,
hideki.c:
  "temperature_C", "Temperature",   DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp/10.f,
oil_watchman.c:
  "temperature_C", "", DATA_DOUBLE, temperature,
oregon_scientific.c:
  "temperature_C", "Temperature", DATA_FORMAT, "%.02f C", DATA_DOUBLE, get_os_temperature(msg, sensor_id),
prologue.c:
  "temperature_C", "Temperature", DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp,
wt450.c:
  "temperature_C", "Temperature",DATA_FORMAT, "%.02f C", DATA_DOUBLE, temp,
```

merbanan, 2016-01-17:
Sounds good, if you write the code I will merge it.

zuckschwerdt, 2016-01-17:
Well, here you go: https://github.com/merbanan/rtl_433/pull/263 ;)

merbanan, 2016-01-19:
Merged.

closed 2016-01-19 by merbanan