merbanan/rtl_433 #284: "Generic Remote SC226x EV1527" Is not working...

Harvie, 2016-01-31:
Hello,
i've tried generic remote with HS1527 which is probably identical to EV1527 and it's not working... Datasheets are exactly same:

http://sc-tech.cn/en/hs1527.pdf
http://sc-tech.cn/en/sc1527.pdf

This is what i receive for pressing 4 various buttons using ./rtl_433 -R 30 -a

```
*** signal_start = 5049941, signal_end = 5114386
signal_len = 64445,  pulses = 176
Iteration 1. t: 100    min: 51 (106)    max: 150 (70)    delta 13
Iteration 2. t: 100    min: 51 (106)    max: 150 (70)    delta 0
Pulse coding: Short pulse length 51 - Long pulse length 150

Short distance: 47, long distance: 146, packet distance: 1531

p_limit: 100
bitbuffer:: Number of rows: 8 
[00] {1} 00 : 0
[01] {25} 88 3f 09 00 : 10001000 00111111 00001001 0
[02] {25} 88 3f 09 00 : 10001000 00111111 00001001 0
[03] {25} 88 3f 09 00 : 10001000 00111111 00001001 0
[04] {25} 88 3f 09 00 : 10001000 00111111 00001001 0
[05] {25} 88 3f 09 00 : 10001000 00111111 00001001 0
[06] {25} 88 3f 09 00 : 10001000 00111111 00001001 0
[07] {25} 88 3f 09 00 : 10001000 00111111 00001001 0

*** signal_start = 6394966, signal_end = 6446720
signal_len = 51754,  pulses = 126
Iteration 1. t: 97    min: 48 (82)    max: 147 (44)    delta 13
Iteration 2. t: 97    min: 48 (82)    max: 147 (44)    delta 0
Pulse coding: Short pulse length 48 - Long pulse length 147

Short distance: 50, long distance: 149, packet distance: 1535

p_limit: 97
bitbuffer:: Number of rows: 6 
[00] {1} 00 : 0
[01] {25} 88 3f 01 00 : 10001000 00111111 00000001 0
[02] {25} 88 3f 01 00 : 10001000 00111111 00000001 0
[03] {25} 88 3f 01 00 : 10001000 00111111 00000001 0
[04] {25} 88 3f 01 00 : 10001000 00111111 00000001 0
[05] {25} 88 3f 00 00 : 10001000 00111111 00000000 0

*** signal_start = 14870470, signal_end = 14928583
signal_len = 58113,  pulses = 151
Iteration 1. t: 102    min: 53 (93)    max: 152 (58)    delta 13
Iteration 2. t: 102    min: 53 (93)    max: 152 (58)    delta 0
Pulse coding: Short pulse length 53 - Long pulse length 152

Short distance: 45, long distance: 144, packet distance: 1530

p_limit: 102
bitbuffer:: Number of rows: 7 
[00] {1} 00 : 0
[01] {25} 88 3f 0a 00 : 10001000 00111111 00001010 0
[02] {25} 88 3f 0a 00 : 10001000 00111111 00001010 0
[03] {25} 88 3f 0a 00 : 10001000 00111111 00001010 0
[04] {25} 88 3f 0a 00 : 10001000 00111111 00001010 0
[05] {25} 88 3f 0a 00 : 10001000 00111111 00001010 0
[06] {25} 88 3f 00 00 : 10001000 00111111 00000000 0

*** signal_start = 15464167, signal_end = 15515940
signal_len = 51773,  pulses = 126
Iteration 1. t: 104    min: 54 (82)    max: 154 (44)    delta 8
Iteration 2. t: 104    min: 54 (82)    max: 154 (44)    delta 0
Pulse coding: Short pulse length 54 - Long pulse length 154

Short distance: 44, long distance: 143, packet distance: 1529

p_limit: 104
bitbuffer:: Number of rows: 6 
[00] {1} 00 : 0
[01] {25} 88 3f 02 00 : 10001000 00111111 00000010 0
[02] {25} 88 3f 02 00 : 10001000 00111111 00000010 0
[03] {25} 88 3f 02 00 : 10001000 00111111 00000010 0
[04] {25} 88 3f 02 00 : 10001000 00111111 00000010 0
[05] {25} 88 3f 00 00 : 10001000 00111111 00000000 0

```

Do you think it's easy to be fixed?

rct, 2016-01-31:
IIRC, that device has different timings depending on a passive like a resistor or capacitor choice.

Looks like you are receiving the signal very cleanly, so compare your timings from the analyzer against the pulse timings in the device definition, usually at the bottom of the source file.

If it's what I think, It's easily "fixed" by duplicating the device structure  to register a new device with the other timings, but that's not really a good solution.

Harvie, 2016-01-31:
@rct i am not sure which are the right values... rtl_sdr -A says this:

```
Detected OOK package
Analyzing pulses...
Total count:  201,  width: 50944		(203.8 ms)
Pulse width distribution:
 [ 0] count:  123,  width:    52 [50;54]	( 208 us)
 [ 1] count:   78,  width:   152 [150;154]	( 608 us)
Gap width distribution:
 [ 0] count:    8,  width:  1536 [1535;1538]	(6144 us)
 [ 1] count:   78,  width:    46 [45;49]	( 184 us)
 [ 2] count:  114,  width:   146 [144;148]	( 584 us)
Pulse period distribution:
 [ 0] count:    8,  width:  1588 [1586;1590]	(6352 us)
 [ 1] count:  192,  width:   198 [196;202]	( 792 us)
Level estimates [high, low]:  15980,     90
Frequency offsets [F1, F2]:   16891,      0	(+64.4 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package
Analyzing pulses...
Total count:  176,  width: 44589		(178.4 ms)
Pulse width distribution:
 [ 0] count:  107,  width:    52 [50;54]	( 208 us)
 [ 1] count:   69,  width:   152 [150;155]	( 608 us)
Gap width distribution:
 [ 0] count:    7,  width:  1536 [1534;1537]	(6144 us)
 [ 1] count:   69,  width:    46 [45;49]	( 184 us)
 [ 2] count:   99,  width:   146 [145;149]	( 584 us)
Pulse period distribution:
 [ 0] count:    7,  width:  1589 [1588;1590]	(6356 us)
 [ 1] count:  168,  width:   198 [196;202]	( 792 us)
Level estimates [high, low]:  15931,     86
Frequency offsets [F1, F2]:   16692,      0	(+63.7 kHz, +0.0 kHz)
Guessing modulation: No clue...
```

And devices/generic_remote.c says this:

```
...

PWM_Precise_Parameters pwm_precise_parameters_generic = {
	.pulse_tolerance	= 50,
	.pulse_sync_width	= 0,	// No sync bit used
};

r_device generic_remote = {
	.name			= "Generic Remote SC226x EV1527",
	.modulation		= OOK_PULSE_PWM_PRECISE,
	.short_limit	= 464,
	.long_limit		= 1404,
//tried this but not working:
//	.short_limit	= 600,
//	.long_limit		= 1600,
	.reset_limit	= 1800,
	.json_callback	= &generic_remote_callback,
	.disabled		= 0,
	.demod_arg		= (unsigned long)&pwm_precise_parameters_generic,
};
```

Anyway... what is better solution in your opinion? I guess that there may be some way to match codes without having to rely on fixed timing... just do some statistics to tell which codes are long and which are short...

Harvie, 2016-01-31:
I've made following changes to generic_remote.c:

```
        // Validate package
+       printf("Bits: %u\n", bits);
+       /*
        if ((bits == 25)
         && (b[3] & 0x80)       // Last bit (MSB here) is always 1
         && (b[0] || b[1])      // Reduce false positives. ID 0x0000 not supported
         && (b[2])      // Reduce false positives. CMD 0x00 not supported
-       ) {
+       ) */ {
```

```
-       .pulse_tolerance        = 50,
+       .pulse_tolerance        = 20,
```

```
-       .short_limit    = 464,
-       .long_limit             = 1404,
-       .reset_limit    = 1800,
+       .short_limit    = 50,
+       .long_limit             = 150,
+       .reset_limit    = 200,
```

And produced output after pressing the button is this:

```
Bits: 1
Generic remote keypress / sensor
ID 16bit = 0xFFFF
CMD 8bit = 0xFF
TRISTATE = 111111111111
```

I guess that there's something wrong as there should be more bits than 1...

merbanan, 2018-11-30:
This type of device should be handled with the flex device decoder now. And with no sample signals there is nothing more that can be done.

zuckschwerdt, 2018-11-30:
Does `rtl_433 -X` help?

zuckschwerdt, 2018-11-30:
Also try using `-A`, you will get a suggestion for a flex decoder.

zuckschwerdt, 2018-12-04:
Maybe we should add -d, -g, -X, -F, -r, and -w help output to the README along the -h and -R output there. I can adapt the maintainer-update to do that.

merbanan, 2018-12-05:
With the -A analyzer there is a suggestion for the flex decoder and -X help works fine also.

closed 2018-12-05 by merbanan