merbanan/rtl_433 #285: AlectoV1 Temperature Sensor

kropochev, 2016-02-01:
It shows incorrect data
`2016-02-01 20:27:05 AlectoV1 Temperature Sensor: House Code 159, Channel 2, Battery LOW, Temperature 204.70 C, Humidity 3 %`
It should be about temperature 0.4 C / 32.7 F, humidity 83%
Debug info:
```
rtl_433 -R 16 -D
Registering protocol "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 8000.
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
2016-02-01 20:36:38 AlectoV1 Temperature Sensor: House Code 159, Channel 2, Battery LOW, Temperature 204.70 C, Humidity 3 %
Checksum      = 0 (calculated 0)
Received Data = f9 ff fe c0 00
pulse_demod_ppm(): AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon) 
bitbuffer:: Number of rows: 8 
[00] {0} : 
[01] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[02] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[03] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[04] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[05] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[06] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[07] {23} f9 ff fe : 11111001 11111111 1111111
2016-02-01 20:38:26 AlectoV1 Temperature Sensor: House Code 159, Channel 2, Battery LOW, Temperature 204.70 C, Humidity 3 %
Checksum      = 0 (calculated 0)
Received Data = f9 ff fe c0 00
pulse_demod_ppm(): AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon) 
bitbuffer:: Number of rows: 8 
[00] {0} : 
[01] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[02] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[03] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[04] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[05] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[06] {28} f9 ff fe c0 : 11111001 11111111 11111110 1100
[07] {23} f9 ff fe : 11111001 11111111 1111111
```

merbanan, 2016-02-01:
Can you provide a list of 5 entries with the printed temperature, the correct temperature and the raw bitbuffer ?

jcarduino, 2016-02-01:
There are samples in the repository including protocol description. 

Mvg Johan Corstjens

Verstuurd via mobile

merbanan, 2016-02-01:
Well for me to debug this I need the information I listed. This actually looks like some other type of sensor. 28 vs 36 bits. The alecto decoder needs to have that check added so bad data can be filtered out better.

jcarduino, 2016-02-01:
Alecto also has a v2 protocol. Sounds to me this is a candidate. 

Mvg Johan Corstjens

Verstuurd via mobile

kropochev, 2016-02-01:
Unfortunately, I am not the owner of the sensor, I just saw the invalid value.
Now the sensor shows the following
```
2016-02-01 22:29:24 AlectoV1 Checksum/Parity error
2016-02-01 22:30:35 AlectoV1 Checksum/Parity error
```
Attached files and a text description, but how to find there the same sensor, I do not know.
[debug info](https://gist.github.com/kropochev/8d0a9d50fd6d12d8895b)
[data.tar.gz](https://github.com/merbanan/rtl_433/files/112678/data.tar.gz)

merbanan, 2016-02-01:
Ok, I'll make sure the alecto decoder rejects data it can not handle. This data can be pretty much anything.

merbanan, 2016-02-01:
Ok, the test files now pass. Hopefully it will reject more false positives now.

closed 2016-02-01 by merbanan

jcarduino, 2016-02-01:
From memory I think the decoder checks number of bytes (37) and than checks crc. 

i can test current revision as I have several electo V1 sensors but has to wait till weekend. However... How does decoder work if you feed it with the alecto v1 sample files? It is some time ago since I updated rtl433 so cannot verify behaviour right now.

What is exact type of sensor, please include picture.

Mvg 

Johan Corstjens

jcarduino, 2016-02-01:
What did you change?

Mvg 

Johan Corstjens

merbanan, 2016-02-01:
Just look at the commit log.

dterweij, 2018-11-08:
Saw this issue when I looked up some sensors that I receive here.
Temperature looks odd.

```
8-11-08 10:49:17 :   AlectoV1 Temperature Sensor
        House Code:      90
        Channel:         2
        Battery:         LOW
        Temperature:     154.40 C
        Humidity:        81 %
        :       CHECKSUM

```

zuckschwerdt, 2018-11-08:
Can you capture a sample for this (`-a -t`) and provide values what the sensor should have read?