merbanan/rtl_433 #302: Adding new or modified device to rtl_433

BudEnsleyIII, 2016-02-11:
Efergy TPM Tx as new device
I would like to add a new device but I am getting a Seg-Fault when I include the new device.  If I replace the efergy_e2_classic source with new source there is no Seg-Fault.  Name of source or callback name does not seem to part of the issue as replacement of source works and new source has new names.  It s only when both devices are included in the build does the Seg-Fault occur.

Running on a rpi2-jessie and Mac-Yosemite, testing with Elite e2 and Elite-pm Tx.

ygator, 2016-02-11:
You probably forgot to increase  MAX_PROTOCOLS in include/rtl_433.h

zuckschwerdt, 2016-02-11:
`MAX_PROTOCOLS` is easy to miss. In summary, when adding a new devices you need to touch these files:
- `include/rtl_433.h (MAX_PROTOCOLS)`
- `include/rtl_433_devices.h`
- `src/CMakeLists.txt`
- `src/Makefile.am`

BudEnsleyIII, 2016-02-11:
Thanks.  
Could someone add that info to the WIKI that tells what needs to be done when adding a device.

closed 2016-02-11 by BudEnsleyIII